pimoroni/hyperpixel4 #81: hyperpixel 4.0 screen blank running octoprint on a Raspberry Pi 4.0 8gb

minibusdriver, 2020-07-05:
To summarise I am having problem configuring a new hyperpixel 4.0 screen to work with Octoprint and a Raspberry Pi4 8gb.  The issue appears to be a software one as there appears to be evidence the screen is actually working. If any one can help resolve this matter it would be appreciated.

Please help

I am having problems setting up an hyper pixel 4.0 rectangular touchscreen with a raspberry pi4b 8gb. I would like to to use this to remotely monitor my 3D printer (Prusa I3 MK3s) with octoprint. Both were recently purchased from Pimoroni. This is my first Pi purchase and I am not familiar with Raspberry Pis, Terminal or Github so please be nice! However I am reasonably intelligent and enjoy learning new things and don't necessarily expect thing to work straight away.  I have spent several  days trying to get it to work properly using various forums and you tube videos. I must confess I am at a loss as what to do next.
I have connected the hyperpixel 4.0 to the GPIO header on the raspberry pi using the standoffs supplied with the screen. Nothing else is connected to the Raspberry Pi.  The Raspberry Pi is powered using a standard Raspberry USB C PSU. 
I successfully flashed Octoprint (OctoPI64-0.18.0-beta.img) to a 16gb sd card using etcher.  Maybe the fact that Octoprint is Beta version is the issue but according to the Octoprint webpage the standard version doesn’t play nicely with the 8gb Raspberry Pi4 so I have no alternative. I modified the WPA/WPA2 settings in |octopi-wpa-supplicant.txt and successfully logged onto the Raspberry Pi using a web browser (Safari). I configured Octoprint and then connected it to the printer and all appears to be normal.  For all of this time the screen was blank as expected as the screen drivers were not installed but you could see portions of the screen backlit.  
I then tried to follow this video to set up Octoscreen https://youtu.be/OJ59hXSyBoI which is where my problems started …..
(timeline 5.00 minutes) The curl https://get.pimoroni.com/hyperpixel4 | bash gave me the error message.  ‘This hardware is not supported, sorry! Config files have been left untouched’. 
So I used….

Last login: Sun Jul  5 17:01:53 on ttys000

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
phils-MBP-17530:~ philbrown$ ssh pi@octopi.local
pi@octopi.local's password: 
Linux octopi 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Jul  5 16:17:09 2020 from fe80::49d:4acf:331c:9023%wlan0

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

------------------------------------------------------------------------------
Access OctoPrint from a web browser on your network by navigating to any of:

    http://octopi.local
    http://192.168.1.95

https is also available, with a self-signed certificate.
------------------------------------------------------------------------------
OctoPrint version : 1.4.0
OctoPi version    : 0.18.0
------------------------------------------------------------------------------

pi@octopi:~ $ git clone https://github.com/pimoroni/hyperpixel4 -b pi4
Cloning into 'hyperpixel4'...
remote: Enumerating objects: 54, done.
remote: Counting objects: 100% (54/54), done.
remote: Compressing objects: 100% (41/41), done.
remote: Total 291 (delta 20), reused 36 (delta 12), pack-reused 237
Receiving objects: 100% (291/291), 730.19 KiB | 1.42 MiB/s, done.
Resolving deltas: 100% (135/135), done.

Which appeared to have worked. Followed by …..

pi@octopi:~ $ cd hyperpixel4
pi@octopi:~/hyperpixel4 $ sudo ./install.sh

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for pi: 
Notice: building hyperpixel4.dtbo
Created symlink /etc/systemd/system/multi-user.target.wants/hyperpixel4-init.service → /etc/systemd/system/hyperpixel4-init.service.
Installed: /usr/bin/hyperpixel4-init
Installed: /etc/systemd/system/hyperpixel4-init.service
Installed: /boot/overlays/hyperpixel4.dtbo
[... 119 lines cut ...]
You should see something like:

```text
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-7
```

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

For example:

```text
pi@raspberrypi:~ $ i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Gadgetoid, 2020-07-06:
When the screen is blank, are you able to ascertain if the backlight is on or off?

Also for the sake of everyone's sanity do you have time to try this setup using the regular Raspberry Pi OS, to eliminate the possibility that either the Pi or HyperPixel is at fault.

We don't test outside of Raspian/Raspberry Pi OS since there just aren't enough hours in the day (and by we, I mean.. there's only really me), but with any luck we'll get to the bottom of this. The HyperPixel4 is a particularly easy device to upset if OctoPrint does anything non-standard with video, or changes any GPIO pin state.

minibusdriver, 2020-07-06:
Philip
1st of all thanks for coming to my rescue 
Will give it a go tonight when I get home and let you know
Is there a Particular source for the OS you want me to use?
And full version not lite confirm?
I will try and get some pictures of the screen back light can’t remember exactly what I saw when now 🤦🏼

Cheers
Phil

Gadgetoid, 2020-07-06:
The last OS I've fired up and tested with was- "Raspberry Pi OS (32-bit) with desktop" - speciffically `2020-05-27-raspios-buster-armhf.zip` - direct link: https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2020-05-28/2020-05-27-raspios-buster-armhf.zip

This is the desktop (not lite) without the extra cruft version, so it's a 1.1G download.

I'll see if I can re-image this for testing.

Do you have a link to the specific Octoprint image you were using?

minibusdriver, 2020-07-06:
Hi

Here you go this is where I got the version of Octoprint from….

Octoprint on a pi4 8gb - Get Help - OctoPrint Community Forum <https://community.octoprint.org/t/octoprint-on-a-pi4-8gb/20670/23>

After your 1st Email I was going to use this Raspberry Pi Imager -

Raspberry Pi Downloads - Software for the Raspberry Pi <https://www.raspberrypi.org/downloads/>

shall I try it tonight or do you want me to wait until you reimage the other file?

Once again thanks for your help
Cheers
Phil

Gadgetoid, 2020-07-06:
Thanks I've imaged Octoprint and think I'm now encountering the same issue as you.

My initial debugging steps don't seem to suggest that any DPI interface is being set up at all, and now I can't seem to SSH back into my Pi. D'oh!

This might be a fundamental problem with the beta 64bit Raspberry Pi OS and need a bit of sleuthing and a bug report.

minibusdriver, 2020-07-06:
Im working through the install as you suggested….

So far ……
Ok so I downloaded Raspberry Pi imager for Mac OS from the raspberrypi.org site.  
I unpacked the imager and using Raspberry Pi Imager v1.3. flashed the recommended OS to the SD card. It was labelled ‘ a Port of Debian with the \raspberry Pi Desktop’ (although I note it was 32-Bit version...presumably still runs on 64Bit Pi) released 2020-05-27. The write was successful
I needed a headless setup so I followed these instructions…..
For headless setup, SSH can be enabled by placing a file named ssh, without any extension, onto the boot partition of the SD card from another computer. When the Pi boots, it looks for the ssh file. If it is found, SSH is enabled and the file is deleted. The content of the file does not matter; it could contain text, or nothing at all.

And also….

You will need to define a wpa_supplicant.conf file for your particular wireless network. Put this file in the boot folder, and when the Pi first boots, it will copy that file into the correct location in the Linux root file system and use those settings to start up wireless networking.
wpa_supplicant.conf file example:
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=<Insert 2 letter ISO 3166-1 country code here>

network={
 ssid="<Name of your wireless LAN>"
 psk="<Password for your wireless LAN>"
}

So on first boot up the Hyperpixel display was backlit with a blank screen (but no hyperpixel drivers installed so far).

I too am struggling to connect with SSH but I forgot to insert the country code in wpa_supplicant.conf.
Not sure if that is critical but will correct anyway.
To answer one of your other questions the screen is backlit now I have the raspberry Pi desktop on (see attached but wasn’t like this when using octoprint !)
Once again thanks for your help

Gadgetoid, 2020-07-06:
I failed to connect via SSH the first time, also, but it magically worked after I rebooted. 🤷 I knew WiFi was working, at least, because I could ping the Pi. Now imaging vanilla Raspberry Pi OS 64bit Beta to see if I encounter the same issue, because I totally broke my setup and had to do *something*

minibusdriver, 2020-07-06:
Hi
No that didn’t work either and I am unable to see my raspberry Pi on my network
I will try ethernet and see if I can see it that way 😩🤷🏼🤦🏼🔨
Cheers Phil

minibusdriver, 2020-07-06:
Sounds like you are having more fun than me!

minibusdriver, 2020-07-06:
Ok some progress….
I can see the raspberry pi on the network connected via ethernet and I was able to sucessfully ping.
Presumably I could load the the hyperpixel 4.0  Drivers through the router but you will have to explain how.  Sorry you are working with a total NOOb!!!
Cheers
Phil

Gadgetoid, 2020-07-07:
It looks like the `hyperpixel4.dtbo` just wont load on an 64bit Pi 4 configuration, or against a 5.x kernel. Device-tree stuff baffles me at the best of times, so I'm going to have to appeal to the powers-that-be for some help with this.

In the mean time I have managed to get display output to work (on the Raspberry Pi OS Beta 64bit image at least) with some changes to `/boot/config.txt`.

Specifically the end of my config now looks like this:

```
#dtoverlay=hyperpixel4
gpio=0-25=a2
gpio=19=op,dh
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```

Note: `dtoverlay=hyperpixel4` is commented-out because it just doesn't work.

Then we use `gpio=` to configure the alt-mode of the DPI (display parallel interface) pins to alt2 so the video signal is sent out on GPIO. And finally set BCM19 to an output, driven high to enable the backlight.

Touch input wont work at all with this config- unfortunately- but it might be possible to whip up a dtoverlay that *just* configures touch. Also the system wont be aware of the backlight in any meaningful way, so it will be always-on.

Finally, setting the display orientation with `hyperpixel4-rotate` wont persist because it fails to set the touch mapping and crashes out.

So... I have *some* progress.

minibusdriver, 2020-07-07:
Once again thanks for your help. Sorry for the late response only just got home. 

So at this stage is there anything else you want me to do?

I was unable to ‘ssh’ my pi yesterday despite setting up wpa_supplicant. I was able to access my pi through ssh previously using octoprint so I know the wireless side of thing is working on the pi. 

Cheers
Phil

Gadgetoid, 2020-07-07:
Right now I think there's nothing to be done- though I've just pushed up a Pi4 64bit beta branch you culd try it's still a work in progress: https://github.com/pimoroni/hyperpixel4/tree/pi4-64bit-beta

Install process should be as follows:

```
git clone https://github.com/pimoroni/hyperpixel4 -b pi4-64bit-beta
cd hyperpixel4
sudo ./install.sh
```

minibusdriver, 2020-07-07:
Ok will give it a go thanks

Cheers
Phil

mikenosko, 2020-07-27:
Just commented on the now closed #88 so putting it here too

I have the 2gb version of pi4 `b03112` and a rectangular hyperpixel4 (updated pi4 branch); i run an up-to-date octopi `Raspbian GNU/Linux 10 (buster)` with the `5.4.51-v7l+` kernel the 32bit one (`armv7l`)

Problem started after the last 1-3 updates: `cat /var/log/dpkg.log | grep upgrade`; im 100% sure it was working earlier in the day on the 22 july; here are the upgrades i did since to narrow it down:

> 2020-07-21 20:48:25 upgrade libnss3:armhf 2:3.42.1-1+deb10u2 2:3.42.1-1+deb10u3
> 2020-07-21 20:48:26 upgrade libopenmpt0:armhf 0.4.3-1 0.4.3-1+deb10u1
> 2020-07-21 20:48:26 upgrade libraspberrypi-doc:armhf 1.20200601-1 1.20200717-1
> 2020-07-21 20:48:29 upgrade libraspberrypi-dev:armhf 1.20200601-1 1.20200717-1
> 2020-07-21 20:48:29 upgrade raspberrypi-kernel:armhf 1.20200601-1 1.20200717-1
> 2020-07-21 20:49:23 upgrade libraspberrypi-bin:armhf 1.20200601-1 1.20200717-1
> 2020-07-21 20:49:24 upgrade libraspberrypi0:armhf 1.20200601-1 1.20200717-1
> 2020-07-21 20:49:25 upgrade raspberrypi-bootloader:armhf 1.20200601-1 1.20200717-1
> 2020-07-21 20:49:30 upgrade rpi-eeprom:all 7.5-1 7.8-1
> 2020-07-25 11:06:44 upgrade libraspberrypi-doc:armhf 1.20200717-1 1.20200723-1
> 2020-07-25 11:06:47 upgrade libraspberrypi-dev:armhf 1.20200717-1 1.20200723-1
> 2020-07-25 11:06:49 upgrade raspberrypi-kernel:armhf 1.20200717-1 1.20200723-1
> 2020-07-25 11:07:55 upgrade libraspberrypi-bin:armhf 1.20200717-1 1.20200723-1
> 2020-07-25 11:07:56 upgrade libraspberrypi0:armhf 1.20200717-1 1.20200723-1
> 2020-07-25 11:07:59 upgrade raspberrypi-bootloader:armhf 1.20200717-1 1.20200723-1
> 2020-07-26 14:31:58 upgrade raspberrypi-kernel:armhf 1.20200723-1 1.20200723-1
> 2020-07-26 14:33:07 upgrade raspberrypi-bootloader:armhf 1.20200723-1 1.20200723-1

The touch seems to be working - i touched the screen a couple of times as i have the screensaver on for it to wake up and while the screen didn't come on the hotend began to heat up. The screen comes on briefly when starting and then just blank with no backlight

`dmesg | grep Goodix`

> Goodix-TS 11-005d: 11-005d supply AVDD28 not found, using dummy regulator
> Goodix-TS 11-005d: 11-005d supply VDDIO not found, using dummy regulator
> Goodix-TS 11-005d: i2c test failed attempt 1: -6
> Goodix-TS 11-005d: i2c test failed attempt 2: -6
> Goodix-TS 11-005d: I2C communication failure: -6
> Goodix-TS 11-0014: 11-0014 supply AVDD28 not found, using dummy regulator
> Goodix-TS 11-0014: 11-0014 supply VDDIO not found, using dummy regulator
> Goodix-TS 11-0014: ID 911, version: 1060
> input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-11/11-0014/input/input0

`i2cdetect -y 11` the UU is in 4:10 as per your template

morphias2004, 2020-08-01:
Also posting details from my closed duplicate for the benefit of others.

**Describe the bug**

I previously opened this as a faulty on the OctoDash GitHub page: UnchartedBull/OctoDash#848

I have since conclusively ruled out OctoDash as a factor as a clean installation of OctoPrint/OctoPi, installing the HyperPixel 4 drivers and then running sudo apt update and sudo apt full-upgrade causes the issue.

This points to the HyperPixel 4 drivers having an issue with the latest Pi updates.

**To Reproduce**

Image SD card with the current OctoPi image

https://octopi.octoprint.org/latest

Update to Octoprint 1.4.0 from within OctoPi web interface and reboot.

Install HyperPixel 4 driver and reboot.

curl https://get.pimoroni.com/hyperpixel4 | bash

Screen is working correctly at this point and is in 'portait'.

Run latest Pi updates

sudo apt update

sudo apt full-upgrade

Reboot and you get a black screen. OctoPi is up and running and available via the web interface and SSH.

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

    Is it Square or Rectangular?
    **Rectangular**

    Touch or non Touch?
    **Touch**

    Approximately when was it purchased
    **2 months ago**

    How is it connected to your Pi
    **Direct GPIO onto Raspberry Pi 4 4GB**

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

    What model of Raspberry Pi are you using (the result of cat /proc/cpuinfo | grep Revision can tell us this exactly
    **c03112**

    Which OS release are you using: lsb_release --description and uname -r
    **Raspbian GNU/Linux 10 (buster)
    4.19.75-v7l+**

    Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?
    **Fresh**

**Extra debugging information**

If you're having a problem with touch, try checking dmesg for related errors:

    dmesg | grep Goodix for HyperPixel 4.0" Rectangular
    **Blank response to that command**

    dmesg | grep ft5 for HyperPixel 4.0" Square
[... 198 lines cut ...]
Get:136 http://archive.raspberrypi.org/debian buster/main armhf libdrm-radeon1 armhf 2.4.100-4+rpi1bpo10+1 [29.0 kB]
Get:137 http://archive.raspberrypi.org/debian buster/main armhf libllvm9 armhf 1:9.0.1-6+rpi1bpo10+1 [13.1 MB]
Get:132 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf libncursesw5 armhf 6.1+20181013-2+deb10u2 [94.3 kB]
Get:138 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf libtinfo5 armhf 6.1+20181013-2+deb10u2 [309 kB]
Get:139 http://mirror.datamossa.io/raspbian buster/main armhf libpango-1.0-0 armhf 1.42.4-8deb10u1 [169 kB]
Get:140 http://mirror.datamossa.io/raspbian buster/main armhf libpangoft2-1.0-0 armhf 1.42.4-8deb10u1 [62.3 kB]
Get:141 http://mirror.datamossa.io/raspbian buster/main armhf libpangocairo-1.0-0 armhf 1.42.4-8deb10u1 [52.0 kB]
Get:142 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf libss2 armhf 1.44.5-1+deb10u3 [71.0 kB]
Get:143 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf libxmuu1 armhf 2:1.1.2-2+b3 [23.4 kB]
Get:144 http://archive.raspberrypi.org/debian buster/main armhf libgl1-mesa-dri armhf 19.3.2-1bpo10+1rpt1 [5,550 kB]
Get:145 http://mirror.datamossa.io/raspbian buster/main armhf python-apt-common all 1.8.4.1 [95.7 kB]
Get:146 http://mirror.datamossa.io/raspbian buster/main armhf python3-apt armhf 1.8.4.1 [167 kB]
Get:147 http://archive.raspberrypi.org/debian buster/main armhf libpam-chksshpwd armhf 1.3.1-5+rpt2 [81.7 kB]
Get:148 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-sys-mods armhf 20200514 [10.6 kB]
Get:149 http://archive.raspberrypi.org/debian buster/main armhf pi-bluetooth all 0.1.13 [4,908 B]
Get:150 http://archive.raspberrypi.org/debian buster/main armhf rpi.gpio-common armhf 0.7.0-0.1bpo10+1 [5,776 B]
Get:151 http://archive.raspberrypi.org/debian buster/main armhf python-rpi.gpio armhf 0.7.0-0.1~bpo10+1 [20.8 kB]
Get:152 http://archive.raspberrypi.org/debian buster/main armhf raspi-config all 20200707 [26.4 kB]
Get:153 http://archive.raspberrypi.org/debian buster/main armhf rpi-eeprom-images all 7.8-1 [6,112 B]
Get:154 http://archive.raspberrypi.org/debian buster/main armhf rpi-eeprom all 7.8-1 [316 kB]
Get:155 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf rpcbind armhf 1.2.5-0.3+deb10u1 [41.2 kB]
Get:156 http://archive.raspberrypi.org/debian buster/main armhf rpi-update all 20200409 [7,488 B]
Get:157 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf sudo armhf 1.8.27-1+deb10u2 [1,201 kB]
Get:158 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf wpasupplicant armhf 2:2.7+git20190128+0c1e29f-6+deb10u2 [1,008 kB]
Fetched 258 MB in 2min 11s (1,968 kB/s)

TxBillbr, 2020-08-02:
The biggest thing I see is that the xwindow system is not running at all. I suspect that the desktop components are missing because the octopi image is built without them as a headless system.

Gadgetoid, 2020-08-03:
In all cases it's worth first trying the Kernel 5.4.51 fixes in #89 

```
git clone https://github.com/pimoroni/hyperpixel4 -b pi4-64bit-beta
cd hyperpixel4
sudo ./install.sh
```

I believe OctoPi is academic in this case, since these same problems occur on stock Raspberry Pi OS when updating.

mikenosko, 2020-08-03:
the `pi4-64bit-beta` fixes the blank screen issue but now i can't seem to able to turn the screen off! i know...

tried `xset force off` editing `/sys/class/... bl_power/brightness` even `gpio toggle 24` (tbh didn't check if the command actually toggled the pin or if its the right pin - should be BCM19?) but none of these work for me

Gadgetoid, 2020-08-03:
You're a step ahead of me! I'm just updating my non-64bit OctoPi image now. IIRC I had backlight working via the /sys/class interface, but the whole shebang is nothing if not a miserable mess when it comes to display power management.

The dtoverlay still sets up the backlight module, and the /sys/class interface being present is a good sign.

The backlight pin is BCM 19.

Gadgetoid, 2020-08-03:
Just before I reboot with the new updates, I can confirm on Kernel 4.19.75-v7l+ with these new changes to the HP4 overlay that the following works:

Backlight off: `echo 1 | sudo tee /sys/class/backlight/rpi_backlight/bl_power`
Backlight on: `echo 0 | sudo tee /sys/class/backlight/rpi_backlight/bl_power`

TxBillbr, 2020-08-03:
I have been running **32-bit** (waiting to take on yet another early system). For that I have published a my work-around for the 32-bit space. See https://github.com/TxBillbr/octodash-hyperpixel-fix . the 3Dprinting community is rapidly adopting OctoDash running on a pi screen and the HyperPixel has been difficult for a number of people. The work-around has been tested on 4.19.118. and I am testing 5.4 today.

reneb86, 2020-08-03:
Issue #88 has been marked as being a duplicate of this, but reading the thread, this seems to be focusing on there being a bug in how it runs on the 64bit version of Raspberry OS. But right now I'm seeing how 2x Pi4@2G and 2x Pi4@4G are all crashing whenever I do:

1. - fresh flash of Raspberry OS (32bit) using Pi imager 
2. - sudo apt-get upgrade
3. - hyperpixel4 install via curl

The only way it works is if I skip step 2. Something in the upgrade is breaking things, and #88 describes it better. Pi3B has no issues whatsoever even when upgrading.

TxBillbr I'm keen to try out your workaround tomorrow and see if that solves my issue. 

My test setup for whoever is interested (4x Pi3B, 2x Pi4@2G, 2x Pi4@4G): [pic](https://rengineering86.files.wordpress.com/2020/08/8pis3.gif?w=460)

TxBillbr, 2020-08-03:
I upgraded to 5.4 kernel and the solution breaks. Here are the dmesg lines for Goodix...

TxBillbr, 2020-08-03:
Same problem as @mikenosko. Only difference is pi revision C03111

Gadgetoid, 2020-08-04:
@TxBillbr in my case I have `Linux octopi 5.4.51-v7l+ #1331 SMP Fri Jul 31 16:51:51 BST 2020 armv7l` and touch input is working correctly.

If you're running HyperPixel4 on a Pi 4 you might be running into the race condition on startup that causes the touchscreen to swap i2c address. You should try the `patch-pi4-i2c-fix` branch which has both the i2c fix and Kernel 5.x fixes- https://github.com/pimoroni/hyperpixel4/tree/patch-pi4-i2c-fix

Gadgetoid, 2020-08-04:
@reneb86 it is my understanding that the problem, and solution, for both Pi 4 64bit and 32bit are the same. I've confirmed this on a Pi 4 4GB using both the OctoPi 64bit beta and latest OctoPi 32bit release.

For anyone wanting to try the fix, this is the version without the weird i2c glitch fix:

```
git clone https://github.com/pimoroni/hyperpixel4 -b pi4-64bit-beta
cd hyperpixel4
sudo ./install.sh
```

And if touch doesn't work for you, remove that clone, remove `dtoverlay=hyperpixel4` from `/boot/config.txt` and run the i2c-fix version of the installer:

```
git clone https://github.com/pimoroni/hyperpixel4 -b patch-pi4-i2c-fix
cd hyperpixel4
sudo ./install.sh
```

You can check which i2c address the touchscreen is using:

```
sudo modprobe i2c-dev
ls /dev/i2c-*
i2cdetect -y <number of i2c bus found above>
```

It will be either 0x14 or 0x54.

To re-iterate above there is currently an issue with the backlight control kernel module on the released version of kernel 5.4.51. We'll have to sit tight for a fix to that.

TxBillbr, 2020-08-05:
@Gadgetoid#4451, I did a full gamut of testing with the patch branch and it seems that applying the installation on a 4.x kernel machine and then upgrading still breaks the solution (yields a black screen, no backlight).  Running it on a clean installed Raspian 5.4-based image seems to work well. In the upgrade scenario I get no i2c ports and no DSI shows up. The /devices tree does have the i2c input assigned. xrandr sees no screen to manage (as one might expect). On shutdown I get a quick glimpse of the kernel shutdown messaging. So something is running that prevents the operation of the screen. As soon as that is killed the screen operates for a split second before it is killed too. I have a video of it if that is useful.

pi@octopi:~ $ dmesg | grep Goodix
[    6.023807] Goodix-TS 11-0014: 11-0014 supply AVDD28 not found, using dummy regulator
[    6.023933] Goodix-TS 11-0014: 11-0014 supply VDDIO not found, using dummy regulator
[    6.025951] Goodix-TS 11-0014: ID 911, version: 1060
[    6.048953] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-11/11-0014/input/input11

Gadgetoid, 2020-08-05:
@TxBillbr ah! You've set the little rusty cogs in my brain in motion and a thought has clicked into place- 

Insofar as I understand, a dtoverlay may be compiled differently depending upon the running kernel. This means the overlay compiled against 4.x, and unmanaged by the official upstream kernel packages, will sit unchanged during the upgrade to 5.x and ultimately be incompatible. D'oh!

The correct fix here is for us to upstream our dtoverlay so that a new version is installed for each kernel update. There's a small hitch here in that our i2c-compatibility fudge probably wouldn't meet the scrutiny for upstream inclusion. This is something I haven't pushed for before, but I'm going to look into it now.

The other problem with upstream is it's generally a much slower process to get a modified dtoverlay through the pipeline and onto a users Pi. I don't think our dtoverlays change all that much though, so perhaps that isn't a problem. Maybe also there's a more immediate fix to this to catch issues during upgrades... but I'm not sure.

Thank you for your thorough testing!

TxBillbr, 2020-08-05:
@Gadgetoid Just did an rpi-update and the kernel 5.4.51-v7l+ #1332.. Screen is on! Rotation and i2c don't appear to be correct, but continuing tests.

tusing, 2020-08-09:
> For anyone wanting to try the fix, this is the version without the weird i2c glitch fix:
>
> ```
> git clone https://github.com/pimoroni/hyperpixel4 -b pi4-64bit-beta
> cd hyperpixel4
> sudo ./install.sh [...]

Just did an apt full-upgrade, and had the problem in this thread after. My display is still completely black after running these steps. Any idea what might be the issue?

Gadgetoid, 2020-08-09:
@tusing if you've used the `patch-pi4-i2c-fix` make sure you don't have `dtoverlay=hyperpixel4` in your `/boot/config.txt`- you should have `hyperpixel4-common` and the two i2c address overlays instead. It imght be that these are conflicting and breaking it for you.

tusing, 2020-08-11:
> @tusing if you've used the `patch-pi4-i2c-fix` make sure you don't have `dtoverlay=hyperpixel4` in your `/boot/config.txt`- you should have `hyperpixel4-common` and the two i2c address overlays instead. It imght be that these are conflicting and breaking it for you.

Thanks @Gadgetoid  - I already did that though. Here's what I did:

* `git checkout pi4-64bit-beta && sudo ./install.sh`
* `git checkout  git checkout pi4-i2c-fix && sudo ./install.sh`
* Comment out `dtoverlay=hyperpixel4` in `boot.txt`

Here is my `/config/boot.txt` now:

```
# (Comments removed)
dtparam=audio=on

[pi4]
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6

dtoverlay=hyperpixel4-common
dtoverlay=hyperpixel4-0x14
dtoverlay=hyperpixel4-0x5d
gpio=0-25=a2
```

Unfortunately I still have a black screen :(

Gadgetoid, 2020-08-11:
If you're not building from a fresh GitHub directory be careful that the dtoverlay files might not regenerate- you should remove `*.dtbo` files from dist/ and try building again. This caught me and I probably need to force rebuild in all cases.

Gadgetoid, 2020-08-11:
I have now merged #89 and #92 to hopefully head off some more instances of this issue.

Anyone who has upgraded from Kernel 4.x to 5.x will need to re-run the HyperPixel 4 installer.

TxBillbr, 2020-08-12:
@Gadgetoid Thanks for all your efforts on this. I have updated my OctoDash document to reflect the current state. https://github.com/TxBillbr/octodash-hyperpixel-fix

Gadgetoid, 2020-08-12:
@TxBillbr thanks for all your feedback & support here. Though I think I'm far from done with this cursed display 😆 

I guess my next port of call for this is to see how much of the HyperPixel 4 driver I can get upstreamed so we don't have a repeat of this issue with another kernel release.

carcamerarig, 2020-09-21:
I have this same issue where the screen is blank with a small fixed cusor top left screen, Pi4b 8gb 4" hyperpixel and loaded up 18.1 beta. Dcustom Carlin has been helping me but we are now properly stuck, did this issue ever get resolved?

Thanks guys
Justin

carcamerarig, 2020-09-21:
[video-1600630855.mp4.zip](https://github.com/pimoroni/hyperpixel4/files/5256597/video-1600630855.mp4.zip)
Here's a quick preview on boot up

divinenanny, 2020-09-29:
While I now have a picture again (after removing all old hyperpixel stuff and running the install from the branch patch-pi4-i2c-fix), it flickers in lines. As if the image is moving quickly from top to bottom and leaving some artifacts on the screen. Touch works, backlight works.

My setup:
OS: Raspbian GNU/Linux 10 (buster) armv7l
Host: Raspberry Pi 4 Model B Rev 1.1
Kernel: 5.4.51-v7l+
Hyperpixel 4 Rectangle (in default rotation)

morphias2004, 2020-09-29:
> 
> 
> @TxBillbr thanks for all your feedback & support here. Though I think I'm far from done with this cursed display 😆
> 
> I guess my next port of call for this is to see how much of the HyperPixel 4 driver I can get upstreamed so we don't have a repeat of this issue with another kernel release.

@Gadgetoid 

Are you able to provide an update on this issue?

I imagine a lot of users, my self included; have been holding off on running updates on our Pi's becuase we'll lose our displays.

Pimoroni really need to get this sorted because they are still selling the screens with no bulletin to new owner's that their screens are not going to work.

TxBillbr, 2020-09-30:
I'll try to spend some time this week to update my pi4s with latest code and os updates to determine latest status from my side.

morphias2004, 2020-10-25:
> 
> 
> I'll try to spend some time this week to update my pi4s with latest code and os updates to determine latest status from my side.

Hi @TxBillbr ,

Any updates on how you went.

I know there are a bunch of us waiting to run updates on our Pi4's to fix/allow updating of other Octoprint related plugins, etc.

TxBillbr, 2020-10-25:
Sorry, I haven't gotten to it. I've been sidelined with some health issues for a few weeks. I'll do my best to get to that as soon as possible.

morphias2004, 2020-11-28:
Hi @TxBillbr 

Can we please have an update on this?

TxBillbr, 2020-11-28:
I run a large Holiday lighting show every year. It will likely be a couple of weeks before I can get to this. I'll do my best to get to this.

kronflux, 2021-05-20:
Has there since been any progress on this? Running out of options and may have to return screen..

TxBillbr, 2021-05-26:
I haven't looked at it in a while and thought it had been resolved with a 
later firmware/kernel release.

Gadgetoid, 2025-01-08:
I think this may have been resolved by a newer OS or firmware?

Closing to clear our decks for 2025 but please re-open (or open a new issue) if you need to.