pimoroni/hyperpixel4 #82: Weird banding using Hyperpixel Rectangular with RPi4

docbobo, 2020-07-07:
**Describe the bug**

From the moment the RPi4 enables video output, there is weird band of colors on the screen, which fades out after a while. However, it seems to reappear whenever significant changes are made to the screen output.

[image]

**To Reproduce**

Always happens for me.

**Your HyperPixel 4**

Hyperpixel 4 Rectangular on a RPi  4. Bought December 2019, never used until now.

**Your Raspberry Pi**

RPi4, 4GB. Running latest version of OctoPi.

Gadgetoid, 2020-07-12:
Whoa- I can't say I've seen this before.

It could- potentially- be related to something else running on the Pi, but I don't quite see how.

I'll ask our engineers/testers to see if anything like this has cropped up.

docbobo, 2020-07-12:
It does happen immediately when the system starts up, right when the raspberries appear at the top. So I suspect it’s not related to anything running in the Pi. 

The RPi4 is one of the very first, btw.

Gadgetoid, 2020-07-13:
I think it's probably a hardware issue with the screen. You should drop an email to support@pimoroni.com to proceed.

closed 2020-07-20 by Gadgetoid