pimoroni/hyperpixel4 #87: hyperpixel4-rotate XRandR "No protocol specified"

Gadgetoid, 2020-07-23:
When running `hyperpixel4-rotate` it can fail with `Exception: XRandR returned error code 1: No protocol specified`.

EG:

```
pi@raspberrypi:~ $ DISPLAY=:0.0 hyperpixel4-rotate left
Rotating display
No protocol specified
Can't open display :0.0
Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 43, in __init__
    version_output = self._output("--version")
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 64, in _output
    raise Exception("XRandR returned error code %d: %s"%(status,err))
Exception: XRandR returned error code 1: No protocol specified
Can't open display :0.0

Setting matrix: 0 -1 1 1 0 0
No protocol specified
Unable to connect to X server
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules
```

To fix this, the `XAUTHORITY` environment variable must be set. For example:

```
export XAUTHORITY=/home/pi/.Xauthority
```

I haven't run into this before, so I'm not sure if it's a Raspberry Pi OS problem or just some specific quirk of my setup. But I have put it here for posterity, and for potential inclusion of this fix into the rotate tool.

tusing, 2020-08-11:
Unfortunately, even with `XAUTHORITY` set, I still get the error:

```
$ XAUTHORITY=/home/pi/.Xauthority DISPLAY=:0.0 sudo hyperpixel4-rotate right
Rotating display
xrandr: Failed to get size of gamma for output default
warning: output DSI-1 not found; ignoring
/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py:66: UserWarning: XRandR wrote to stderr, but did not report an error (Message was: 'xrandr: Failed to get size of gamma for output default\n')
  warnings.warn("XRandR wrote to stderr, but did not report an error (Message was: %r)"%err)
Saving display settings to /usr/share/dispsetup.sh
Setting matrix: 0 1 0 -1 0 1
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules
```

Gadgetoid, 2020-08-11:
`dtoverlay=vc4-fkms-v3d` *must* be enabled for any of the new fancy HyperPixel 4 stuff to work. All the new Pi 4 multi-display support and X desktop rotation relies upon it.

I suspect you've probably still got an issue with your HyperPixel 4 install- I've replied on the main issue thread, but stuff is quickly getting buried there. You need to make sure your dtoverlay (`*.dtbo`) files are actually being rebuilt, and that the installer isn't just copying old ones from the local GitHub repo you're updating (they're out of tree, so they don't disappear when you switch branch).

closed 2026-08-25 by Gadgetoid