pimoroni/hyperpixel4 #88: HyperPixel 4.0 Touch black screen after Pi updates

morphias2004, 2020-07-25:
**Describe the bug**

I previously opened this as a faulty on the OctoDash GitHub page:  https://github.com/UnchartedBull/OctoDash/issues/848

I have since conclusively ruled out OctoDash as a factor as a clean installation of OctoPrint/OctoPi, installing the HyperPixel 4 drivers and then running **sudo apt update** and **sudo apt full-upgrade** causes the issue.

This points to the HyperPixel 4 drivers having an issue with the latest Pi updates.

**To Reproduce**

Image SD card with the current OctoPi image

https://octopi.octoprint.org/latest

Update to Octoprint 1.4.0 from within OctoPi web interface and reboot.

Install HyperPixel 4 driver and reboot.

curl https://get.pimoroni.com/hyperpixel4 | bash

Screen is working correctly at this point and is in 'portait'.

Run latest Pi updates

sudo apt update

sudo apt full-upgrade

Reboot and you get a black screen. OctoPi is up and running and available via the web interface and SSH.

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular?
**Rectangular**

2. Touch or non Touch?
**Touch**

3. Approximately when was it purchased
**2 months ago**

4. How is it connected to your Pi
**Direct GPIO onto Raspberry Pi 4 4GB**

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
**c03112**

2. Which OS release are you using: `lsb_release --description` and `uname -r`
**Raspbian GNU/Linux 10 (buster)
4.19.75-v7l+**

3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?
**Fresh**

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular
**Blank response to that command**

* `dmesg | grep ft5` for HyperPixel 4.0" Square

And check i2c is working:
[... 199 lines cut ...]
Get:136 http://archive.raspberrypi.org/debian buster/main armhf libdrm-radeon1 armhf 2.4.100-4+rpi1bpo10+1 [29.0 kB]
Get:137 http://archive.raspberrypi.org/debian buster/main armhf libllvm9 armhf 1:9.0.1-6+rpi1bpo10+1 [13.1 MB]
Get:132 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf libncursesw5 armhf 6.1+20181013-2+deb10u2 [94.3 kB]
Get:138 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf libtinfo5 armhf 6.1+20181013-2+deb10u2 [309 kB]
Get:139 http://mirror.datamossa.io/raspbian buster/main armhf libpango-1.0-0 armhf 1.42.4-8deb10u1 [169 kB]
Get:140 http://mirror.datamossa.io/raspbian buster/main armhf libpangoft2-1.0-0 armhf 1.42.4-8deb10u1 [62.3 kB]
Get:141 http://mirror.datamossa.io/raspbian buster/main armhf libpangocairo-1.0-0 armhf 1.42.4-8deb10u1 [52.0 kB]
Get:142 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf libss2 armhf 1.44.5-1+deb10u3 [71.0 kB]
Get:143 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf libxmuu1 armhf 2:1.1.2-2+b3 [23.4 kB]
Get:144 http://archive.raspberrypi.org/debian buster/main armhf libgl1-mesa-dri armhf 19.3.2-1bpo10+1rpt1 [5,550 kB]
Get:145 http://mirror.datamossa.io/raspbian buster/main armhf python-apt-common all 1.8.4.1 [95.7 kB]
Get:146 http://mirror.datamossa.io/raspbian buster/main armhf python3-apt armhf 1.8.4.1 [167 kB]
Get:147 http://archive.raspberrypi.org/debian buster/main armhf libpam-chksshpwd armhf 1.3.1-5+rpt2 [81.7 kB]
Get:148 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-sys-mods armhf 20200514 [10.6 kB]
Get:149 http://archive.raspberrypi.org/debian buster/main armhf pi-bluetooth all 0.1.13 [4,908 B]
Get:150 http://archive.raspberrypi.org/debian buster/main armhf rpi.gpio-common armhf 0.7.0-0.1bpo10+1 [5,776 B]
Get:151 http://archive.raspberrypi.org/debian buster/main armhf python-rpi.gpio armhf 0.7.0-0.1~bpo10+1 [20.8 kB]
Get:152 http://archive.raspberrypi.org/debian buster/main armhf raspi-config all 20200707 [26.4 kB]
Get:153 http://archive.raspberrypi.org/debian buster/main armhf rpi-eeprom-images all 7.8-1 [6,112 B]
Get:154 http://archive.raspberrypi.org/debian buster/main armhf rpi-eeprom all 7.8-1 [316 kB]
Get:155 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf rpcbind armhf 1.2.5-0.3+deb10u1 [41.2 kB]
Get:156 http://archive.raspberrypi.org/debian buster/main armhf rpi-update all 20200409 [7,488 B]
Get:157 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf sudo armhf 1.8.27-1+deb10u2 [1,201 kB]
Get:158 http://mirror.internode.on.net/pub/raspbian/raspbian buster/main armhf wpasupplicant armhf 2:2.7+git20190128+0c1e29f-6+deb10u2 [1,008 kB]
Fetched 258 MB in 2min 11s (1,968 kB/s)

Gadgetoid, 2020-07-27:
Duplicate of https://github.com/pimoroni/hyperpixel4/issues/81

closed 2020-07-27 by Gadgetoid

mikenosko, 2020-07-27:
Happened to me as well.

I have the 2gb version of pi4 `b03112` and a rectangular hyperpixel4; i run an up-to-date octopi `Raspbian GNU/Linux 10 (buster)` with the `5.4.51-v7l+` kernel the 32bit one (`armv7l`)

Problem started after the last 1-3 updates: `cat /var/log/dpkg.log | grep upgrade`; im 100% sure it was working earlier in the day on the 22 july, so here are the upgrades i did since to narrow it down:

> 2020-07-21 20:48:25 upgrade libnss3:armhf 2:3.42.1-1+deb10u2 2:3.42.1-1+deb10u3
2020-07-21 20:48:26 upgrade libopenmpt0:armhf 0.4.3-1 0.4.3-1+deb10u1
2020-07-21 20:48:26 upgrade libraspberrypi-doc:armhf 1.20200601-1 1.20200717-1
2020-07-21 20:48:29 upgrade libraspberrypi-dev:armhf 1.20200601-1 1.20200717-1
2020-07-21 20:48:29 upgrade raspberrypi-kernel:armhf 1.20200601-1 1.20200717-1
2020-07-21 20:49:23 upgrade libraspberrypi-bin:armhf 1.20200601-1 1.20200717-1
2020-07-21 20:49:24 upgrade libraspberrypi0:armhf 1.20200601-1 1.20200717-1
2020-07-21 20:49:25 upgrade raspberrypi-bootloader:armhf 1.20200601-1 1.20200717-1
2020-07-21 20:49:30 upgrade rpi-eeprom:all 7.5-1 7.8-1
2020-07-25 11:06:44 upgrade libraspberrypi-doc:armhf 1.20200717-1 1.20200723-1
2020-07-25 11:06:47 upgrade libraspberrypi-dev:armhf 1.20200717-1 1.20200723-1
2020-07-25 11:06:49 upgrade raspberrypi-kernel:armhf 1.20200717-1 1.20200723-1
2020-07-25 11:07:55 upgrade libraspberrypi-bin:armhf 1.20200717-1 1.20200723-1
2020-07-25 11:07:56 upgrade libraspberrypi0:armhf 1.20200717-1 1.20200723-1
2020-07-25 11:07:59 upgrade raspberrypi-bootloader:armhf 1.20200717-1 1.20200723-1
2020-07-26 14:31:58 upgrade raspberrypi-kernel:armhf 1.20200723-1 1.20200723-1
2020-07-26 14:33:07 upgrade raspberrypi-bootloader:armhf 1.20200723-1 1.20200723-1

The touch seems to be working - i touched the screen a couple of times as i have the screensaver on for it to wake up and while the screen didn't come on the hotend began to heat up. The screen comes on briefly when restarting and then just blank with no backlight

`dmesg | grep Goodix`

> Goodix-TS 11-005d: 11-005d supply AVDD28 not found, using dummy regulator
Goodix-TS 11-005d: 11-005d supply VDDIO not found, using dummy regulator
Goodix-TS 11-005d: i2c test failed attempt 1: -6
Goodix-TS 11-005d: i2c test failed attempt 2: -6
Goodix-TS 11-005d: I2C communication failure: -6
Goodix-TS 11-0014: 11-0014 supply AVDD28 not found, using dummy regulator
Goodix-TS 11-0014: 11-0014 supply VDDIO not found, using dummy regulator
Goodix-TS 11-0014: ID 911, version: 1060
input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-11/11-0014/input/input0

`i2cdetect -y 11` the UU is in 4:10 as per your template