pimoroni/hyperpixel4 #90: Banding appears when screen content changes, and disappears after time.

TheLion, 2020-07-28:
**Describe the bug**

Just got the Hyperpixel4 Rectangular working on my Raspberry Pi 4B 1Gb. 

Every time the content on the screen changes (initiated from the screen or remotely), some white bands appear. These also appear when first starting up.

After some time, the banding disappears.

See video here: https://youtu.be/YOKXyE8YWiU

**To Reproduce**

Just change anything on the screen. Even clicking in a blank space, which moves the mouse-cursor.

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Rectangular
2. Touch
3. 2 days ago
4. i2c

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. Revision        : a03111
2. Description:    Raspbian GNU/Linux 10 (buster) 5.4.51-v7l+
3. Fresh setup

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

No problem with touch

Gadgetoid, 2020-07-28:
Did you install more recently than 5 days ago? I merged some screen configuration tweaks that may rectify this: https://github.com/pimoroni/hyperpixel4/pull/84/files

May be worth checking first if your `/usr/bin/hyperpixel4-init` is the new or old version, but grabbing the updated `hyperpixel4-init` and replacing your local `/usr/bin/hyperpixel4-init` should suffice to see if these updates fix your problem- https://github.com/pimoroni/hyperpixel4/blob/pi4/dist/hyperpixel4-init

TheLion, 2020-08-02:
Yes, I installed it on the day I created the issue ;-)

Will take a look at the options you mentioned and get back to you.

TheLion, 2020-08-15:
I checked my /usr/bin/hyperpixel4-init and it is exactly the same as the one you linked to.

Replaced it just to be sure, but no difference in the banding unfortunately.

TheLion, 2020-09-01:
Anything I can do/check to solve this banding issue?

Gadgetoid, 2020-09-18:
Sorry for the delay responding- I believe you've got in touch with support?

We don't know if this is a hardware or software issue yet, but it's being looked into.

TheLion, 2020-09-18:
No problem. I indeed contacted support and a replacement is on the way.

TheLion, 2021-01-28:
Screen has been replaced and issue is solved.

closed 2021-01-28 by TheLion