pimoroni/hyperpixel4 #93: Black screen on Pi 4 and hyperpixel 4

rfkinline, 2020-08-08:
**Describe the bug**

What's gone wrong?
I am getting a blank screen. The display seems to be working as I was able to see it flickering at startup

**To Reproduce**

What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?
Brand new Raspberry Pi OS installation + I followed the install for the Hyperpixel 4 (pi 4)

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular? Rectangular
2. Touch or non Touch? non
3. Approximately when was it purchased this week
4. How is it connected to your Pi: GPIO

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
2. Which OS release are you using: `lsb_release --description` and `uname -r`
3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular
* `dmesg | grep ft5` for HyperPixel 4.0" Square

pi@raspberrypi:~ $ dmesg | grep Goodix
[    5.605515] Goodix-TS 11-0014: 11-0014 supply AVDD28 not found, using dummy regulator
[    5.605633] Goodix-TS 11-0014: 11-0014 supply VDDIO not found, using dummy regulator
[    5.606407] Goodix-TS 11-0014: i2c test failed attempt 1: -6
[    5.652077] Goodix-TS 11-0014: i2c test failed attempt 2: -6
[    5.701486] Goodix-TS 11-0014: I2C communication failure: -6

And check i2c is working:

`ls /dev/i2c-*`
/dev/i2c-11

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-11
pi@raspberrypi:~ $ i2cdetect -y 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

Gadgetoid, 2020-08-11:
This is most likely another instance of - https://github.com/pimoroni/hyperpixel4/issues/81

You'll need to reinstall HyperPixel 4 so that the dtoverlays build against the new kernel, and include the latest fixes:

```
git clone https://github.com/pimoroni/hyperpixel4 -b patch-pi4-i2c-fix
cd hyperpixel4
sudo ./install.sh
```

rfkinline, 2020-08-11:
I tried that one many times. In all variations including changing the config.txt file. I also tried with different Pis and different SD cards. It did not solve the problem. I guess it has to do with the operating system as when I installed according to these instructions: [https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-hyperpixel-4]  I got it working. The only difference is the version of the downloaded Raspberry Pi OS.

Gadgetoid, 2021-05-20:
Closing stale issue. Please contact support@pimoroni.com if you're still having problems.

closed 2021-05-20 by Gadgetoid