pimoroni/hyperpixel4 #95: Hyperpixel4 and 64-bit Ubuntu

alexanderbsd, 2020-08-20:
Hi all,

I understand that 64-bit Ubuntu Server is not supported, but just wanted to understand if it is possible to fix the issue below. I installed hyperpixel4 -b 64bit-beta drivers and managed to get to a point where the driver seems to be loading, the touch is working and there is video output, but there seems to be some issues with timings? Or maybe something more advanced? Thanks

[image]

config.txt

arm_64bit=1
device_tree_address=0x03000000
#dtoverlay=vc4-kms-v3d
dtoverlay=hyperpixel4
#gpio=0-25=a2
#gpio=19=op,dh
#overscan_left=0
#overscan_right=0
#overscan_top=0
#overscan_bottom=0
#framebuffer_width=800
#framebuffer_height=480
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6

alexanderbsd, 2020-08-29:
Thanks a lot for checking. Turns out it was a hardware issue. I got a replacement today and tried the 64bit-beta branch again. Everything works well - display, rotation and touch. Install.sh is broken, but it is easy to fix -> manually add lines to config.txt and install RPio and python-is-python3 packages. Just in case, decided to use hyperpixel4-init from the pi4 branch.

eaceto, 2020-08-31:
Hello @Gadgetoid It's great to see your Ubuntu booting screen. I would like to know if you can share with us which Image you used and steps you take in order to reach to that config.
Thanks very much!

Arudinne, 2020-09-02:
Hello,

I managed to get the installer script to work on a fresh install of Ubuntu Mate 64-bit 20.04.1. my changes can be found here: https://github.com/Arudinne/hyperpixel4/tree/pi4-ubuntu-mate

The following packages needed to be installed via apt-get first:

rpi.gpio
python-is-python3

The only thing that doesn't seem to work as it should is restoring the LCD's rotation after a reboot (touch screen config seems fine). I thought I managed to resolve that by adding the following line to /etc/lightdm/lightdm.conf:

display-setup-script=/usr/share/dispsetup.sh

but upon testing with a clean install the pi seems stuck in a boot loop.
---------

Apologies if any of this isn't exactly best practice, I'm not really a Linux expert, I know just enough to break stuff.

alexanderbsd, 2020-09-02:
I had the same issue when I compiled a 32bit driver (pi4 branch) for 64-bit Ubuntu Server. I used the 64bit-beta branch and replaced the content of hyperpixel4-init from the pi4 branch. Also, I made same changes as @Gadgetoid in config files (config.txt and usercfg.txt). There are still some issues (see screenshots below), but not sure whether they are software or hardware related.

Here is my setup. I only tested the 64-bit server image.

1. Ubuntu 20.04.1 LTS 64-bit installed through Raspberry Imager
2. Sudo apt-get update & upgrade to 5.4-raspi-1016 kernel
3. Install rpi.io and python-is-python3
3. Desktopify —en ubuntu
4. Hyperpixel4 -b 64bit-beta branch
5. Edit manually configs and copy the compiled driver in firmware/overlays
6. Replace the content of hyperpixel4-init from the pi4 branch (original also works, but output is a bit noisy)
7. All seems to work ok in the portrait mode, but couldn’t get the touch work correctly in other orientations. I think Ubuntu uses another default driver for Goodix. I think it can be fixed easily, but not sure what to do about the issues below:

[image]

[image]
[image]

Arudinne, 2020-09-02:
I started with the pi4-i2c-fix branch and had zero issues with the touchscreen.

eaceto, 2020-09-04:
I tried following the steps some of you did, but could see either the
ubuntu login screen or anything else...

mogenson, 2020-11-29:
Just wanted to mention that I've had a good experience with this display + Pi 400 + Ubuntu 20.10 + Phosh + Wayland.

The orientation and scale can be set through the UI or from the command line via: `wlr-randr --output DSI-1 --scale 1 --transform 90`. Touch follows the rotation correctly.

[image]

eaceto, 2020-11-29:
Great! Which branch are you using? Is Ubuntu 64bits?

mogenson, 2020-11-29:
Yep.

eaceto, 2020-11-29:
That's bloody brilliant! Would you mind share the full how-to with us?

tristansokol, 2022-02-27:
I just got this up and running after quite a bit of time and wanted to share my steps in case it could help others. 

**Goal:** Cyberdeck with Raspi400 & hyperpixel4 touch with Ubuntu Desktop (21.10)
I started with a fresh ubuntu install from the raspi imager, and below are all the steps that I think you would need to do to get a successful install. 

Some prerequisites: 
 ```bash
sudo apt-get update
sudo apt-get upgrade #this was important to get the i2c to work correctly
sudo apt get install rpi.gpio python-is-python3
```

* The advertised instructions (`curl https://get.pimoroni.com/hyperpixel4 | bash`) will not work with ubuntu, probably for lots of reasons, but least of all is that your config file is in `/boot/firmware/config.txt` instead of `/boot/config.txt`. 

* Get a copy (either clone or download zip) of the install files (for me it was [a1fcf3f7](https://github.com/pimoroni/hyperpixel4/tree/a1fcf3f7c7d35d35de81a29ae26fe5bdf827fbb4)) and update the directories for configs and overlays in the `install.sh`. Alternatively you could just run the commands, your call. The main things you need are to copy over the lines for your config file, and get the overlay built and moved to `/boot/firmware/overlays`

* trying to debug the init.sh script also let me to adding my user to the `dialout` group. `sudo adduser tristansokol dialout`. I think this needs a restart after.  You can verify your user's groups with `groups tristansokol ` (where tristansokol is obviously _my_ username) 

* Around this time I believe I had the screen partially working, the screen was blank (backlit black) but the touch functionality worked. `hyperpixel-init` would run without errors, but nothing changed with the screen. 

* The last change I made was replacing `dtoverlay=vc4-kms-v3d` with `dtoverlay=vc4-fkms-v3d` in the config file. I have no clue what that does, but I noticed the difference when trying the rotate command. 

**/boot/firmware/config.txt**
```
[pi4]
max_framebuffers=2

[all]
kernel=vmlinuz
cmdline=cmdline.txt
initramfs initrd.img followkernel

# Enable the audio output, I2C and SPI interfaces on the GPIO header
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=spi=on

# Enable the KMS ("full" KMS) graphics overlay, and allocate 128Mb to the GPU
# memory. The full KMS overlay is required for X11 application support under
# wayland
dtoverlay=vc4-fkms-v3d
gpu_mem=128

# Uncomment the following to enable the Raspberry Pi camera module firmware.
# Be warned that there *may* be incompatibilities with the "full" KMS overlay
#start_x=1

# Comment out the following line if the edges of the desktop appear outside
# the edges of your display
disable_overscan=1

# If you have issues with audio, you may try uncommenting the following line
# which forces the HDMI output into HDMI mode instead of DVI (which doesn't
# support audio output)
#hdmi_drive=2

# If you have a CM4, uncomment the following line to enable the USB2 outputs
# on the IO board (assuming your CM4 is plugged into such a board)
#dtoverlay=dwc2,dr_mode=host

# Config settings specific to arm64
arm_64bit=1
dtoverlay=dwc2

#https://github.com/pimoroni/hyperpixel4/issues/95

enable_uart=1
cmdline=cmdline.txt

enable_uart=0
dtparam=audio=on
dtparam=i2c_arm=off
dtparam=spi=off
cmdline=cmdline.txt

dtoverlay=hyperpixel4
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```
This got me to a screen that worked, but was the incorrect cyber deck rotation. I tried `arandr` and `wlr-randr` as both are mentioned above but arandr just didn't seem to work, or apply changes and the other complained about wlr-output-management-unstable-v1 not being implemented. I ended up just rotating under settings in ubuntu GUI.

Also it wasn't obvious to me reading some of the debugging tips but when you run ` ls /dev/i2c-*` you should see a device like `/dev/i2c-22`

Other issues that helped me #167  #111,

eaceto, 2022-03-18:
Wow! Congrats!!! hope this can get into the official doc!

KjeldJensen, 2022-09-13:
For what it is worth I got it to work on Ubuntu 22.04.1 64bit (including rotation) with help from Tristan above (thanks a lot) and another forum thread focusing on Raspberry Pi OS. Here is what worked for me:

It is important to update-upgrade before updating config.txt, otherwise it will be overwritten (which I learned the hard way):

sudo apt update
sudo apt upgrade

sudo adduser pi dialout 
sudo apt install rpi.gpio-common python-is-python3

sudo nano /boot/firmware/config.txt

Change these lines:

dtparam=i2c_arm=off
dtparam=spi=off

Add these lines at the end of the file:

# Hyperpixel4
dtoverlay=vc4-kms-dpi-hyperpixel4
dtparam=rotate=270,touchscreen-swapped-x-y,touchscreen-inverted-x

Reboot and it works...

d3k4-k3rb3r0s, 2024-04-23:
> For what it is worth I got it to work on Ubuntu 22.04.1 64bit (including rotation) with help from Tristan above (thanks a lot) and another forum thread focusing on Raspberry Pi OS. Here is what worked for me:
> 
> It is important to update-upgrade before updating config.txt, otherwise it will be overwritten (which I learned the hard way):
> 
> ```
> sudo apt update
> sudo apt upgrade
> ```
> 
> ```
> sudo adduser pi dialout 
> sudo apt install rpi.gpio-common python-is-python3
> ```
> 
> `sudo nano /boot/firmware/config.txt `
> 
> Change these lines:
> 
> ```
> dtparam=i2c_arm=off
> dtparam=spi=off
> ```
> 
> Add these lines at the end of the file:
> 
> ```
> # Hyperpixel4
> dtoverlay=vc4-kms-dpi-hyperpixel4
> dtparam=rotate=270,touchscreen-swapped-x-y,touchscreen-inverted-x
> ```
> 
> Save, reboot and it works...

did you get the login screen to rotate?

closed 2026-08-25 by Gadgetoid