pimoroni/hyperpixel4 #97: Unable to combine all dtoverlay parameters

tswonke, 2020-09-06:
**Describe the bug**

I'm not able to combine the commands like this. The last one seems not to be respected (already tried first -y and then -x)
 
`dtoverlay=hyperpixel4,touchscreen-swapped-x-y,touchscreen-inverted-x,touchscreen-inverted-y`

**To Reproduce**

First I used the one-line-installer, but I had troubles to rotate the touch. Then I found the parameters on https://github.com/pimoroni/hyperpixel4/tree/pi3 and was very hopeful, but I don't get it work completely.

My intend ist very simple:
Use Hyperpixel4 rectangular in normal or inverted orientation (in any case portrait mode) on an RPI 3A+ with chromium-browser but without desktop. (I have a working setup with the same screen on a Raspberry Pi Zero WH, but with desktop and and older OS).

Without modifying `dtoverlay` it will be like this:

Swipe → : Cursor ↑
Swipe ↑: Cursor →

`dtoverlay=hyperpixel4,touchscreen-swapped-x-y`:

Swipe →: Cursor ←
Swipe ↑: Cursor ↓

`dtoverlay=hyperpixel4,touchscreen-swapped-x-y,touchscreen-inverted-x,touchscreen-inverted-y`:

Swipe →: Cursor →
Swipe ↑: Cursor ↓

`dtoverlay=hyperpixel4,touchscreen-swapped-x-y,touchscreen-inverted-y,touchscreen-inverted-x`:

Swipe →: Cursor ←
Swipe ↑: Cursor ↑

I don't get all parameters running at the same time, but that is what I need... 😢
Am I missing something? I don't mind installing an older OS version if that helps...

**Your HyperPixel 4**

1. Rectangular
2. Touch
3. 2019
4. Directly installed (GPIO)

**Your Raspberry Pi**

1. Raspberry 3 A+, Revision	: 9020e0
2. Raspbian GNU/Linux 10 (buster), 5.4.51-v7+
3. Fresh setup + xinit/xserver + chromium-browser

**Extra debugging information**

[    7.537595] Goodix-TS 11-005d: 11-005d supply AVDD28 not found, using dummy regulator
[    7.537693] Goodix-TS 11-005d: 11-005d supply VDDIO not found, using dummy regulator
[    7.539584] Goodix-TS 11-005d: ID 911, version: 1060
[    7.560093] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-11/11-005d/input/input0

And check i2c is working:

`ls /dev/i2c-*`

ls: cannot access '/dev/i2c-*': No such file or directory

Is this an issue in my case?

Gadgetoid, 2020-09-07:
It's highly possible you're exceeding some line-length limit in `/boot/config.txt`. There's an 80 character limit to any one line in config.txt, and your line is 91 characters.

You should try (if I recall correctly):

```
dtoverlay=hyperpixel4
dtparam=touchscreen-swapped-x-y
dtparam=touchscreen-inverted-x
dtparam=touchscreen-inverted-y
```

tswonke, 2020-09-07:
Wow, didn't know about a limit...
This indeed helped!

Thank you very much! 👍

closed 2020-09-07 by tswonke