pimoroni/hyperpixel4 #99: Only at the half of the display touch is working - i2c problem? some failures

SebiWolze, 2020-09-15:
Description:
After installing Hyperpixel 4 Driver (with Touch support) on my Pi4 and editing my /boot/config.txt and installing a graphical interface I am able to use only one half of the touch display.

I tried everything about swapping X/Y and so on in the config.txt but nothing worked.
Touch works good, but only on one half.
I believe there is anything wrong calibrated, but not sure.

Content of my config.txt:

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2
display_lcd_rotate=3

[all]
#dtoverlay=vc4-fkms-v3d
# enable raspicam
start_x=1
gpu_mem=128

enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6

dtoverlay=hyperpixel4
dtoverlay=hyperpixel4-common
dtoverlay=hyperpixel4-0x14
dtoverlay=hyperpixel4-0x5d

What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?
Installed a freh raspbian and also a fresh octopi
Installed the driver with experimental touch
Install graphical octopi interface
Configured config.txt

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular?
Rectangular 4.0
2. Touch or non Touch?
Touch
3. Approximately when was it purchased
Before 2 days at amazon germany, direct from pimoroni
4. How is it connected to your Pi
GPIO

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
Revision        : b03111

2. Which OS release are you using: `lsb_release --description` and `uname -r`

Description:    Raspbian GNU/Linux 10 (buster)
Octopi

3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?
Fresh Octopi
Buster Lite and installed to full for graphical interface

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular

pi@octopi:~ $ dmesg | grep Goo
[    4.558226] Goodix-TS 7-005d: ID 911, version: 1060
[    4.582607] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-7/7-005d/input/input0
[    4.586019] Goodix-TS 7-0014: i2c test failed attempt 1: -6
[    4.629838] Goodix-TS 7-0014: i2c test failed attempt 2: -6
[    4.662208] Goodix-TS 7-0014: I2C communication failure: -6

And check i2c is working:

`ls /dev/i2c-*`

You should see something like:

```text
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-7
```

**pi@octopi:~ $ ls /dev/i2c-*
ls: cannot access '/dev/i2c-*': No such file or directory**

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

pi@octopi:~ $  i2cdetect -y 7
-bash: i2cdetect: command not found

**i2c fails - i dont know why!**

SebiWolze, 2020-09-15:
I removed 

dtoverlay=hyperpixel4-common
dtoverlay=hyperpixel4-0x14
dtoverlay=hyperpixel4-0x5d

But after that, touch stop working.

SebiWolze, 2020-09-15:
https://youtu.be/7asnhlxKpls

Gadgetoid, 2020-09-15:
If the same one half of the screen fails when you have rotated touch, then it's likely a hardware issue- ie: some issue with the ribbon cable. You should drop a message to support@pimoroni.com.

SebiWolze, 2020-09-23:
Okay, will do.
Thanks, close pls.

closed 2020-09-23 by SebiWolze