pimoroni/hyperpixel4 #104: How do I uninstall?

Daveyr, 2020-09-24:
There's a section in the instructions titled Install/Uninstall but I only see installation instructions. Please can you add uninstallation instructions? After installing the hyperpixel4 on a fresh Raspbian image on a Raspberry Pi 2, my scrollphat HD no longer works.

Gadgetoid, 2020-09-24:
It mostly involves removing `dtoverlay=hyperpixel4` from `/boot/config.txt`, but there are also other lines that might look something like this:

```
dtoverlay=hyperpixel4
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```
you'll need to comment these out like so:

```
#dtoverlay=hyperpixel4
#enable_dpi_lcd=1
#dpi_group=2
#dpi_mode=87
#dpi_output_format=0x7f216
#dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```

`sudo nano /boot/config.txt` should get you editing the right file.

Will flag this up to add instructions to the various READMEs.

Daveyr, 2020-09-25:
Thanks very much; that is exactly what I need. Do you think there is a need for an on/off script? If so, and you accept pull requests, I will create a bash script to do it.

closed 2026-08-25 by Gadgetoid