pimoroni/hyperpixel4 #109: 800x480 landscape mode for RetroPie

jonasschnelli, 2020-10-26:
Using the Raspberry Pi 4.

The HyperPixel4 800x480 works fine in the default portrait mode.
However, switching to 90 or 270° mode does not.
Retropie requires the **VC4 FKMS V3D** (vc4-fkms-v3d) driver to work (otherwise no OpenGLES I assume).

I tried `display_lcd_rotate=1` which seems to not work correctly.

Is there a way to rotate the screen to landscape mode without loosing the VC4 support (and without X11)?

Gadgetoid, 2020-10-26:
This is, as far as I'm aware, a hardware limitation of the Pi 4. Without some interim software layer to handle the rotation it would be down to RetroPie itself to support it.

This isn't exactly crystal clear, but seems to support my conclusion - https://www.raspberrypi.org/forums/viewtopic.php?t=266833

There is, unfortunately, nothing we can do about this on the Pi side. I will see if we can find any magic to instruct our display itself to act as 800x480 rather than 480x800.

Gadgetoid, 2021-05-20:
Did some digging and unfortunately making it 800x480 is impossible. At best I can flip X/Y to "rotate" the screen in hardware, but this just goes from one portrait orientation to another. I've made it a point to search for a landscape oriented display for a future product, but short of software (or accelerated rotation such as VC4 and the older display_lcd_rotate options) there's simply no way to make the current Hyperpixel4 behave as a landscape display.

closed 2026-08-25 by Gadgetoid