pimoroni/hyperpixel4 #110: Dark border when showing dark background

RobertSmart, 2020-10-28:
Hi,

I'm using a hyperpixel 4 touch with a RPi4 to show a fullscreen dark themed webpage. I'm seeing a dark border around the edge or a light patch in the middle (depending on how you want to look at it) has anyone else seen this? 

Its not like a solid black border, looks more like backlight bleeding through or something.

Thanks

RobertSmart, 2020-11-04:
I'm also seeing a fair bit of screen burn in, when using a dark background with white text. 

Is this normal?

ntamas, 2020-12-07:
I'm also seeing this on my Hyperpixel4. Could it be related to #79?

RobertSmart, 2020-12-07:
Is this a defect in the product or an issue in software?

Gadgetoid, 2021-05-20:
Closing stale issue. Unfortunately these issues are not monitored by tech support and are intended for software bugs only. Please contact support@pimoroni.com if you're still having problems.

closed 2021-05-20 by Gadgetoid