pimoroni/hyperpixel4 #111: `hyperpixel4-rotate` commands not working fresh install rectangular Pi4

jehanalvani, 2020-11-07:
**Describe the bug**

Unable to rotate display using `hyperpixel4-rotate` commands.

**To Reproduce**

1. Fresh install of Raspbian using Raspberry Pi imager. 
2. `touch ssh` on boot volume to enable ssh
3. Boot rPi, login
4. `sudo raspi-config`
5. Set locales [en.US]
6. `curl https://get.pimoroni.com/hyperpixel4 | bash`, option 2 (Rpi4 Rect exp touch) affirm where needed, reboot when prompted
7. Connect via SSH when rebooted (desktop in Portrait (normal) on display)
8. `hyperpixel4-rotate right` results in

```
pi@raspberrypi:~ $ hyperpixel4-rotate right
Rotating display
Can't open display 
Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 43, in __init__
    version_output = self._output("--version")
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 64, in _output
    raise Exception("XRandR returned error code %d: %s"%(status,err))
Exception: XRandR returned error code 1: Can't open display 

Setting matrix: 0 1 0 -1 0 1
Unable to connect to X server
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules

```

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Rectangular?
2. Touch
3. Oct 15th 2020
4. full 40-pin GPIO

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. Raspberry Pi Model: `Model		: Raspberry Pi 4 Model B Rev 1.1` 4gb

2.
```
pi@raspberrypi:~ $ lsb_release --description
Description:	Raspbian GNU/Linux 10 (buster)
pi@raspberrypi:~ $ uname -r
5.4.51-v7l+ 

```

3. Fresh as the morning dew

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular

```text
pi@raspberrypi:~ $ dmesg | grep Goodix
[    5.417804] Goodix-TS 11-005d: 11-005d supply AVDD28 not found, using dummy regulator
[    5.417918] Goodix-TS 11-005d: 11-005d supply VDDIO not found, using dummy regulator
[    5.418688] Goodix-TS 11-005d: i2c test failed attempt 1: -6
[    5.454211] Goodix-TS 11-005d: i2c test failed attempt 2: -6
[    5.491025] Goodix-TS 11-005d: I2C communication failure: -6
[    5.491476] Goodix-TS 11-0014: 11-0014 supply AVDD28 not found, using dummy regulator
[    5.491599] Goodix-TS 11-0014: 11-0014 supply VDDIO not found, using dummy regulator
[    5.493618] Goodix-TS 11-0014: ID 911, version: 1060
[    5.536595] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-11/11-0014/input/input4
```

And check i2c is working:

`ls /dev/i2c-*`

```text
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-11
```

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

For example:

```text
pi@raspberrypi:~ $ i2cdetect -y 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```

LordRembo, 2020-11-08:
There's a couple of issues in here already to do with screen rotation and/or touch rotation not working on Pi4, you should look around a bit. I'm having similar ones.
The gist of it is: the default drivers don't support 90° or 270° rotation on Pi4.
There's a few workarounds through the commandline. The easiest one is disabling `dtoverlay=vc4-fkms-v3d` and replacing it with `dtoverlay=hyperpixel4:rotation` + adding a line for the rotation setting ([See the comments sprinkled throughout here](https://github.com/pimoroni/hyperpixel4/issues/39)).
But the downside of it is that you lose some functionality (eg. can't run RetroPie). Which is why I'm still looking for an alternative.

jehanalvani, 2020-11-08:
Sounds like it'd be useful to me, thanks for the heads up. Devs, though, LMK if you want me to provide anything more. I can re-flash this Pi anytime.

jehanalvani, 2020-11-08:
[This comment](https://github.com/pimoroni/hyperpixel4/issues/39#issuecomment-552190708) Included the following

> /boot/config.txt:
>
>
> ```
> [pi4]
> # Enable DRM VC4 V3D driver on top of the dispmanx display stack
> #dtoverlay=vc4-fkms-v3d
> max_framebuffers=2
> 
> [all]
> #dtoverlay=vc4-fkms-v3d
>
> dtoverlay=hyperpixel4
> gpio=0-25=a2
> enable_dpi_lcd=1
> dpi_group=2
> dpi_mode=87
> dpi_output_format=0x7f216
> dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
> display_rotate=1
> ```

My `/boot/config.txt` _after_ running the installation script and choosing option 2 for rect w/ exp pi 4 fix did _not_ comment out 

`#dtoverlay=vc4-fkms-v3d` from the [pi4] section, nor did it include `display_rotate=1`  in [all]. I assume the latter was because the `hyperpixel4-rotate` command couldn't execute successfully. Manually updating the file resolved my issue for my use-case, though I suspect the bug should remain open (or closed as dupe if there's another issue that is addressing this).

David-26, 2020-11-29:
I was experiencing the same issues as @jehanalvani with latest Raspberry OS, Pi 4 and HyperPixel4. The following worked for me without using having to disable vc4-fkms-v3d @LordRembo .

**Process**
`#Download Latest HP4 Pi4 Drivers
git clone https://github.com/pimoroni/hyperpixel4 -b pi4
#Swap x y screen size in hyperpixel4-overlay
sed -i 's/screen-size-x = <480>/screen-size-x = <800>;/' ~/hyperpixel4/src/hyperpixel4-overlay.dts
sed -i 's/screen-size-y = <800>/screen-size-x = <480>;/' ~/hyperpixel4/src/hyperpixel4-overlay.dts
#Install
sudo ~./hyperpixel4/install.sh
#Reboot
sudo reboot
#Rotate the screen (in my case right)
hyperpixel4-rotate right
#Correct issue where touch is 90 degrees to screen is 'right' orientation.
echo 'ATTRS{name}=="Goodix Capacitive TouchScreen", ENV{LIBINPUT_CALIBRATION_MATRIX}="-1 0 1 0 1 0"' > /etc/udev/rules.d/98-hyperpixel4-calibration.rules`

A note, if you were to want a rotation different than 'right' then LIBINPUT_CALIBRATION_MATRIX would need to be set to something else.

**Credit Where it's Due**
[Modification of hyperpixel4-overlay.dts](https://github.com/pimoroni/hyperpixel4/issues/42#issuecomment-581158390
)
[Touch Screen Calibration](https://developer.toradex.com/knowledge-base/touch-screen-calibration-linux)

travellingkiwi, 2021-01-12:
Is this ever going to be fixed?  I've tried the process above, and no luck.

David-26, 2021-01-12:
I've had to image subsequent pi devices since this post. Found that the part done by sed needed done manually using vi. Results were inconsistent but got there eventually.

Previously when I had a Pi3 the solution that worked for me was...

[image]

LordRembo, 2021-01-12:
@David-26 We know howt works on Pi3, that's been documented extensively. It's Pi4 that's the issue. 
I never got it to work either and went back to a Pi3.

David-26, 2021-01-12:
Modifying the standard hyperpixel 4 for Pi4 install worked for me. Must have build it 6-6 times and had a few failtures. The make or break point was after modifying ~/hyperpixel4/src/hyperpixel4-overlay.dts installing and rebooting, should I be able to move the mouse to all extremes of the display then the rest worked ok.

alphex, 2021-01-16:
@David-26 
I've followed your directions from Nov 29.
Still doesn't work.

This is a brand new HyperPixel 4.0 on a brand new raspberrypi 4...

happy to help debug if I can, but any more info is appreciated.
Thank you!

alphex, 2021-01-16:
Alright, I think I found a fix.
Using @jehanalvani 's comment, and just reinstalling the software again,  I was able to get it to work.
With the USB C port on the bottom. 

I picked option 2 for the `2 : Rectangular    - Pi 4 - Experimental Touch Fix`
and
then my boot/config.txt looks like this.

```
[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
# dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

dtoverlay=hyperpixel4
gpio=0-25=a2
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
display_rotate=3

dtoverlay=hyperpixel4-common
dtoverlay=hyperpixel4-0x14
dtoverlay=hyperpixel4-0x5d
```

I hope that helps someone!

tubbybtch, 2021-01-27:
alphex: after three weeks mucking with this problem, I must say this:

THANK YOU, THANK YOU, THANK YOU, THANK YOU, THANK YOU.

There is no reason this was so incredibly difficult!

> Alright, I think I found a fix.
> Using @jehanalvani 's comment, and just reinstalling the software again, I was able to get it to work.
> With the USB C port on the bottom.
>
> I picked option 2 for the [...]

jehanalvani, 2021-01-29:
@alphex Yes, that's rad. Thank you. I still get the errors below, but the screen works in this orientation. 

`jehan@displaypi:~ $ hyperpixel4-rotate left
Rotating display
Can't open display 
[sudo] password for jehan: 
Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 43, in __init__
    version_output = self._output("--version")
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 64, in _output
    raise Exception("XRandR returned error code %d: %s"%(status,err))
Exception: XRandR returned error code 1: Can't open display 

Setting matrix: 0 -1 1 1 0 0
Unable to connect to X server
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules`

tubbybtch, 2021-01-29:
try 

export DISPLAY=:0.0 

before executing the command

> @alphex Yes, that's rad. Thank you. I still get the errors below, but the screen works in this orientation.
>
> ```
> jehan@displaypi:~ $ hyperpixel4-rotate left
> Rotating display
> Can't open display [...]

jehanalvani, 2021-01-29:
@tubbybtch 

```
jehan@displaypi:~ $ export DISPLAY=:0.0
jehan@displaypi:~ $ hyperpixel4-rotate right
Rotating display
Invalid MIT-MAGIC-COOKIE-1 keyCan't open display :0.0
[sudo] password for jehan: 
Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 43, in __init__
    version_output = self._output("--version")
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 64, in _output
    raise Exception("XRandR returned error code %d: %s"%(status,err))
Exception: XRandR returned error code 1: No protocol specified
Can't open display :0.0

Setting matrix: 0 1 0 -1 0 1
Invalid MIT-MAGIC-COOKIE-1 keyUnable to connect to X server
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules
```

No bueno, unfortunately. 3rd and 12th lines in output above.

Gadgetoid, 2021-01-29:
You may also need: `export XAUTHORITY=/home/pi/.Xauthority` or equivalent.

Sorry to sweep in here and be less than useful, but I'm between homes and all my Pi stuff is packed away in boxes. As such I've currently got no access to test this stuff and iron out any wrinkles in the documentation, but it's on my radar.

jehanalvani, 2021-01-29:
@Gadgetoid No worries; happy to be a ginuea pig if it makes resolving this easier. 

```
jehan@displaypi:~ $ export XAUTHORITY=/home/jehan/.Xauthority
jehan@displaypi:~ $ hyperpixel4-rotate right
Rotating display
Invalid MIT-MAGIC-COOKIE-1 keyCan't open display :0.0
Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 43, in __init__
    version_output = self._output("--version")
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 64, in _output
    raise Exception("XRandR returned error code %d: %s"%(status,err))
Exception: XRandR returned error code 1: Invalid MIT-MAGIC-COOKIE-1 keyCan't open display :0.0

Setting matrix: 0 1 0 -1 0 1
Invalid MIT-MAGIC-COOKIE-1 keyUnable to connect to X server
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules
```

metaColin, 2021-03-09:
*bump*

Does the Pimoroni crew have any plans to address this issue with a fix?

Would love to be able to use my Hyper Pixel in a reliable way...

Gadgetoid, 2021-03-09:
It's on my radar, but contingent on me having space to set up a Pi and familiarise myself with whatever is causing all this chaos.

cmdshft, 2021-03-29:
Just did a Pi4 install with the fix that was posted above because in the last 20 days, this has not been resolved apparently.

LordRembo, 2021-04-18:
The only fixes that seem work semi-consistently,  are the ones that involve first removing the original vc4-fkms-v3d overlay. Since that removes original functionality (eg. 3D acceleration) and the core problem is Pi4 hardware based (there's links in previous comments and related threads), I think the best course of action is to 1) document the suggested fixes in this project's install info (including a caveat about losing functionality) and 2) throw in the towel and admit this is just not fully fixable for this screen + device combo. You need to buy a different screen or use a Pi3 to use horizontal orientation is you absolutely need 3D acceleration or run things like RetroPie.

David-26, 2021-04-18:
Which reminds me, even when I did get it working with 3D acceleration, as soon as I run 'apt upgrade' it reverts back to not working correctly. Which is fine for my use case but definitely not for most.

iainp999, 2021-04-28:
Not sure if this helps anyone, but it got me past the `MIT-MAGIC_COOKIE` error, but only as root.  FWIW I still can't get the screen to rotate.

In any case, I ran `ps aux | grep -i auth` and noticed that `lightdm` was specifying an auth file

```
root       827  0.5  2.4 137480 47432 tty7     Ssl+ 16:01   0:01 /usr/lib/xorg/Xorg :0 -seat seat0 -auth /var/run/lightdm/root/:0 -nolisten tcp vt7 -novtswitch
```

I then logged in as root, and symlinked that file to become my `$HOME/.Xauthority` file

```
ln -s /var/run/lightdm/root/:0 $HOME/.Xauthority
```

Now I'm left with the following

```
root@raspberrypi4:~# DISPLAY=:0.0 hyperpixel4-rotate normal
Rotating display
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
ImportError: No module named screenlayout
Setting matrix: 1 0 0 0 1 0
unable to find device pointer:Goodix Capacitive TouchScreen
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules
```

iainp999, 2021-04-28:
To be able to connect as non-root, what I did next was a little naughty, but I'll live with it for now.

As root

```
DISPLAY=:0.0 xhost +
access control disabled, clients can connect from any host
```

Still no further forward with the rotating though.

```
DISPLAY=:0.0 hyperpixel4-rotate normal
Rotating display
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
ImportError: No module named screenlayout
Setting matrix: 1 0 0 0 1 0
unable to find device pointer:Goodix Capacitive TouchScreen
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules```

iainp999, 2021-04-28:
In the end, doing the above and running `DISPLAY=:0.0 xrandr --output DSI-1 --rotate left` worked.  Couldn't get the Pimorini script to play ball.

iainp999, 2021-04-28:
OK, finally got the script working.  Added comments at https://github.com/pimoroni/hyperpixel4/issues/51

joaolisboa, 2021-05-13:
I've been trying other methods, back to the point where I don't even have rotate_display any longer. It's like the touchscreen is still assuming a portrait orientation. dtparams touchscreen_inverted_x or y, swapped_x_y and a combinations of these and still no luck
I've also tried the usual xinput_calibrator and the 99-calibration.conf file and still no luck, when I do that the cursor just goes to the edges of the display.

joaolisboa, 2021-05-28:
Found a fix. This command properly sets the touchscreen calibration rotation for me:

xinput set-prop 'Goodix Capacitive TouchScreen' 'Coordinate Transformation Matrix' 0 -1 1 1 0 0 0 0 1

This setting isn't persisted after rebooting though. For now I put it inside a .desktop file to run on startup and it's working fine.

Gadgetoid, 2021-06-18:
> xinput set-prop 'Goodix Capacitive TouchScreen' 'Coordinate Transformation Matrix' 0 -1 1 1 0 0 0 0 1

This is... odd... what OS are you running? I'm guessing it doesn't use libinput.

joaolisboa, 2021-06-18:
> This is... odd... what OS are you running? I'm guessing it doesn't use libinput.

Latest Raspberry Pi OS.

maccampb, 2021-06-19:
this worked for me... I have to repeat on reboot, so will have to stuff the commands somewhere for boot up.
I'm ssh'ing into the pi from a terminal app on my MAC. YMMV.
-------------------------------------------------------
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Jun 18 
pi@raspberrypi:~ $ hyperpixel4-rotate left
This rotate utility only works with the Raspberry Pi OS desktop version.
You need to set a DISPLAY variable.
Try: DISPLAY=:0.0 hyperpixel4-rotate <orientation>
pi@raspberrypi:~ $ export DISPLAY=:0.0
pi@raspberrypi:~ $ hyperpixel4-rotate left
This rotate utility only works with the Raspberry Pi OS desktop version.
Rotating display
Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 43, in __init__
    version_output = self._output("--version")
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 64, in _output
    raise Exception("XRandR returned error code %d: %s"%(status,err))
Exception: XRandR returned error code 1: No protocol specified
Can't open display :0.0

Failed to set display orientation. Make sure you're running Raspberry Pi OS desktop on a Pi 4.

pi@raspberrypi:~ $ xhost +.   <------ this is the trick.
access control disabled, clients can connect from any host
pi@raspberrypi:~ $ hyperpixel4-rotate left
This rotate utility only works with the Raspberry Pi OS desktop version.
Rotating display
Saving display settings to /usr/share/dispsetup.sh
Setting matrix: 0 -1 1 1 0 0
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules
pi@raspberrypi:~ $ 
---------------------------------
hope this helps someone.

maccampb, 2021-06-19:
this worked for me... I have to repeat on reboot, so will have to stuff the commands somewhere for boot up.
I'm ssh'ing into the pi from a terminal app on my MAC. YMMV.
————————
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Jun 18
pi@raspberrypi:~ $ hyperpixel4-rotate left
This rotate utility only works with the Raspberry Pi OS desktop version.
You need to set a DISPLAY variable.
Try: DISPLAY=:0.0 hyperpixel4-rotate 
pi@raspberrypi:~ $ export DISPLAY=:0.0
pi@raspberrypi:~ $ hyperpixel4-rotate left
This rotate utility only works with the Raspberry Pi OS desktop version.
Rotating display
Traceback (most recent call last):
File "", line 15, in 
File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 43, in init
version_output = self._output("--version")
File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 64, in _output
raise Exception("XRandR returned error code %d: %s"%(status,err))
Exception: XRandR returned error code 1: No protocol specified
Can't open display :0.0
Failed to set display orientation. Make sure you're running Raspberry Pi OS desktop on a Pi 4.
pi@raspberrypi:~ $ xhost +      <------ this is the trick.
access control disabled, clients can connect from any host
pi@raspberrypi:~ $ hyperpixel4-rotate left
This rotate utility only works with the Raspberry Pi OS desktop version.
Rotating display
Saving display settings to /usr/share/dispsetup.sh
Setting matrix: 0 -1 1 1 0 0
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules
pi@raspberrypi:~ $
————
hope this helps someone.

fcatrambone, 2021-07-10:
Hi everyone!
I'm running FluiddPi, tried everything above but no luck in rotating the screen :(
Any help please? Thanks!

Gadgetoid, 2021-07-12:
> Hi everyone!
> I'm running FluiddPi, tried everything above but no luck in rotating the screen :(
> Any help please? Thanks!

Assuming Fluidd doesn't use X and you're on a Pi 4, you'd need to comment-out `dtoverlay=vc4-fkms-v3d` in `/boot/config.txt` and use `lcd_rotate`.

There's no hardware-level accelerated rotation on Pi 4/Pi 400 so you must either turn `vc4-fkms-v3d` off or support rotation in the application itself.

If it *does* use X, these instructions might help - https://github.com/pimoroni/hyperpixel4/commit/cf175ccc22017417c3d8be24a26afa6433705abd

morphias2004, 2021-10-02:
It's reprehensible that this issue is still causing trouble all over the place.

Repeated commitments from @Gadgetoid over the last 18 months to address it have gone nowhere. It's an utter disgrace that Pimoroni continue to sell and market this screen and tout it's compatability with the Pi4, yet we are no better off and have to implement hack workarounds.

As least get a simple to follow set of instructions in the README to stop all this BS.

I have literally just loaded up the latest OctoPi 0.18 image from scratch on my Pi4 4GB and have it working with the latest OctoDash.

I loaded the latest OctoPi image, did the basic config on OctoPrint to get it on the network, ran apt update and upgrade, , ran all the updates on OctoPrint to bring it up to v1.6.1 and then ran:

`sudo apt install git`

`git clone https://github.com/pimoroni/hyperpixel4 -b pi4`

`sudo raspi-config`

System Options > Boot/Auto Login    and enabled B1 Console and B2 Console Autologin

Rebooted

`cd /boot`

`sudo nano config.txt`

Edit the bottom so it looks like this:

```
[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
**#dtoverlay=vc4-fkms-v3d**
max_framebuffers=2
**display_lcd_rotate=3**

[all]
#dtoverlay=vc4-fkms-v3d
# enable raspicam
start_x=1
gpu_mem=128

dtoverlay=hyperpixel4
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```

Save and reboot and the screen will now be orientated with the USB C port at the bottom. If you want the port at the top, then you need to set **display_lcd_rotate=1**

The TOUCH function will still be incorrect, but to correct this:

`cd /usr/share/X11/xorg.conf.d`

`sudo nano 40-libinput.conf`

Edit the section for the touchscreen - it is normally the second last section - by adding this line:

```
Section "InputClass"
        Identifier "libinput touchscreen catchall"
        MatchIsTouchscreen "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
        **Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1"**
EndSection
```

Leave all the spaces.

Save and reboot and your touch screen should now be orientated correctly the and touch screen also calibrated correctly. If you went with the USB C port at the top, then you need to make the line **Option "TransformationMatrix" "0 1 0 -1 0 1 0 0 1"**

This is so simple and straight forward. Surely Pimoroni could script this to take the leg work out of it? At least pull your fingers out and document it.

I do not claim credit for this solution. It's a combination of the following video and articles:

https://github.com/TxBillbr/octodash-hyperpixel-fix

https://www.youtube.com/watch?v=K0A-sIUBFfU&t=2015s

https://www.instructables.com/Rotate-Raspberry-Pi-Display-and-Touchscreen/

morphias2004, 2021-10-09:
> THANK YOU. Just followed your instructions and it worked the first time.
> 
> I've been using a smaller and easier to install screen for the last few months after not being able to find a way to get my Hyperpixel 4 to work with my Pi 4.

Glad I could help mate!  :)

I can't tell you how many days I've burned trawling the internet for solutions and have been posting my procedure everywhere to try and help the countless thousands that are in the same boat.

Ben.

Gadgetoid, 2021-10-12:
> pi@raspberrypi:~ $ xhost + <------ this is the trick.

@maccampb this is useful, thank you. I've added some xhost auth disabling/enabling magic into the hyperpixel4-rotate scripts for the two rectangular branches and square. Along with the save-to-Xorg feature that should cover more bases (for non-Pi OS platforms that don't use the `/usr/share/dispsetup.sh` kludge.

See:
* https://github.com/pimoroni/hyperpixel4/pull/145
* https://github.com/pimoroni/hyperpixel4/pull/146
* https://github.com/pimoroni/hyperpixel4/pull/144

There was also a bugfix in my Xorg-rotate patches that puts the "xrandr" command *before* the invocation of "arandr" or "Screen Layout" which actually persists it. That made rotation of both touch/display in-sync require *two* successive runs of the script.

I've tested this on RPi OS (buster). As always YMMV on other distributions.

This should, with any luck, fix @jehanalvani's original issue.

bchau21, 2021-10-15:
> > Shut down the PI4 last night for something and booted it up earlier this evening and now my touch isn't working.
> 
> @bchau21 have you tried the i2c-touch fix version of the drivers?

Just going to preface this by saying i'm still relatively new to the Raspberry Pi system overall. Got my first one a little over a year ago and I'm really only familiar with OctoPi, OctoPrint, and OctoDash. 

I see your post above mine mentioned a fix, how would I go about trying that out?

bchau21, 2021-10-15:
> > > > THANK YOU. Just followed your instructions and it worked the first time.
> > > > I've been using a smaller and easier to install screen for the last few months after not being able to find a way to get my Hyperpixel 4 to work with my Pi 4.
> > > 
> > > 
> > > Glad I could help mate! :)
> > > I can't tell you how many days I've burned trawling the internet for solutions and have been posting my procedure everywhere to try and help the countless thousands that are in the same boat.
> > > Ben.
> > 
> > 
> > Any issues with yours recently?
> > It's been working with no issues after several reboots and shutdowns.
> > Shut down the PI4 last night for something and booted it up earlier this evening and now my touch isn't working. Checked all files to make sure everything was still there and they are.
> > Edit: I see something was updated and i'm guessing that broke it. Seriously about to throw this thing into the trash.
> 
> That's one of the major bug bears with this thing. Once you get it working, any Pi OS updates will break it.
> 
> Updates to apps on the Pi doesn't seems to break it - OctoPrint, OctoDash, OctoPrint plug-ins., etc. Just the actual updates that get installed from an **apt update**
> 
> I would try re-following my steps, less the **apt update** and **apt upgrade** from the start.

I did try your steps, minus the apt update and apt upgrade parts, after a friend (who's a little more familiar with the Pi) recommended it to see if something wasn't saved. Sadly, it didn't do anything, and I tried it twice. All the text I added previously were still there in both conf files.

morphias2004, 2021-10-20:
> I did try your steps, minus the apt update and apt upgrade parts, after a friend (who's a little more familiar with the Pi) recommended it to see if something wasn't saved. Sadly, it didn't do [...]

I have had the same experience in the past.

Sadly, it means a full reinstall of OctoPi.

Back up your OctoPi config with the OctoPrint backup utility - accessed from settings. It will back up everything including your plugins.

Reinstall OctoPi 0.18, do you apt updates, do you base setup of OctoPrint, restore your backup, reinstall OctoDash and then followmy HyperPixel install guide. I've updated my instructions as I left out the lines to actually install the HyperPixel drivers and reboot.

Never do any Raspberry apt updates again - they break the HyperPixel drivers. What specifically, I do not know, but I know it's a deal breaker.

OctoPrint, OctoPrint plugin and OctoDash updates will not break anything.

GruntyAssassin, 2021-11-08:
> 
> 
> It's reprehensible that this issue is still causing trouble all over the place.
> 
> Repeated commitments from @Gadgetoid over the last 18 months to address it have gone nowhere. It's an utter disgrace that Pimoroni continue to sell and market this screen and tout it's compatability with the Pi4, yet we are no better off and have to implement hack workarounds.
> 
> As least get a simple to follow set of instructions in the README to stop all this BS.
> 
> I have literally just loaded up the latest OctoPi 0.18 image from scratch on my Pi4 4GB and have it working with the latest OctoDash.
> 
> I loaded the latest OctoPi image, did the basic config on OctoPrint to get it on the network, ran apt update and upgrade, , ran all the updates on OctoPrint to bring it up to v1.6.1 and then ran:
> 
> `sudo apt install git`
> 
> `git clone https://github.com/pimoroni/hyperpixel4 -b pi4`
> 
> `cd hyperpixel4`
> 
> `sudo ./install.sh`
> 
> `sudo reboot now`
> 
> Log back in and then run
> 
> `sudo raspi-config`
> 
> System Options > Boot/Auto Login and enabled B1 Console and B2 Console Autologin
> 
> Rebooted
> 
> `cd /boot`
> 
> `sudo nano config.txt`
> 
> Edit the bottom so it looks like this:
> 
> ```
> [pi4]
> # Enable DRM VC4 V3D driver on top of the dispmanx display stack
> **#dtoverlay=vc4-fkms-v3d**
> max_framebuffers=2
> **display_lcd_rotate=3**
> 
> [all]
> #dtoverlay=vc4-fkms-v3d
> # enable raspicam
> start_x=1
> gpu_mem=128
> 
> dtoverlay=hyperpixel4
> enable_dpi_lcd=1
> dpi_group=2
> dpi_mode=87
> dpi_output_format=0x7f216
> dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
> ```
> 
> Save and reboot and the screen will now be orientated with the USB C port at the bottom. If you want the port at the top, then you need to set **display_lcd_rotate=1**
> 
> The TOUCH function will still be incorrect, but to correct this:
> 
> `cd /usr/share/X11/xorg.conf.d`
> 
> `sudo nano 40-libinput.conf`
> 
> Edit the section for the touchscreen - it is normally the second last section - by adding this line:
> 
> ```
> Section "InputClass"
>         Identifier "libinput touchscreen catchall"
>         MatchIsTouchscreen "on"
>         MatchDevicePath "/dev/input/event*"
>         Driver "libinput"
>         **Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1"**
> EndSection
> ```
> 
> Leave all the spaces.
> 
> Save and reboot and your touch screen should now be orientated correctly the and touch screen also calibrated correctly. If you went with the USB C port at the top, then you need to make the line **Option "TransformationMatrix" "0 1 0 -1 0 1 0 0 1"**
> 
> This is so simple and straight forward. Surely Pimoroni could script this to take the leg work out of it? At least pull your fingers out and document it.
> 
> I do not claim credit for this solution. It's a combination of the following video and articles:
> 
> https://github.com/TxBillbr/octodash-hyperpixel-fix
> 
> https://www.youtube.com/watch?v=K0A-sIUBFfU&t=2015s
> 
> https://www.instructables.com/Rotate-Raspberry-Pi-Display-and-Touchscreen/

Holy shit, I wish I could shake your hand. I spent the last few days struggling to finish my Prusa OctoDash side project. 90% of it was getting the Hyperpixel to work with my RPi 4 and I was minutes away from just doing an Amazon return for the whole project until I stumbled on this comment that fixed everything.

NOMAD1172, 2022-01-13:
> Allow me to also join the party of people who wasted and unnecessary amount of time getting this to work on a fluiddpi system. Thank you @morphias2004! You saved the day!

What was the magic sauce? I followed the steps exactly but still result in a rotated display but no touch functionality. (KlipperScreen, Fluidpi all updated to latest versions as of a few hours ago, and raspi 4)

closed 2026-08-25 by Gadgetoid