pimoroni/hyperpixel4 #113: Touch is not working on raspbian 64bit

excelsior091224, 2020-11-24:
**Describe the bug**

What's gone wrong?

Touch is not working.

**To Reproduce**

What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?

patch-pi4-rotate-error-handling from GitHub.

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular?
Rectangular
2. Touch or non Touch?
Touch
3. Approximately when was it purchased
2018/08
4. How is it connected to your Pi
Connected with GPIO directly

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
```Revision        : c03112```
2. Which OS release are you using: `lsb_release --description` and `uname -r`
```Description:    Raspbian GNU/Linux 10 (buster)
5.4.72-v8+```
3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?
fresh OS setup

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular
* `dmesg | grep ft5` for HyperPixel 4.0" Square

```[    4.962222] Goodix-TS 11-0014: 11-0014 supply AVDD28 not found, using dummy regulator
[    4.962573] Goodix-TS 11-0014: 11-0014 supply VDDIO not found, using dummy regulator
[    4.963389] Goodix-TS 11-0014: i2c test failed attempt 1: -6
[    4.991531] Goodix-TS 11-0014: i2c test failed attempt 2: -6
[    5.018683] Goodix-TS 11-0014: I2C communication failure: -6```

And check i2c is working:

`ls /dev/i2c-*`

You should see something like:

```text
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-7
```

```pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-11```

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

For example:

```text
pi@raspberrypi:~ $ i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

```pi@raspberrypi:~ $ i2cdetect -y 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- 5d -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --```

```pi@raspberrypi:~ $ curl -sSL https://raw.githubusercontent.com/pimoroni/hyperpixel4/master/hyperpixel4-debug.sh | bash

## Platform Information

Revision        : c03112
Description:    Raspbian GNU/Linux 10 (buster)
5.4.72-v8+

## Touchscreen logs

Rectangular: Goodix
[    4.962222] Goodix-TS 11-0014: 11-0014 supply AVDD28 not found, using dummy regulator
[    4.962573] Goodix-TS 11-0014: 11-0014 supply VDDIO not found, using dummy regulator
[    4.963389] Goodix-TS 11-0014: i2c test failed attempt 1: -6
[    4.991531] Goodix-TS 11-0014: i2c test failed attempt 2: -6
[    5.018683] Goodix-TS 11-0014: I2C communication failure: -6
Square: ft5

## I2C Devices and Mappings

/dev/i2c-11
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory

## Boot Config

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

#
arm_64bit=1

dtoverlay=hyperpixel4
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6```

Gadgetoid, 2020-11-24:
I think I know what's up. You should try the pi4-i2c-fix version of the Hyperpixel4 software:

```
git clone https://github.com/pimoroni/hyperpixel4 -b pi4-i2c-fix
cd hyperpixel4
sudo ./install.sh
```

And make sure `dtoverlay=hyperpixel4` is commented-out in `/boot/config.txt` since this fix uses three different overlays to try and catch the touch IC on the right address.

The what and why of this glorious race-condition bug is explained here - https://github.com/pimoroni/hyperpixel4/issues/41#issuecomment-549759417

excelsior091224, 2020-11-26:
> I think I know what's up. You should try the pi4-i2c-fix version of the Hyperpixel4 software:
> 
> ```
> git clone https://github.com/pimoroni/hyperpixel4 -b pi4-i2c-fix
> cd hyperpixel4
> sudo ./install.sh
> ```
> 
> And make sure `dtoverlay=hyperpixel4` is commented-out in `/boot/config.txt` since this fix uses three different overlays to try and catch the touch IC on the right address.
> 
> The what and why of this glorious race-condition bug is explained here - [#41 (comment)](https://github.com/pimoroni/hyperpixel4/issues/41#issuecomment-549759417)

It works. Thank you!

Gadgetoid, 2020-11-26:
Whew! Great. I really should make it the standard for Pi 4 / HyperPixel 4 to avoid this but it's such a weird hack for an even weirder issue 😬

closed 2020-11-26 by Gadgetoid