pimoroni/hyperpixel4 #117: HyperPixel 4 Square RPi 4 64bit support

ncohen17, 2020-12-27:
Hi, 

I am trying to install the hyperpixel 4 square driver on the Pi 4 64bit Raspbian OS. I have tried using the 64bit-beta branch but it doesn't seem to work. when installing the driver manually and connecting the display i get inverted colours and touch is not working.

could you please advise of a solution ?

arronmabrey, 2021-02-24:
@ncohen17 I have the same issue on the 64-bit Raspberry Pi 4. I tried installing the "pi4-64bit-beta" found here https://github.com/pimoroni/hyperpixel4/tree/pi4-64bit-beta and I got these results 
[image]

ncohen17, 2021-02-24:
@arronmabrey I got the same results as you got. For some reason the 64bit is not working for me using the Square display on the Pi 4.

ncohen17, 2021-05-13:
Hi @Gadgetoid do you know why this is happening ? any fixes to this issue ?

Gadgetoid, 2022-04-12:
This should be wholly fixed in the latest Raspberry Pi OS release. Some details here: https://github.com/pimoroni/hyperpixel4/issues/177

closed 2022-04-12 by Gadgetoid