pimoroni/hyperpixel4 #118: Pi4 fresh install won't rotate

rtho782, 2020-12-30:
**Describe the bug**

Cannot rotate screen

pi@raspberrypi:~ $ sudo hyperpixel4-rotate left
Rotating display
Can't open display
Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 43, in __                         init__
    version_output = self._output("--version")
  File "/usr/lib/python2.7/dist-packages/screenlayout/xrandr.py", line 64, in _o                         utput
    raise Exception("XRandR returned error code %d: %s"%(status,err))
Exception: XRandR returned error code 1: Can't open display

Setting matrix: 0 -1 1 1 0 0
Unable to connect to X server
Saving touch settings to /etc/udev/rules.d/98-hyperpixel4-calibration.rules

**To Reproduce**

Fresh install of RaspberryOS, on Pi4 8gb, run the following commands:
sudo apt-get update
sudo apt-get upgrade
curl https://get.pimoroni.com/hyperpixel4 | bash

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular?
  Rectangular
2. Touch or non Touch?
  Touch
3. Approximately when was it purchased
  Nov 2019
4. How is it connected to your Pi
  Sat on the GPIO header

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
  Revision        : d03114
2. Which OS release are you using: `lsb_release --description` and `uname -r`
  Description:    Raspbian GNU/Linux 10 (buster)
  5.4.79-v7l+
3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?
  Fresh OS

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular

[    7.224135] Goodix-TS 11-0014: 11-0014 supply AVDD28 not found, using dummy regulator
[    7.224265] Goodix-TS 11-0014: 11-0014 supply VDDIO not found, using dummy regulator
[    7.226246] Goodix-TS 11-0014: ID 911, version: 1060
[    7.248702] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-11/11-0014/input/input0

* `dmesg | grep ft5` for HyperPixel 4.0" Square

And check i2c is working:

`ls /dev/i2c-*`
/dev/i2c-11

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

## Platform Information

Revision        : d03114
Description:    Raspbian GNU/Linux 10 (buster)
5.4.79-v7l+

## Touchscreen logs

Rectangular: Goodix
[    7.224135] Goodix-TS 11-0014: 11-0014 supply AVDD28 not found, using dummy r                         egulator
[    7.224265] Goodix-TS 11-0014: 11-0014 supply VDDIO not found, using dummy re                         gulator
[    7.226246] Goodix-TS 11-0014: ID 911, version: 1060
[    7.248702] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i                         2c-11/11-0014/input/input0
Square: ft5

## I2C Devices and Mappings

/dev/i2c-11
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directo                         ry

## Boot Config

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

dtoverlay=hyperpixel4
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6

travellingkiwi, 2021-01-12:
That looks like you didn't set the DISPLAY env

export DISPLAY=:0.0

before you try to rotate... (It may still not work, but at least you'll be able to talk to the display)

Gadgetoid, 2021-05-20:
I'm in the process of updating the rotate script to complain more when it's run in the wrong context. Should hopefully curb some of these :grimacing:

closed 2021-05-20 by Gadgetoid

Shyamranny, 2022-12-31:
Please see [RPiconfig - eLinux.org](http://elinux.org/RPi_config.txt#Video_mode_options)
I added display_rotate=3 in /boot/config.txt and rebooted. Then my display got rotated. But touch screen was not working properly. Then I ran "hyperpixel4-rotate left" which corrected the touch screen issue as well :slight_smile: