pimoroni/hyperpixel4 #119: GPIO Signal Clarification

DonaldSchelle, 2020-12-31:
Hello,

I need a few clarifications regarding the signals on the GPIO pin header.

The standard questions are answered as so:

1. Is it Square or Rectangular?  - Rectangular
2. Touch or non Touch? - Touch
3. Approximately when was it purchased - Last Week
4. How is it connected to your Pi - GPIO

I've looked at the pinout here: https://pinout.xyz/pinout/hyperpixel4

A few questions:

Regarding the SPI interface.   

1.   Is it safe to assume that the pins map to standard SPI signals as follows:

- GPIO27 (pin 13) - SCLK
- GPIO26 (pin 37) - MOSI
- GPIO18 (pin 12) - CS#

2. GPIO27 is labeled as "Touch Interrupt" on the pinout page.   Is this a typo, or is it intended as an ALERT pin on the I2C bus?   This post (https://github.com/pimoroni/hyperpixel4/issues/2) seems to indicate that GPIO27 is part of the SPI bus used to initialize the display.

3. What's the maximum SCLK speed of the SPI interface?

4. What's the I/O voltage on the I2C and SPI bus?  (3.3V?)

Regarding the I2C interface:

5. Please confirm the signal names:

- GPIO11 (pin 23) - SCL
- GPIO10 (pin 19) - SDA

6. With regard to backlight control (GPIO 19, pin 35).   Is this a PWM signal?   If so, what's the I/O voltage and more importantly, the PWM frequency?

I'm in the process of trying to serialize the GPIO data in this application so that the display can be remote located (a ribbon cable doesn't work) using a standard cable (I'm thinking HDMI) in a proprietary manner.   The RGB666 interface is fairly straight forward.    Documentation on the other interfaces is a little light though.    I'm hoping someone here can save me a little time; otherwise, I'll have to reverse engineer the signals with an oscilloscope.

Don

sjtoik, 2021-01-03:
Found this, hope it helps: https://pinout.xyz/pinout/hyperpixel4

Please document your findings, if you end up making measurements.

DonaldSchelle, 2021-01-03:
Appreciate the follow-up.   I've since found an HDMI display that I think will work just as well for my application.   Same size, IPS display, capacitive touch via USB, and cheaper than the HyperPixel.   I'll report back once I confirm that it works and if the display looks as good as the HyperPixel.

Gadgetoid, 2021-05-20:
FWIW (and I know I'm a day late and a dollar short here) those are - indeed - I2C and SPI busses. All logic levels on the Pi are necessarily 3.3v, this applies also to PWM. GPIO 19 has PWM channel 1 as ALT mode 5 and you can.. basically configure this within the limits of the Pi's GPIO. Frequency can be basically whatever you like, since it's just the duty cycle that matters. (Well some would argue low frequency backlight PWM tortures their eyes).

The touch interrupt pin is- indeed- shared with the SPI bus. SPI is actually bitbanged on pins 27, 26, 18 (CLK, MOSI, CS), the full init is here: https://github.com/pimoroni/hyperpixel4/blob/pi4/dist/hyperpixel4-init

i2c is also bitbanged with the i2c-gpio module. The limitations of the Pi's DPI output means everything is a bit wonky.

It sounds like an HDMI display would be best in your application, though.

closed 2026-08-25 by Gadgetoid