pimoroni/hyperpixel4 #121: VLC displays incorrectly when Hyperpixel is rotated

orbrey, 2021-01-19:
Using the hyperpixel display in landscape mode and VLC Viewer to display a 704 x 480 mpeg file, VLC cuts off the display at 480 pixels. So it seems like it is picking up the 480 as the width of the display not the height. If I put the hyperpixel in portrait mode VLC then displays the full width resolution of the video, but I have to move the window back and forth to view it all.

Hyperpixel was installed using the one line installer.
Removing either of the display_rotate =1
statements in config.txt has no effect.

If I take out the config.txt settings for the Hyperpixel 4.0 screen and
remove the screen then VLC displays just fine so I really don't
understand those videolan tickets since they don't specifically 

This is the rectangular Hyperpixel 4, unsure if it's touch or non-touch, but hyperpixel-rotate isn't being used so non-touch is presumed. It's installed on Raspbian, unsure of version, will ask for details.

tahoemike, 2021-01-20:
This was a Hyperpixel 4.0 touch screen.  It was running on Raspbian Buster.  I attach an image showing a 720x480 sized video being displayed with VLC Viewer. 
[image]

Gadgetoid, 2021-05-20:
This is a Raspberry Pi/VLC issue and honestly I don't have the first clue what's causing it, much less who to poke for a fix.

closed 2021-05-20 by Gadgetoid