pimoroni/hyperpixel4 #122: No display with Pi4 and Hyperpixel Touch Screen Help Please !!

pip000, 2021-01-19:
**Describe the bug**I have a big problem with a hyperpixel 4 screen and a new Pi4 , used pimoroni bash scrip for rectangulatr screen but not getting display to work . Your help is very much appreciated as I have hit a wall now after a few hours trying to figure out why  :-(   please see screen report below :

[all]
#dtoverlay=vc4-fkms-v3d

# NOOBS Auto-generated Settings:
start_x=1
gpu_mem=128

dtoverlay=hyperpixel4
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6

dtoverlay=hyperpixel4-common
dtoverlay=hyperpixel4-0x14
dtoverlay=hyperpixel4-0x5d

overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
display_default_lcd=1
display_rotate=0
hdmi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
pi@Raspberry:~ $ 

What's gone wrong?

**To Reproduce**

What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular?
2. Touch or non Touch?
3. Approximately when was it purchased
4. How is it connected to your Pi

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
2. Which OS release are you using: `lsb_release --description` and `uname -r`
3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular
* `dmesg | grep ft5` for HyperPixel 4.0" Square

And check i2c is working:

`ls /dev/i2c-*`

You should see something like:

```text
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-7
```

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

For example:

```text
pi@raspberrypi:~ $ i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Kungpaoshizi, 2021-05-28:
Are you on the Pi 400 or a Pi 4 board itself?
Perhaps we're having similar issues :)

https://forums.adafruit.com/viewtopic.php?f=50&t=179733

pip000, 2021-06-03:
I am using a Raspberry Pi 4 Board :-)  Thanks for your help .

closed 2026-08-25 by Gadgetoid