pimoroni/hyperpixel4 #123: Hyperpixel 4 touchscreen not responding to touch

spw1502, 2021-01-22:
I have fitted a Hyperpixel 4 touch rectangle on Pi4 and installed software using the one line install script. Display works brilliantly except no touch response.
Re-installed software from github clone.
Still no touch operation.
This seems to be a perennial problem.
Have trawled around and can't find anything obvious else to try.
Am I missing something, this seems to have been an ongoing issue for at least a couple of years, surely it shouldn't be this difficult.

tubbybtch, 2021-01-27:
Having the same issue...lack of response *troubling*

Choirin, 2021-01-27:
I'm also having the same problem with a recently acquired one.
After some debugging, I found a resistor might be missing on my device.
(Pad on the right side, below the connector in the picture)
This resistor connects INT signal to raspberry pi.
I mount zero ohm resistor, and the touch function works fine.
[image]

After the fix, I searched in google and found a picture which shows the resistor (4.7 kOhm) mounted.
https://deninet.com/blog/2019/06/30/handheld-raspberry-pi-part-1-if-first-you-dont-succeed

I'm guessing this issue may happen only on my device, but just in case, I'd like to share.

Gadgetoid, 2021-05-20:
For Pi 4 touch on older Hyperpixel 4 units there's an `i2c-touch-fix` option in the installer that may work.

If all else fails, you should contact support@pimoroni.com.

closed 2021-05-20 by Gadgetoid