pimoroni/hyperpixel4 #124: Touch not working on Pi 3B+

dev-null2019, 2021-02-02:
**Describe the bug**
After installation with `curl -sSL https://get.pimoroni.com/hyperpixel4 | bash` the display itself works fine, but touch input is not working. The I2C from the Goodix chip isnt recognized. And the Pi did not recognize any other I2C devices from the outbreak on the display.
I did detach and reattach the display several times, measured every PIN from the Pi to the Hyperpixel, every PIN has good contact.

What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?
It tried both!

**Your HyperPixel 4**
1. Rectangular
2. Touch
3. purchased about two weeks ago
4. It is connected over the PIN header

**Your Raspberry Pi**
Raspberry Pi 3B+ with Raspbian GNU/Linux 10 (buster) and Kernel 5.4.83-v7+ fresh installed.

**Extra debugging information**
```
pi@raspberrypi:~ $ dmesg | grep Goodix
[    8.781951] Goodix-TS 11-005d: 11-005d supply AVDD28 not found, using dummy regulator
[    8.782088] Goodix-TS 11-005d: 11-005d supply VDDIO not found, using dummy regulator
[    8.782974] Goodix-TS 11-005d: i2c test failed attempt 1: -6
[    8.817055] Goodix-TS 11-005d: i2c test failed attempt 2: -6
[    8.856384] Goodix-TS 11-005d: I2C communication failure: -6
```
```

pi@raspberrypi:~ $ dmesg | grep i2c
[    4.963532] i2c /dev entries driver
[    6.660761] gpio-10 (i2c@0): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[    6.664013] gpio-11 (i2c@0): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[    6.670357] i2c-gpio i2c@0: using lines 10 (SDA) and 11 (SCL)
[    8.782974] Goodix-TS 11-005d: i2c test failed attempt 1: -6
[    8.817055] Goodix-TS 11-005d: i2c test failed attempt 2: -6
```

And check i2c is working:
```
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-11
```

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

```
pi@raspberrypi:~ $ i2cdetect -y 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

dev-null2019, 2021-02-02:
Ok, it seems a logic level shifter put 5V on the GPIO10 ... there is no sign of life on this GPIO anymore. I order a new PI and report back if that was the issue.

dev-null2019, 2021-02-03:
Update: Two GPIO Pins on the Pi are dead, with a new one the display works great!

Thanks.

closed 2021-02-03 by dev-null2019

dalilox, 2022-09-22:
Hello,

I know it's very old, but I do have the same issue as yours, is there anything should be changed in touchscreen's device tree configuration ? I've added some nodes and properties according to goodix bindings but the same problem as yours with same logs. 

[    1.348528] Goodix-TS 2-005d: i2c test failed attempt 1: -6
[    1.384533] Goodix-TS 2-005d: i2c test failed attempt 2: -6
[    1.415939] Goodix-TS 2-005d: I2C communication failure: -6

Let me know please, Thanks

dev-null2019, 2022-09-22:
Do you have another PI to test the display?
Its likely a dead GPIO pin on the Pi or a faulty touchscreen controller...

dalilox, 2022-09-22:
No unfortunately I don’t have. Yes maybe it’s the touch pin controller. Can you share with me your device tree modifications ?