pimoroni/hyperpixel4 #127: Incorrectly giving me the message This hardware is not supported, sorry! on the raspberry pi 4

coreybruce, 2021-02-06:
**Describe the bug**
Incorrectly giving me the message This hardware is not supported, sorry! on the raspberry pi 4 using Ubuntu Mate
What's gone wrong?

**To Reproduce**
Run script on any other distro other than Raspberry pi OS on the Raspberry pi 4 and you will get this incorrect message 

What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?

**Your HyperPixel 4**
```curl -sSL https://get.pimoroni.com/hyperpixel4 | bash```

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular?
2. Touch or non Touch?
3. Approximately when was it purchased
4. How is it connected to your Pi
Rectangular

**Your Raspberry Pi**
Raspberry pi 4

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
Revision	: d03114
2. Which OS release are you using: `lsb_release --description` and `uname -r`
Ubuntu 20.04.1 LTS
3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?
Raspbery pi 4 8gb

Can't get past the installer to give extra debugging info for driver as it's not installed.

dev-null2019, 2021-02-15:
Hi!
I think (without having a look at the one-line installer) that there is some kind of OS check inside the script. Have you tried to install the Pi4 drivers, without the on-line installer?

https://github.com/pimoroni/hyperpixel4/tree/pi4

Regards
Stefan

coreybruce, 2021-02-15:
Yeah I noticed which is a bit silly as this should be a universal install for all Linux Arm systems and yes I have with no success, I also made a issue report of it having issues getting it to even work properly and even on Raspberry pi OS I noticed that the touch wouldn't work properly and the DPI needs adjusting as it's all to small.

Gadgetoid, 2021-05-20:
Sorry we're unable to support any platform other than Raspberry Pi OS desktop. There's simply too little time and too many variables... and I'm drowning in Pi OS issues at the moment.

closed 2021-05-20 by Gadgetoid

coreybruce, 2021-05-20:
Once all the issues are fixed would you? because even on raspberry pi OS I ran into with touch and scaling but mainly touch being very buggy.