pimoroni/hyperpixel4 #128: Is there a method to install on omnirom?

EagleDTW, 2021-03-16:
Is there a method to install for android 11 Rpi 4B omnirom? If there’s a manual method or some install scripts that are android compatible I would really appreciate the support? Thank you, Daniel

eaceto, 2021-03-24:
Hello Daniel, I'm also looking for this. Did you make any progress?

EagleDTW, 2021-03-24:
Hi Ezequiel, unfortunately no response yet and no progress on integration of the hyperpixel.

eaceto, 2021-03-25:
I could make it work with Ubuntu OS64bit and Flutter 2.0.3... it even runs some Flutter apps without problems...!

[Uploading 20210324_232129.jpg …]()

Gadgetoid, 2021-05-20:
I'm afraid I don't have the first clue how to support Android, nor the time to look into it. If anyone manages to get it working, I'd be happy to link some instructions here, but unfortunately that's about the best I can do.

@eaceto I'd be interested in step-by-step instructions for getting things running on Ubuntu, I've had requests for support but - again - there aren't enough hours in the day! - see: https://github.com/pimoroni/hyperpixel4/issues/130

eaceto, 2021-09-13:
Hello, I could finish the tutorial I have started related to this display and running Flutter ( as I have recently moved to Europe). I am going to return to this in a few days when my RPi 4 with 8 GB arrives.

Gadgetoid, 2025-01-07:
Closed as effectively wontfix. Though the state of things has probably changed significantly in the intervening years.

closed 2025-01-07 by Gadgetoid