pimoroni/hyperpixel4 #131: dtoverlay settings in /boot/config.txt sporadically applied, cause and solution?

fhb, 2021-05-16:
**Bug description and reproduction steps**

I'm trying to rotate the display and the touch-overlay from portrait to landscape. 

I tried both the one-line-installation for the rpi4 with and without "experimental touch fix". 

The command "hyperpixel4-rotate left" led to a correct display rotation with inverted touch-overlay and also the rotation did not persist after reboot, so I rotated back with hyperpixel4-rotate normal. 

I then edited /boot/config.txt
out-commenting:
#dtoverlay=vc4-fkms-v3d
and adding
display_lcd_rotate=3
and then
dtparam=touchscreen-swapped-x-y=1
dtparam=touchscreen-inverted-x=1
after
dtoverlay=hyperpixel4
or instead writing
dtoverlay=hyperpixel4, touchscreen-swapped-x-y=1, touchscreen-inverted-x=1

This led to a temporary solution with both screens and touch-overlay being rotated properly. But after a reboot of the RPI I'm stuck with the touchscreen x-axis being inverted and my settings in /config.txt regarding the dtoverlay don't seem to have an effect. Reading through the documentation and the documented issues I found that the touchscreen driver tries to connect through i2c to the addresses 0x5d and 0x14, the latter failing as seen below. 

Is the "experimental touch fix" offered by the installer what's been described as trying to reach both display addresses in https://github.com/pimoroni/hyperpixel4/issues/41?

Is the i2c error documented below expected behaviour due to trying both addresses in the "experimental touch fix" version?

Could the fact, that the dtoverlay settings in /boot/config.txt are seemingly randomly applied or ignored, be related to the temperature dependent address switch? 

**Your HyperPixel 4**

Rectangular, Touch, purchased a week ago, connected via gpio

**Your Raspberry Pi**
RPI4 Revision	: b03114 with OctoPi, Raspbian GNU/Linux 10 (buster) 5.10.17-v7l+

**Extra debugging information**

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular:

[    5.959145] Goodix-TS 11-005d: ID 911, version: 1060
[    5.983433] Goodix-TS 11-005d: Checking firmware
[    5.985364] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-11/11-005d/input/input1
[    5.987206] Goodix-TS 11-0014: i2c test failed attempt 1: -6
[    6.034212] Goodix-TS 11-0014: i2c test failed attempt 2: -6
[    6.083566] Goodix-TS 11-0014: I2C communication failure: -6

ls /dev/i2c-*:

/dev/i2c-11

i2cdetect -y 11:

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- UU -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

fhb, 2021-05-19:
There is also flickering and ghosting when connected to 0x5d as you can see in this short video of the display: https://www.dropbox.com/s/y98anzsv20uu9a4/IMG_3753.MOV?dl=0

Gadgetoid, 2021-05-20:
It looks like you've got a mix of old and new settings in your `/boot/config.txt`- 

The non-fix settings are:

```
dtoverlay=hyperpixel4
dtparam=touchscreen-swapped-x-y=1
dtparam=touchscreen-inverted-x=1
```

You need to remove these and, instead, add the touchscreen settings under each instance of the touchscreen overlays, like so:

```
dtoverlay=hyperpixel4-common
dtoverlay=hyperpixel4-0x14
dtparam=touchscreen-swapped-x-y=1
dtparam=touchscreen-inverted-x=1
dtoverlay=hyperpixel4-0x5d
dtparam=touchscreen-swapped-x-y=1
dtparam=touchscreen-inverted-x=1
```

This should ensure that both 0x14 and 0x5d are configured to match.

I'm a little out of touch with HyperPixel 4 and don't know what's causing the flickering/ghosting. From what I understood with the i2c address switch, it was a race condition during bootup that caused the i2c select pin to be pulled. It shouldn't have any bearing on the *display* driver at all.

fhb, 2021-05-20:
Thank you! Somehow with "0x14" I don't have to swap the x/y-axis:
https://gist.github.com/fhb/1bf17c090baa2d33301b622b26505b6c

"#dtoverlay=vc4-fkms-v3d
display_lcd_rotate=3
dtoverlay=hyperpixel4-common
dtoverlay=hyperpixel4-0x14
#dtparam=touchscreen-swapped-x-y=1
dtparam=touchscreen-inverted-x=1
dtoverlay=hyperpixel4-0x5d
dtparam=touchscreen-swapped-x-y=1
dtparam=touchscreen-inverted-x=1"

Gadgetoid, 2021-05-25:
> There is also flickering and ghosting when connected to 0x5d as you can see in this short video of the display: https://www.dropbox.com/s/y98anzsv20uu9a4/IMG_3753.MOV?dl=0

I've just seen this flickering on a Pi 400 when detected as 0x5d. A cold boot seemed to fix it. And also the i2c. I wonder if there's something weirder going on with the pin state when hot-plugging/rebooting.

closed 2026-08-25 by Gadgetoid