pimoroni/hyperpixel4 #135: random output loss

TalleyWhacker, 2021-05-21:
So i have the hyperpixel 4.0 non touch on an adafruit cyberdeck in my pi 400. Took a bit to even get working but now that i can actually get to the desktop i have a new problem. I lose display output randomly, sometimes i wont even see the desktop, sometimes it lets me flip the screen, i've even gotten minecraft to boot but for some reason i just get black screened no matter what. I've tried a normal HDMI display and it works just fine, its only when i try to swap back. I would appreciate any suggestions! i'm very new to this stuff but so far im tried reseating the ribbon cable and checking all GPIO pins.

Kungpaoshizi, 2021-05-28:
May I post here as well? Perhaps we have similar issues TalleyWhacker? Please read this, please do join the discussion there if you could! :)

https://forums.adafruit.com/viewtopic.php?f=50&t=179733

Otherwise may I have the Pimoroni devs look as well? Should I be posting here or should I open another Issue ticket on github?

Thanks!

TalleyWhacker, 2021-05-28:
Im about to make a post on the adafruit forum since there is more activity. I have gotten it to kinda work but it's still iffy!

petaflot, 2022-08-23:
Had this issue with a square touch when powering the Pi with a 5V USB adapter ; voltage loss in the cable and connectors were causing my 5VDC bus to be at around 4.85 V

now using a 12VDC power supply and a dedicated hat that powers the Pi through the 5V pin of the 40-pin header.

As an alternative, you may also try to solder your power wires directly on the header.

closed 2026-08-25 by Gadgetoid