pimoroni/hyperpixel4 #137: Display not working when Overlay File System is enabled

elaurijssens, 2021-05-24:
**Describe the bug**

Hyperpixel 4 rectangular doesn't work when the Overlay File System is enabled

**To Reproduce**

Install from Github. 
Make display work (which isn't always easy anyway). 
Use `raspi-config` to enable Overlay File System. 
Reboot.  

**Your HyperPixel 4**

Hyperpixel 4 rectangular

1. Rectangular
2. Touch
3. Purchased a couple of weeks ago
4. Connected through GPIO

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. Revision	: c03112
2. Description:	Raspbian GNU/Linux 10 (buster) 
5.10.17-v7l+
3. This is a fresh setup

**Extra debugging information**

When Overlay FS is disabled:
```text
# ls /dev/i2c-*
/dev/i2c-11
```

```text
# i2cdetect -l
i2c-11	i2c       	i2c@0                           	I2C adapter
```

```text
$ i2cdetect -y 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- UU -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

When Overlay FS is enabled:
```text
# ls /dev/i2c-*
/dev/i2c-1  /dev/i2c-11
```

```text
# i2cdetect -l
i2c-1	i2c       	bcm2835 (i2c@7e804000)          	I2C adapter
i2c-11	i2c       	i2c@0                           	I2C adapter
```

```text
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

$ i2cdetect -y 11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- UU -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Gadgetoid, 2021-05-25:
I'm guessing it's breaking the `hyperpixel4-init` script somehow.

What do you see if you run `systemctl status hyperpixel4-init` with OverlayFS enabled/disabled?

elaurijssens, 2021-05-25:
```text
pi@Prusa3D-02:~ $ sudo systemctl status hyperpixel4-init
● hyperpixel4-init.service - HyperPixel 4.0" LCD Display Initialization
   Loaded: loaded (/etc/systemd/system/hyperpixel4-init.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Mon 2021-05-24 22:48:54 CEST; 12h ago
  Process: 431 ExecStart=/usr/bin/hyperpixel4-init (code=exited, status=0/SUCCESS)
 Main PID: 431 (code=exited, status=0/SUCCESS)

May 24 22:48:52 Prusa3D-02.internal.talwyn-esp.nl systemd[1]: Starting HyperPixel 4.0" LCD Display Initialization...
May 24 22:48:54 Prusa3D-02.internal.talwyn-esp.nl systemd[1]: hyperpixel4-init.service: Succeeded.
May 24 22:48:54 Prusa3D-02.internal.talwyn-esp.nl systemd[1]: Started HyperPixel 4.0" LCD Display Initialization.
```

elaurijssens, 2021-05-25:
Nothing unusual, except for installing Octoprint. 

Is there a way to ramp up logging so I can try to figure out what the differences are?

Gadgetoid, 2021-05-25:
You could try manually running `hyperpixel4-init` and seeing it the screen springs into life.

With Overlay FS enabled, if you try and read `/boot/config.txt` does it still contain the hyperpixel setup?

There's not much output from `hyperpixel4-init` to log and, as far as I know, not much to see from the system regarding the enabling of DPI LCDs either.

You can get an idea of the pin states with:

```
git clone git@github.com:wiringPi/wiringpi
cd wiringpi
./build
sudo ./gpio/gpio readall
```

If DPI has enabled you'd expect to see most of the pins in Mode "ALT2".

Not sure what else to try!

```

elaurijssens, 2021-05-25:
Found it. 

using raspi-config creates a new /boot/config.txt which should be the same as the original config.txt before running raspi-config with something like `initramfs initrd.img-5.10.17-v7l+-overlay` appended. 

But it's not the same version, as it had `dtparam=i2c_arm=on` also in there.

I remounted /boot rw, commented it out, and rebooted, and now all is well, display is working again.

closed 2021-05-25 by elaurijssens

WilsonBradley, 2023-03-19:
"Commented it out"  , what is "it" ?

WilsonBradley, 2023-03-19:
better yet, post your working with LCD config.txt