pimoroni/hyperpixel4 #139: HyperPixel screen functions, but becomes black screen backlit frequently.

T53UNG, 2021-06-01:
**PLEASE NOTE**

:warning: GitHub issues are for reporting problems with the software, tech support do not monitor these.
Our software team are smol and busy, and you will not receive a timely response.
If you're having a hardware issue with your Hyperpixel 4, you should contact no support@pimoroni.com.

:warning: Hyperpixel 4 is only officially supported on Raspberry Pi OS. There are too many operating systems with too many quirks to begin to support them all. Issues relating to other operating systems will be closed.

**Describe the bug**

What's gone wrong?

The screen is going out. Randomly it seems to function fine for the first few seconds some times even a few minutes, but it always goes black. 

I have tried this with and without a monitor hooked up to and hdmi port.
**To Reproduce**

What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?
I tried both the manual and one line installer.

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular?
2. Touch or non Touch?
3. Approximately when was it purchased
4. How is it connected to your Pi
 
 rectangular non touch, purchased this year, had issues with it since I got it, but never got it fixed. 
 connected via ada fruit cyberdeck.
**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
2. Which OS release are you using: `lsb_release --description` and `uname -r`
3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?

Pi 400 with latest version of raspbian fresh
**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular
* `dmesg | grep ft5` for HyperPixel 4.0" Square

And check i2c is working:

`ls /dev/i2c-*`

You should see something like:

```text
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-7
```

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

For example:

```text
pi@raspberrypi:~ $ i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Fythios, 2021-06-11:
I was having this same issue, seems there might be a small hardware problem (diode to the Chip Select) between the two. 

I initially found [this ](http://www.reddit.com/r/cyberDeck/comments/nfhcxw/help_with_cyberdeck_hat/) on reddit about it and then found that [Adafruit forums ](https://forums.adafruit.com/viewtopic.php?f=50&t=179733) has info about it too

Galbi3000, 2022-03-11:
I had the same problem too. The desktop appeared for a few seconds then went black. 

I looked at the two links Fythios provided above and found that on page 2 of the replies in the Adafruit forums link there is an option to cut a trace from a diode on the Cyberdeck board. That fixed it for me. The only caveat to this fix is that it may disable using the STEMMA connector beside that diode. This is not a problem for me but it may be for others.

T53UNG, 2022-03-11:
I did cut the trace so the screen is functioning now, but since then I upgrade to raspberry is (bullseye) it seems There's a driver issue. After commenting a certain driver in the /boot/config.txt I can get the screen to come up but rotation does not work

markmiro, 2023-10-14:
I have the same issue but with a Raspberry Pi 4 in my case. I don't have it connected via a cyberdeck

closed 2026-08-25 by Gadgetoid