pimoroni/hyperpixel4 #140: Doesn't work on anything other than stock Raspberry OS, please fix

phalkon13, 2021-06-05:
IT seems like the packages required are only available if you install stock Raspberry OS.  I tried installing this on the 1.5 build of Pimiga (which I believe is Raspberry Pi OS lite) as well as multiple builds of Kali Linux.  On Pimiga, it will bring up the screen, but it is in the wrong orientation, and your rotate command fails, stating that the command code is incorrect.  In Kali Linux, it either won't even install, or if it does, it then locks up on boot after install.
Please add a list of repos and packages required to successfully install this software, so that if someone is installing this on a different OS than Raspbian/RPiOS, it will function properly.

Addendum: Just to test, I installed RPiOS stock, and installed the HP4.0 right after, and it worked great (except touch was backwards), but after installing the Katoolin packages that are available, the HP4.0 does not work at all.  The curl command goes through, the the screen stays backlit black.
I also tried waiting until installing all of the Katoolin packages before installing the screen, and it immediately just goes backlit black screen.

Gadgetoid, 2021-06-18:
Added some generic Xorg configuration instructions in https://github.com/pimoroni/hyperpixel4/commit/cf175ccc22017417c3d8be24a26afa6433705abd

Gadgetoid, 2021-06-22:
Working on the ability to save out a basic Xorg configuration for non-Raspberry-Pi-OS distributions that still use X and don't have the xrandr/lightdm configuration that we usually depend on:

https://github.com/pimoroni/hyperpixel4/commit/236fb219db95c7eeb893b293535563c1019f0b2b

Kabouik, 2021-11-24:
Thank you for your efforts towards making this screen more generic across distributions. Just to be sure as I'm about to order one, would the Hyperpixel 4 work on an x64 machine running Debian and having Pi-compatible pins (I'm thnking about the Seeedstudio Odyssey X86J4125)?

Gadgetoid, 2021-11-25:
i2s is audio, i2c is data so it should have no bearing on compatibility.

There's almost no chance the Odyssey SBC or any other random SBC supports DPI in the same configuration across the same pins as the Pi. Only Seeed could really elaborate upon this.

Kabouik, 2021-11-25:
> i2s is audio, i2c is data so it should have no bearing on compatibility.
Thanks for the answer. Yes, that's what I found later but it's good to have your confirmation.

> There's almost no chance the Odyssey SBC or any other random SBC supports DPI in the same configuration across the same pins as the Pi. Only Seeed could really elaborate upon this.
That I don't understand. [These](https://wiki.seeedstudio.com/ODYSSEY-X86J4105-GPIO/) are the pins of the Odyssey SBC. Are they not exactly the same as the Pi pins?

Gadgetoid, 2021-11-25:
> Are they not exactly the same as the Pi pins?

Very, very superficially yes. For anything non-trivial - like GPU-based DPI (Display Parallel Interface) support - you're out of luck. 

RPi's GPIOs have up to six alt modes that are - afaik - not replicated by any "pin compatible" SBC. See: https://elinux.org/RPi_BCM2835_GPIOs

Kabouik, 2021-11-25:
Thank you very much for the detailed answer. I think it's pretty clear then that no pin-based display would work for the Odyssey, since the vast majority of pin displays are designed around the Pi models. I'll have to focus on HDMI displays then.

Great screen though, wish I could have used it!

Gadgetoid, 2021-11-25:
SPI displays *should* work but driver support is spotty and a lot of the nice utilities for mirroring framebuffers are very Pi-specific :grimacing:

HDMI is definitely the path of least resistance. I wish we'd do a little HDMI display and smol cable. So much less hassle :laughing:

Kabouik, 2021-11-25:
You're giving me some hope again! It is not clear to me which utilities I would miss because I have never used SPI displays so don't know what are the limitations and how the driver help working around them. My understanding is it's not really plug-and-play, so I assume rotation would be difficult if not all utilities work on the SBC, and maybe the orientation of the touch matrix too? Also, I am not sure if SPI displays can be combined with HDMI displays to make multi-monitor setups within the OS, or whether they are completely separate things.

My concern is the one-liner installation script you developed may not work at all, partly because, as you said, the pins won't be an exact match with those of a Pi, and also because the Odyssey SBC is an x64 architecture and may lack dependencies in Debian if they have been compiled only for the Pi. Another concern, but that one is not related to this issue on distribution compatibility, is I couldn't find what voltage the Pimoroni SPI displays require. Apparently the Odyssey IO voltage is 1.8V.

Kabouik, 2021-11-28:
> HDMI is definitely the path of least resistance. I wish we'd do a little HDMI display and smol cable. So much less hassle laughing

To end the hardware off-topic I started (sorry), wouldn't [this](https://www.adafruit.com/product/2219) solvie my first issue to make the Hyperpixel4 an HDMI (and hence hardware agnostic) device, even x86)64? This would bring me back on topic in this issue since I would like to use it on Debian Bullseye.

Gadgetoid, 2022-04-12:
There's some upstream - at least to Pi's Linux fork - support for HyperPixel 4 and HyperPixel 4 Square now available in Raspberry Pi OS. It may take some time, but I'd expect this to trickle over to other distributions and start to smooth over the issues.

@Kabouik I don't know if the linked board would work with HyperPixel. It would certainly be a pain to set up.

Hopefully once the mentioned distros get updated the instructions in https://github.com/pimoroni/hyperpixel4/issues/177 will apply.

closed 2022-04-12 by Gadgetoid

Kabouik, 2022-04-12:
Thanks a lot for the clarification on that specific board @Gadgetoid!