pimoroni/hyperpixel4 #148: Love this screen, but I don't like GUI's

snarflakes, 2021-08-15:
Forums directed me here. I'm oldschool. I code for the raspberry pi at the NON-GUI command line in python3 directly to my SPI display using PIL,PILLOW.  Can I do that with this screen? I can't tell if there are any modules that I can import so I can just code in python3 to directly display directly to this screen and not have to boot to the GUI or run program through the GUI.  

many thanks,
Snarflakes

**PLEASE NOTE**

:warning: GitHub issues are for reporting problems with the software, tech support do not monitor these.
Our software team are smol and busy, and you will not receive a timely response.
If you're having a hardware issue with your Hyperpixel 4, you should contact support@pimoroni.com.

:warning: Hyperpixel 4 is only officially supported on Raspberry Pi OS. There are too many operating systems with too many quirks to begin to support them all. Issues relating to other operating systems will be closed.

**Describe the bug**

What's gone wrong?

**To Reproduce**

What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular?
2. Touch or non Touch?
3. Approximately when was it purchased
4. How is it connected to your Pi

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
2. Which OS release are you using: `lsb_release --description` and `uname -r`
3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular
* `dmesg | grep ft5` for HyperPixel 4.0" Square

And check i2c is working:

`ls /dev/i2c-*`

You should see something like:

```text
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-7
```

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

For example:

```text
pi@raspberrypi:~ $ i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Gadgetoid, 2021-08-19:
You can use Pygame to render directly to the screen but I've had mixed results doing this from the command-line. This is mostly because the drivers and version of SDL aren't *quite there* in Raspberry Pi OS yet.

Here's me chasing those kinds of issues with our 480x480 pixel "round" display - https://github.com/pimoroni/hyperpixel2r-python/issues/2

You can probably draw inspiration from the Hyperpixel2r Python examples, but for touch input you'd probably use Pygame's mouse support directly rather than the (incompatible with Hyperpixel 4 anyway) touch driver in this repository.

snarflakes, 2021-08-19:
Hi thanks!
Pimoroni help desk just emailed me this:
_________________________________________________________________________
Yes, it should be. If you install the drivers and then edit /boot/config.txt to either comment out or remove the line:
dtoverlay=hyperpixel4 
That should leave you with the software installed and the Pi using the default display. You can then specify display DSI-1 in PIL to output to the Hyperpixel.

Kind regards,

Sebwap, 2021-09-07:
> Hi thanks!
> Pimoroni help desk just emailed me this:
>
> Yes, it should be. If you install the drivers and then edit /boot/config.txt to either comment out or remove the line:
> dtoverlay=hyperpixel4
> That [...]

Does somebody have tried this ?
I would love to have an answer to this before buying the screen !

Thx :)

fireduck64, 2021-11-24:
I am interested in this as well.  In my use case, I'll be generating images that I will want to push to the display every few minutes.  Basically as a status board with date time, weather and some key metrics.  I guess I could make that work with using xorg, skipping login and running on program full screen but I'd rather just build the image and push it to the display if possible.

SeedSigner, 2021-12-09:
Also curious about this use case; our code base uses Pillow and would like to verify that works before digging in. Not interested in replicating a standard Pi GUI desktop.

fireduck64, 2021-12-13:
Just in case it helps anyone else, I've found that if you follow the instructions in this repo for driver installation, then the hyperpixel will be the console output of the Pi.  From there, you can install a program called 'fim' and use it to display images to the hyperpixel.
Example: fim -q 720.png

The -q makes it now show data about the image and just the raw image itself.  This seems to work, with the limitation that any system kernel logs overwrite the image a bit, but if you are refreshing occasionally that shouldn't be a big problem.

My plan is to have some other processes that output a new image formatted for the screen size and then occasionally kill and restart fim to update the screen.  This works for my use case, which is data that changes at most every few minutes.

thirdr, 2025-01-08:
Closed as answered.

closed 2025-01-08 by thirdr