pimoroni/hyperpixel4 #151: It would be really helpful to have a basic demo python program template for displaying to this DPI screen.

snarflakes, 2021-08-28:
I code in python3 here. 
i'm used to SPI displays, where there is an easy example .py script that shows me how and what module to import and then a basic initialize screen code sequence that describes screen so I can use PIL to disp.image via my python code. I would immensely appreciate it. i already commented out dtoverlay=hyperpixel4 in the config.txt file.

thanks
Snarflakes

Gadgetoid, 2021-08-29:
The screen doesn't have an SPI data interface or anything befitting basic init and usage from Python. It either runs over DPI driven by the Raspberry Pi's GPIO or - for all intents and purposes - it doesn't run at all. The speed and timing requirements for generating a DPI signal are pretty far outside the grasp of Python.

That said if you let the Pi handle the display- you can just use any Python GUI methods to display images on it. This can be anything from PIL running in X and using the "display()" method on an image (if memory serves) or using fullscreen pygame as per these examples - https://github.com/pimoroni/hyperpixel2r-python/tree/master/examples

YMMV though, it's been surprisingly difficult to get pygame/SDL to behave themselves on the Pi - though a more traditional screen resolution probably helps.

I'll have to look into porting those examples over from HyperPixel round. It only got the special treatment since it's pretty difficult to use a small round display for regular desktop use.

snarflakes, 2021-08-31:
thanks for the steering me in the right direction

closed 2021-08-31 by snarflakes