pimoroni/hyperpixel4 #155: Display blank on Raspberry Pi OS Bullseye

shbisson, 2021-11-09:
**PLEASE NOTE**

:warning: GitHub issues are for reporting problems with the software, tech support do not monitor these.
Our software team are smol and busy, and you will not receive a timely response.
If you're having a hardware issue with your Hyperpixel 4, you should contact support@pimoroni.com.

:warning: Hyperpixel 4 is only officially supported on Raspberry Pi OS. There are too many operating systems with too many quirks to begin to support them all. Issues relating to other operating systems will be closed.

**Describe the bug**

Display blank on Raspberry Pi OS Bullseye. Flashes during reboot and then stays blank. Could be an issue with the new KMS display drivers?

**To Reproduce**

Install using online installer on Bullseye, reboot.

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

Square touch 2021

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. Raspberry Pi 4 4GB Revision        : c03114
2. Description:    Raspbian GNU/Linux 11 (bullseye) 5.10.63-v7l+
3. Fresh install

$ i2cdetect -y 22
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

shbisson, 2021-11-09:
Going back to Buster fixed things, so there's clearly some incompatabilty there! Will wait for updated drivers before I attempt the update again...

Gadgetoid, 2021-11-09:
Seeing a similar issue on the Pi Zero 2 W. 😬 Looking into it.

shbisson, 2021-11-09:
It seems to install on i2c 22 not 7; I wonder if this might be the reason?

Gadgetoid, 2021-11-09:
Commenting out `dtoverlay=vc4-kms-v3d` seems to work reliably for me. I'd say it's definitely something up with the display driver.

Touch is working for me (Bullseye, Pi Zero 2 W) after some tweaks. For some reason `gpio-backlight` was stealing the touch pins.

The display is *working* with `dtoverlay=vc4-kms-v3d` and I see stuff during startup, but the X server just doesn't want to achknowledge it.

Looks like the change from `vc4-fkms-v3d` to `vc4-kms-v3d` will also break the rotation support check in `hyperpixel4-rotate`.

mickeyvanolst, 2021-11-09:
@Gadgetoid are you also able to get it working with openGL enabled? I can get it running with your instructions, but not when openGL is enabled in the config.

Gadgetoid, 2021-11-09:
> @Gadgetoid are you also able to get it working with openGL enabled? I can get it running with your instructions, but not when openGL is enabled in the config.

Nope, enabling openGL loads that overlay I think?

You can try adding: `dtoverlay=vc4-kms-dpi-at056tn53v1` to `/boot/config.txt` which has a close approximation of some new device tree settings I need to supply for things to work properly with kms enabled.

This seems to work *but* (and this is a big but) rotation seems to be broken, split output between HyperPixel4 + HDMI is not supported and the HyperPixel rotate script will also not work (well it will break things if you use it in --xorg mode).

On the up side, touch input seems to have grown up a lot. It appears to detect touches as touches (hiding the cursor) rather than ponderously poking the curser around.

mickeyvanolst, 2021-11-10:
Using `dtoverlay=vc4-kms-dpi-at056tn53v1` for me just generates a blank screen after bootup.

rschu68, 2021-11-10:
Same with me: Commenting out dtoverlay=vc4-kms-v3d works for Pi3-2835
? not tested before going into production maybe

rschu68, 2021-11-10:
No worries .. the work you do for the community is verry much appriciated!

Grz
~remke

mickeyvanolst, 2021-11-10:
@Gadgetoid do you have a url to the latest Raspberry OS image that you fully support?

shbisson, 2021-11-10:
After a rebuild I am running happily on downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2021-05-28/; which is the latest Buster image on the Raspberry Pi site.

shbisson, 2021-11-10:
@Gadgetoid Thanks for your work on this!

Gadgetoid, 2021-11-10:
Looks like I could have a solution using DKMS this week, albeit rotation support is horribly broken and that's just something we're going to have to wait for Raspberry Pi to fix.

Gadgetoid, 2021-11-11:
# This is EXTREMELY ALPHA

* Rotation/positioning in Screen Layout/arandr is currently kinda broken you need to manually run `/usr/share/dispsetup.sh` to restore it.
* This *only works on HyperPixel4 Rectangular*, no square yet!

https://github.com/pimoroni/hyperpixel4/tree/pi4-kms-dpi

To install:

```
git clone https://github.com/pimoroni/hyperpixel4 -b pi4-kms-dpi
cd hyperpixel4
sudo ./install.sh
```

The pi4 kms DPI branch includes some prototype (proof of concept, mostly) setup that should work with Bullseye.

It uses a DKMS package to build a new (somewhat hacked together) panel driver for HyperPixel 4. It has to install kernel headers and this is painfully slow. But it works.

Over the coming weeks I'm going to delve into panel drivers and try to write a complete driver for HyperPixel 4 Rectangular that handles display init, etc, with a view to upstreaming that into the mainline kernel.

The same applies to HyperPixel 4 Square (pre and post 2021) except I think that will just be display timings added to simple-panel.c.

⚠️ If you use this on a production Pi, expect to have to re-image later since this is not representative of the final state of software support for HyperPixel products on Bullseye.

bert269, 2021-11-14:
Thanks - yes, this 'alpha' seems to work to at least get the screen to come up under Bullseye. I must report that the rotation is still not working, I get these messages:
**_hyperpixel4-rotate left**
HyperPixel 4: Square - Display/Touch Rotation
This rotate utility only works with the Raspberry Pi OS desktop version or X-based alternatives.

No DISPLAY variable set, trying :0.0
xhost:  unable to open display ":0.0"
xhost:  unable to open display ":0.0"
Disabling xhost access control...
Rotating display left
Can't open display :0.0
grep: /etc/lightdm/lightdm.conf: No such file or directory
The display setup script is not configured in /etc/lightdm/lightdm.conf so your saved settings *will not be restored upon reboot*
Maybe you're not using Raspberry Pi OS/lightdm?
Add "display-setup-script=/usr/share/dispsetup.sh" to /etc/lightdm/lightdm.conf or use the Xorg config method (add --xorg to this command)

Saving display settings to /usr/share/dispsetup.sh
chown: cannot access '/usr/share/dispsetup.sh': No such file or directory
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
ModuleNotFoundError: No module named 'screenlayout'
chown: cannot access '/usr/share/dispsetup.sh': No such file or directory
chmod: cannot access '/usr/share/dispsetup.sh': No such file or directory
Failed to persist display orientation.
Make sure you're running Rasperry Pi OS desktop on a Pi 4 or Pi 400
xhost:  unable to open display ":0.0"
Re-enabling xhost access control..._

I tried this after the initial attempt - so I do not know what 'older stuff' was stuck there before I tried the alpha.
I will wipe the sdcard and try again with this with a clean install and report back.

Gadgetoid, 2021-11-16:
Hmm looks like dispsetup/lightdm conf have gone entirely since I last tested. I'll have to dig into arandr and see what it does to preserve rotation settings upon reboot now. Assuming it actually does yet! It was all very, very broken and getting daily updates. :grimacing:

ramstik, 2021-11-17:
I'd just like to report my findings here. Had the same issue, black screen after `curl | bash` install, SSH was fine, etc.

I stumbled upon [this comment](https://github.com/pimoroni/hyperpixel4/issues/111#issuecomment-723654745) in which I added the line `gpio=0-25=a2` to the very bottom underneath `dtoverlay=hyperpixel4`. This brought the screen online. I had originally included the very last rotate command but I discovered that this put my touch 270 degrees out of alignment, so I went and removed it.

Attempted `hyperpixel4-rotate` and it spat out the "you're missing `dtoverlay=vc4-fkms-v3d` in /boot/config.txt" crap. Went in and un-commented it under `[all]` and rebooted. Rotate command worked as expected, touch and all.

Here's my current /boot/config.txt **with all of the commented items stripped out**
```
dtparam=i2c_arm=on
dtparam=spi=on

dtparam=audio=on

camera_auto_detect=1

display_auto_detect=1

max_framebuffers=2

disable_overscan=1

[cm4]
otg_mode=1

[all]
dtoverlay=vc4-fkms-v3d

[pi4]
arm_boost=1

[all]
gpu_mem=256
dtoverlay=w1-gpio

dtoverlay=hyperpixel4
gpio=0-25=a2
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
# display_rotate=1
```

And system info:
```
pi@rpi4:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
pi@rpi4:~ $ uname -r
5.10.63-v7l+
```

RAYLEGION, 2021-11-29:
having the same issue, Display blank on Raspberry Pi OS Bullseye. Flashes during reboot and then stays blank. I am assuming that a fix has not yet been released, any indication of release date ?

Tubaleviao, 2021-12-02:
I have the same issue using the raspberry pi 400.

catch-twenty-two, 2021-12-04:
Same issue here on Rasberry pi 400

catch-twenty-two, 2021-12-04:
> I have the same issue using the raspberry pi 400.

I thought I had the same issue on the 400, but it turns out it was the connector I was using, this might be of help: https://github.com/pimoroni/hyperpixel4/issues/139

itzskillzz, 2021-12-11:
I have the same issue using the raspberry pi 400, is there a fix soon?

Nopzen, 2021-12-23:
> Commenting out `dtoverlay=vc4-kms-v3d` seems to work reliably for me. I'd say it's definitely something up with the display driver.
>
> Touch is working for me (Bullseye, Pi Zero 2 W) after some [...]

Had issue where my Pi4 4GB with Raspberry Pi OS 32-bit also just gave black screen on start up, replacing: `dtoverlay=vc4-kms-v3d` with `vc4-fkms-v3d` made it work with me :)

sjtoik, 2021-12-28:
Yes, with `vc4-fkms-v3d` in dtoverlay and with 32bit armv7l version of Raspberry OS, the current drivers work, although with a few pixel folding over when you do a rotation. The overlap is always at the same side where the orange display ribbon is located. I got this with a clean install of current Raspberry OS.

As the overlay logic is changing, and you need to rewrite the device tree drivers eventually, please consider extending the builds to aarch64. The beta version of Raspberry OS is 64bit already and a lot of new development is there also.

wohali, 2022-01-13:
Hey, thanks for all your hard work on this. I'm watching https://forums.raspberrypi.com/viewtopic.php?p=1958231#p1958231 with keen interest, though I have not yet tried compiling @6by9 's kernel modules. Very thankful to you both - looks like a difficult journey.

Until then, I'm able to get the display working under bullseye with the `pi4-kms-dpi` branch of this repo as described above. 

I did end up having to compile the `libinput` branch of `xinput_calibrator` from [here](https://github.com/kreijack/xinput_calibrator) to get touch calibration working, and stuck  its output in `/etc/X11/xorg.conf.d/99-calibration.conf`. I'm not sure where the canonical version of these settings is supposed to go, but it actually worked with (and I feel more confident in) putting the settings in `/etc/X11/xorg.conf.d` rather than under the distro's rules in `/usr/share/X11/xorg.conf.d/`. Would this be a better choice for the `hyperpixel4-rotate` script?

Unfortunately, I'm unable to get rotation working either, which is what I'm waiting for at the moment. (I tried `hyperpixel4-rotate` and `arandr` both.)

6by9, 2022-01-13:
I haven't looked in detail at the pi4-kms-dpi branch, but taking the raspberrypi/linux kernel from after December 3 when https://github.com/raspberrypi/linux/pull/4753 was merged, and then add:
```
dtoverlay=vc4-kms-dpi-generic
dtparam=hactive=480,hfp=10,hsync=16,hbp=59
dtparam=vactive=800,vfp=15,vsync=113,vbp=15
dtparam=clock-frequency=32000000
dtparam=hsync-invert,vsync-invert,pixclk-invert
dtparam=rgb666-padhi
```
alongside the "normal" ```hyperpixel4-init``` bitbash SPI script should get the panel side up and running.
Memory says that the ```hyperpixel4-touch-overlay.dts``` overlay does then work for the touchscreen too, but mainly as hyperpixel4-init bypasses the kernel and just uses GPIOs.

wohali, 2022-01-13:
I pulled down the latest `rip-5.10.y` sources today and cross-compiled/installed it via [these instructions](https://www.raspberrypi.com/documentation/computers/linux_kernel.html): `[    0.000000] Linux version 5.10.90-v7l+ (joant@deb-11) (arm-linux-gnueabihf-gcc (Debian 10.2.1-6) 10.2.1 20210110, GNU ld (GNU Binutils for Debian) 2.35.2) #1 SMP Thu Jan 13 16:13:58 EST 2022
`

So far, no joy with this overlay at all, and the touchscreen driver [throws a backtrace](https://gist.github.com/wohali/8e04f8a738abd09cc4c37bcc544c3cad).

I'll try a fresh install of bullseye and try again via a local compile next.

6by9, 2022-01-14:
https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpio/gpiolib.c#L3323
```
		/*
		 * We must not be output when using IRQ UNLESS we are
		 * open drain.
		 */
		WARN_ON(test_bit(FLAG_IS_OUT, &desc->flags) &&
			!test_bit(FLAG_OPEN_DRAIN, &desc->flags));
```
Trying to enable an IRQ on an output.
If you're using the above workaround with ```dtoverlay=vc4-kms-dpi-generic```, then it probably means something has left the GPIO as an output when the touch driver comes around to initialise. Not the end of the world.

Reading the info I have, the backlight is just on/off via GPIO 19. With https://github.com/raspberrypi/linux/pull/4815 merged you can now add that to ```dtoverlay=vc4-kms-dpi-generic``` with ```dtparam=backlight-gpio=19```, but it won't change that it should be either on or off.
GPIO19 function ALT5 is PWM1, so if you're not using audio then you could send PWM to the backlight. I have no idea if it's happy with a PWM drive though.

Gadgetoid, 2022-01-14:
> I have no idea if it's happy with a PWM drive though.

PWM was actually something I looked into donkeys years ago- the LCD would be happy with a dimmable backlight but:

1. AFAIK the `pwm-backlight` module is not shipped with Raspberry Pi OS
2. That's partially on me, because I requested it but had nowhere near the level of device-tree experience to make it work properly
3. So the issue died on the vine, see: https://github.com/raspberrypi/linux/issues/2063

2017 😲 

Thanks for weighing in here @6by9 - I can't imagine you're stretched any less thin than me, so your time is appreciated!

6by9, 2022-01-14:
pwm-backlight is shipped by default for the Pi4/CM4 config following https://github.com/raspberrypi/linux/pull/4742 for the CutiePi display.

Seeing as we have a generic DPI display overlay then I don't see why we shouldn't add ```pwm-backlight``` as a module, and provide it as an option in that overlay. It does block using analogue audio at the same time, so it might need a little more thought.
https://github.com/raspberrypi/linux/pull/4812 can then be updated to use pwm-backlight too.

BTW did you note my comment on https://forums.raspberrypi.com/viewtopic.php?p=1958259#p1958259 that I'm happy to look at adding your round 2.1 (PIM579) and 4.0 square (PIM470) variants of HyperPixel display if you want to ship samples to Pi Towers for my attention. If they are are also based on ILI9806E then it would change my approach slightly in the driver.

rcarmo, 2022-01-28:
Well, I came to this issue after experiencing the same thing, and flashed `buster` instead - the display now works, but the backlight has a vignette effect (which it didn't during boot on Debian 11...), so I suppose neither is a full solution. 

Is there an easy fix for the vignetting in `buster`? I can't divine that from reading the discussion above, since most recent commentary seems to be about the latest release.

rcarmo, 2022-01-28:
OK, I seem to have (at least temporarily) fixed my problem. I reflashed `bullseye` and changed `config.txt` like so:

```
# original value: dtoverlay=vc4-kms-v3d
dtoverlay=vc4-fkms-v3d

# after the following line:
dtoverlay=hyperpixel4
# added this line:
gpio=0-25=a2
```

I also did a full `dist-upgrade`, but I don't think that had anything to do with it (I do that after every new install, just after I run `raspi-config` for minor tweaks).

rstarks, 2022-04-03:
I was in a similar situation with a hyperpixel4 rectangular display connected by an adafruit cyberdeck hat to a raspberry pi 400. I would see a splash screen briefly and then sometimes a grayish vignette followed by all black, Some of the suggestions in this thread seemed to help, but ultimately what fixed my problem was cutting a trace on the hat. I had read this prior to acquiring everything but spaced it awaiting delivery. Anyway, after troubleshooting it for a bit, I re-read this [important detail](https://learn.adafruit.com/cyberdeck-bonnet-and-hat/fixing-pin-conflicts#option-1-cut-a-trace-3104511-6), which worked. I figured I'd mention it here in case anyone is utilizing a similar setup with an extender board. Pin conflicts from other components might result in similar symptoms.

Gadgetoid, 2022-04-12:
I'd appreciate if everyone having issues here - who hasn't fled the scene due to these long-standing and understandably frustrating problems - giving the new Raspberry Pi OS and drivers a try to see how they fare: https://github.com/pimoroni/hyperpixel4/issues/177

Please assume all previous setup instructions, install methods, hacks, tricks and tips to be obsolete and legacy- albeit some Xorg rotation and touch config still comes in useful. Continue discussion and feedback on the linked thread!

Thank you, and sorry this has taken so very, very long.

closed 2022-04-12 by Gadgetoid

icurnvs, 2022-05-11:
Thanks for this!  Just tried the instructions in [#177](https://github.com/pimoroni/hyperpixel4/issues/177) on a 3a+ and it worked swimmingly.  I'd tried the legacy drivers before since I didn't think the 3a+ could be lumped in with the pi4 and pi3b.  Guess I was wrong! Ran into the issues outlined above.  Followed the instructions in #177 and it all worked like a charm.