pimoroni/hyperpixel4 #157: Hyperpixel4 TOUCH not responding to touch on rpi4

omegaforc3, 2021-12-12:
**Describe the bug**

I have bought a raspberry pi 4b and an hyperpixel4 screen (touch, rectangular). The display works properly but the touchscreen is not responding to touch.

**To Reproduce**

I have followed those instructions in order to install everything (Octopi, hyperpixel 4 drivers, octodash) : https://kevenaar.name/octopi-with-octod ... pixel-4-0/

I have used the one line installer (curl -sSL https://get.pimoroni.com/hyperpixel4 | bash) and the installation seemed to work properly, no error messages.

**Your HyperPixel 4**

1. Is it Square or Rectangular? 
RECTANGULAR

2. Touch or non Touch?
TOUCH

3. Approximately when was it purchased
2 weeks ago

4. How is it connected to your Pi
to the GPIO header using a ribbon cable

**Your Raspberry Pi**

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
Raspberry pi 4b

above command returns : Revision : c03114

2. Which OS release are you using: `lsb_release --description` and `uname -r`
I installed Octopi on a blank SD card

lsb_release returns : Raspbian GNU/Linux 10 (buster)
uname -r returns : 5.10.63-v71+

3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?
This is a fresh setup. I formatted the sd card and only installed, in this order, Octopi, hyperpixel4 drivers and Octodash

**Extra debugging information**
If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular

This returns : 

[    5.843941] Goodix-TS 22-0014: supply AVDD28 not found, using dummy regulator
[    5.844203] Goodix-TS 22-0014: supply VDDIO not found, using dummy regulator
[    5.845297] Goodix-TS 22-0014: i2c test failed attempt 1: -6
[    5.886115] Goodix-TS 22-0014: i2c test failed attempt 2: -6
[    5.925297] Goodix-TS 22-0014: I2C communication failure: -6
[    5.926202] Goodix-TS 22-005d: supply AVDD28 not found, using dummy regulator
[    5.926440] Goodix-TS 22-005d: supply VDDIO not found, using dummy regulator
[    5.929400] Goodix-TS 22-005d: i2c test failed attempt 1: -6
[    5.975880] Goodix-TS 22-005d: i2c test failed attempt 2: -6
[    6.015305] Goodix-TS 22-005d: I2C communication failure: -6

**And check i2c is working:**

**`ls /dev/i2c-*`** returns 
/dev/i2c-22

`**i2cdetect -y 22**` returns 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

Also found this when requesting sudo desmg

(theres are pages of stuff but this has caught my attention, its written in red) :

[ 4.778716] pinctrl-bcm2835 fe200000.gpio: pin gpio2 already requested by rpi_backlight; cannot claim for fe804000.i2c
[ 4.778742] pinctrl-bcm2835 fe200000.gpio: pin-2 (fe804000.i2c) status -22
[ 4.778775] pinctrl-bcm2835 fe200000.gpio: could not request pin 2 (gpio2) from group gpio2 on device pinctrl-bcm2711
[ 4.778794] i2c-bcm2835 fe804000.i2c: Error applying setting, reverse things back

The follwing is a copy of my /boot/config.txt

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
display_lcd_rotate=3
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
# enable raspicam
start_x=1
gpu_mem=128

dtoverlay=hyperpixel4
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6

morphias2004, 2022-03-07:
Try following the guide I wrote that is now part of the OctoDash Wiki.

https://github.com/UnchartedBull/OctoDash/wiki/HyperPixel-4

Gadgetoid, 2022-04-12:
@morphias2004 the way display drivers work, and specifically how HyperPixel4 is set up, has changed quite dramatically in newer versions of Raspbian. I'm still not entirely familiar with how quickly OctoPi etc stays ahead of new releases, but this is likely to affect (hopefully positively) the install process in future.

See: https://github.com/pimoroni/hyperpixel4/issues/177

rschweter, 2023-01-11:
Saving this here for posterity. I had the same issue and my fresh in stall of octopi did not have i2c enabled.