pimoroni/hyperpixel4 #162: [HARDWARE][BUG][INCOMPATIBILITY][??] HyperPixel 4 Display abnormal bars using Sandisk U3 MicroSD card!?

bolsoncerrado, 2022-01-06:
Hi!

I was upgrading my Pi3B+ from a SAMSUNG U1 microSD card to a SANDISK U3 microSD when the system just booted fine but the Hyperpixel4 screen just flickers showing vertical lines and blurs when scrolling up and down. The touch feature works fine, but the display is definetly wrong.

Reverting back to the Samsung U1, the problem goes away!?!?! I don't think it's voltage related as the Pi does not report the lightning bolt and I have an identical Hyperpixel 4 running on a Pi3B+ and Samsung U3 running right below this system (I use these PIs+HP4 alongside the Prusa MK3 printers).

So I'm not sure what's wrong here? Hardware incompatibility!?!?

I leave you video of the effect with both the Samsung fine display and the wrong Sandisk one:

SAMSUNG:
https://www.dropbox.com/s/2jnhizae3q48hm9/20220106_095617.mp4?dl=0

SANDISK:
https://www.dropbox.com/s/j1fb0uszcmh6xvt/20220106_095117.mp4?dl=0

And this is the offending card:

https://www.dropbox.com/s/tx92gdvj7ih7oxp/20220106_095454.jpg?dl=0

Let me know what do you think? I'm really puzzled!

bolsoncerrado, 2022-01-10:
Now this is interesting!! (I hope?) 

In my OP I suggested a Samsung U3 setup was running fine... I just noticed this today on that setup: a white vertical line flickering on the leftmost side of the screen :???

https://www.dropbox.com/s/dl69po4s98m8dho/20220109_143656.mp4?dl=0

Just hope all is related here!

Tomorrow I'm planing to add more U3s to our Farm so I'll keep reporting any findings, either for good or bad.

Gadgetoid, 2022-01-11:
Sorry for the lack of response here- I didn't have any useful insight to contribute! I've not come across different SD cards changing behavior before, nor anything close enough to intuit what might be causing this. Actually stumped!

Thanks for continuing to prod and poke!

bolsoncerrado, 2022-01-11:
start_x=1

gpu_mem=128

dtoverlay=hyperpixel4,hyperpixel4-rotate left,touchscreen-swapped-x-y,touchscreen-inverted-x,touchscreen-inverted-y

overscan_left=0

overscan_right=0

overscan_top=0

overscan_bottom=0

framebuffer_width=800

framebuffer_height=480

enable_dpi_lcd=1

display_default_lcd=1

dpi_group=2

dpi_mode=87

dpi_output_format=0x7f216

display_rotate=3

hdmi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6

#dtoverlay=gpio-poweroff,gpiopin=19,active_low=1

Do you spot anything out of bonds in this config.txt OctoPI image uses?

Thanks!

closed 2026-08-25 by Gadgetoid