pimoroni/hyperpixel4 #163: i2c/spi mappings disappear from /dev on Pi Zero 2W

justSem, 2022-01-06:
As many know by now, to get the display working under bullseye you have to remove:

dtoverlay=vc4-kms-3d from /boot/config.txt

As I was intending to use the alternate i2c bus I noticed that, when removing this overlay all SPI and I2C mappings seem to disappear from /dev.

I've been tinkering around but haven't really been able to determine a root cause or a proper solution.
The only thing I've noticed is the error thrown by i2c-bcm2835 in the kernel log.

Toying around by enabling/disabling the various i2c modules doesn't make a difference.

Another fun fact in this case is that I have another Pi zero 2W, but without the Hyperpixel display. This one - however - _does_ give me the i2c mappings even with vc4-kms-3d enabled.

So my guess is that something's going wrong in the device-tree mappings, but since that's not really my area of expertise I haven't been able to find out much as of yet.

**Things I've tried so far**
- I've tried different branches from the Hyperpixel repo (rpi3, rpi4-i2c-fix, rpi4-kms-dpi)
- I've tried enabling/disabling i2c-related kernel modules (in order to match with my other i2c-working Pi)
- I've tried various options in /boot/config.txt as was stated in various issues on GitHub

I specifically disabled/enabled the i2c_gpio and i2c_algo_bit modules as they weren't loaded on my pi that _does_ work. That didn't seem to make any difference

**Environment**
Debian Bullseye on a Pi Zero 2 W.
Hyperpixel 4 Rectengular Touch, purchased a few weeks ago.

**Debug script output**

```

## Platform Information

Revision	: 902120
Description:	Raspbian GNU/Linux 11 (bullseye)
5.10.63-v7+

## Touchscreen logs

Rectangular: Goodix
Square: ft5

## I2C Devices and Mappings

ls: cannot access '/dev/i2c-*': No such file or directory

Error: I2C bus name doesn't match any bus present!
Usage: i2cdetect [-y] [-a] [-q|-r] I2CBUS [FIRST LAST]
       i2cdetect -F I2CBUS
       i2cdetect -l
  I2CBUS is an integer or an I2C bus name
  If provided, FIRST and LAST limit the probing range.

## Boot Config

dtparam=i2c_arm=on
dtparam=spi=on

dtparam=audio=on

camera_auto_detect=1

display_auto_detect=1

max_framebuffers=2

disable_overscan=1

[cm4]
otg_mode=1

[... 127 lines cut ...]
[    8.394822] bcm2835-codec bcm2835-codec: Loaded V4L2 image_fx
[    8.462164] pinctrl-bcm2835 3f200000.gpio: pin gpio2 already requested by soc:fb; cannot claim for 3f804000.i2c
[    8.462192] pinctrl-bcm2835 3f200000.gpio: pin-2 (3f804000.i2c) status -22
[    8.462211] pinctrl-bcm2835 3f200000.gpio: could not request pin 2 (gpio2) from group gpio2  on device pinctrl-bcm2835
[    8.462226] i2c-bcm2835 3f804000.i2c: Error applying setting, reverse things back
[    8.462264] i2c-bcm2835: probe of 3f804000.i2c failed with error -22
[    8.791514] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    8.951842] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    9.015823] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[    9.141471] brcmfmac: F1 signature read @0x18000000=0x1542a9a6
[    9.151533] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
[    9.151851] usbcore: registered new interface driver brcmfmac
[    9.328618] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
[    9.331515] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
[    9.359251] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Oct  9 2020 14:44:32 version 9.88.4.65 (test) (f149b32@shgit)  (r679549) FWID 01-f40f3270
[   10.774611] random: crng init done
[   10.774636] random: 7 urandom warning(s) missed due to ratelimiting
[   10.874268] 8021q: 802.1Q VLAN Support v1.8
[   11.030810] uart-pl011 3f201000.serial: no DMA platform data
[   11.044778] Adding 102396k swap on /var/swap.  Priority:-2 extents:2 across:1544188k SSFS
[   11.369733] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   11.406284] Bluetooth: Core ver 2.22
```

Any ideas, pointers or suggestions are greatly welcome!

Gadgetoid, 2022-01-06:
Does touch still work? The pins for i2c/SPI are used by Hyperpixel's DPI bus, so they should never be enabled alongside Hyperpixel and wouldn't work anyway.

I'm not sure how or why `dtoverlay=vc4-fkms-v3d` would make a difference, but I guess it could break Hyperpixel just enough that it doesn't load and trounce the i2c/SPI interfaces.

eg:

```
[    8.230453] pinctrl-bcm2835 3f200000.gpio: pin gpio9 already requested by soc:fb; cannot claim for 3f204000.spi
[    8.230481] pinctrl-bcm2835 3f200000.gpio: pin-9 (3f204000.spi) status -22
[    8.230500] pinctrl-bcm2835 3f200000.gpio: could not request pin 9 (gpio9) from group gpio9  on device pinctrl-bcm2835
[    8.230516] spi-bcm2835 3f204000.spi: Error applying setting, reverse things back
[    8.230558] spi-bcm2835: probe of 3f204000.spi failed with error -22
```

Running something like `modprobe i2c_dev` (it's this driver that actually makes the device nodes for i2c available) might get you back the `/dev/i2c-XX` node of the i2c-gpio instance HyperPixel uses for touch.

justSem, 2022-01-06:
Yes, touch still works.
I was actually going for the i2c-3 interface as the Hyperpixel provides a breakout for that, but even that doesn't show up.
(Also: enabling the vc4-fkms-v3d breaks the hyperpixel for me, so it's just left disabled).
So my actual aim here is to be able to access that alternate bus to enable communication with a secondary microcontroller.

justSem, 2022-01-06:
To clarify this some more:
I first went looking into the loaded kernel modules, as this more often takes a role in 'missing' interfaces.
This revealed 4 loaded modules containing the string "i2c"

```
sem@dispi:~ $ lsmod | grep i2c
i2c_bcm2835            16384  0
i2c_gpio               16384  0
i2c_algo_bit           16384  1 i2c_gpio
i2c_dev                20480  0
```

Giving the fact that the second and the third weren't present on another Pi I tried disabling it to see what it would change. For me it didn't change anything in either device listing or display functionality, so I went looking further.

So basically I'm trying to get `i2c-3` to work as described here: [https://learn.pimoroni.com/article/getting-started-with-hyperpixel-4#using-the-alternate-i2c-interface-on-hyperpixel-4-0-for-advanced-users](https://learn.pimoroni.com/article/getting-started-with-hyperpixel-4#using-the-alternate-i2c-interface-on-hyperpixel-4-0-for-advanced-users )

rcarmo, 2022-01-29:
I'm interested in this since I would love to get an audio SHIM working... USB is possible, but would take up more space in my case.

Gadgetoid, 2022-01-31:
@justSem sorry for the slow response. Have you tried removing `dtoverlay=w1-gpio` since *something* is causing a pin conflict on BCM 9. I don't know how or why that would break the i2c bus getting mapped in /dev but device tree is weird and I wont pretend to fully understand it :grimacing: 

@rcarmo i2s and i2c are different but incredibly confusingly named buses. i2s conflicts with the DPI bus used for HyperPixel and thus wont work. See HyperPixel pinout: https://pinout.xyz/pinout/hyperpixel4 vs PCM/i2s pinout: https://pinout.xyz/pinout/pcm

justSem, 2022-02-01:
@Gadgetoid Yes, that was one of my attempts to see if there was anything like that bugging it.
I even tried a completely fresh bullseye install with only the Hyperpixel stuff installed, but that yielded nothing.

closed 2026-08-25 by Gadgetoid