pimoroni/hyperpixel4 #165: Hyperpixel4 does not work on freshly flashed Rpi OS

klundry, 2022-02-11:
I've made three attempts to flash a fresh image of RPi OS, headless and desktop. I've run updates, rebooted. Ran the hyperpixel4 install script and gotten no errors. Rebooted and...get nothing on the hyperpixel. 

Seems to me this has to be a bug since I've done absolutely nothing to the install other boot it up and run the install script. Maybe something that has changed in the most recent Pi OS images?

PS. I have used this hyperpixel display successfully in the past so the hardware is fine.

rizese, 2022-02-12:
is it the latest (Bullseye) OS? It says on the product page that Hyperpixel is not yet compatible with Raspberry Pi OS Bullseye. You should have success with a freshly flashed version of Buster

klundry, 2022-02-12:
Yeah, I tried the legacy buster image and it is working now. Is supporting bullseye in the works or are hyperpixel owners stuck with buster forever?

Gadgetoid, 2022-02-13:
It’s in the works - https://github.com/raspberrypi/linux/pull/4812

It’s a total paradigm shift from the old way it was accomplished on Raspberry Pi so it’s been a bit of a misadventure. Thankfully we’ve had rather a lot of help from the experts.

thirdr, 2025-01-07:
Closed as stale