pimoroni/hyperpixel4 #167: black screen

alexanderspacex, 2022-02-13:
hello 
i have problems installing hyperpixel 4.0 on raspberry pi 3B+.
I follow the instructions of: https://www.youtube.com/watch?v=gYP2j-dUrBY&t=736s&ab_channel=TeachingTech
nothing comes up on my screen just the background light.
I would use it as clipper screen for 3d printer.

alexander

alexanderspacex, 2022-02-13:
pi@ratos:~/hyperpixel4 $ curl -sSL https://raw.githubusercontent.com/pimoroni/hyperpixel4/master/hyperpixel4-debug.sh | bash

## Platform Information

Revision        : a020d3
Description:    Raspbian GNU/Linux 10 (buster)
5.10.63-v7+

## Touchscreen logs

Rectangular: Goodix
Square: ft5

## I2C Devices and Mappings

ls: cannot access '/dev/i2c-*': No such file or directory

bash: line 24: i2cdetect: command not found

## Boot Config

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
# enable raspicam
start_x=1
gpu_mem=128

#MainsailOS Specific
#Enables UART Communication by default
enable_uart=1
dtoverlay=disable-bt

[pi0w]
#pi0(w) Specific
enable_uart=1
dtoverlay=pi3-disable-bt

dtoverlay=hyperpixel4
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
enable_dpi_lcd=1
display_default_lcd=1
display_rotate=0
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
hdmi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6

Gadgetoid, 2022-02-14:
Is this HyperPixel 4 Rectangular? Directly connected to your Pi?

Does manually running `hyperpixel4-init` kick it into life?

What do you see if you:

```
sudo modprobe i2c-dev
ls /dev/i2c-*
```

Then run `i2cdetect` on whatever device shows up, eg:

```
sudo i2cdetect -y 22
``

Also output of the following can be helpful:

```
dmesg | grep Goodix
dmesg | grep i2c
```

BryceAshey, 2022-02-15:
to add to this - I'm seeing the same behavior - does running 64-bit raspbian make a difference?

alexanderspacex, 2022-02-16:
> to add to this - I'm seeing the same behavior - does running 64-bit raspbian make a difference?
Hello Bryce,
thanks for your help!
64 bit raspbian installed. Tried installing hyperpixel 4.0 rectangular again. but still doesn't work. stays the same just the background light is on.

Gadgetoid, 2022-02-16:
@alexanderspacex it doesn't look like you have any drivers set up for the display at all. I wonder if that `[pi0w]` header in `/boot/config.txt` is breaking things.

Try changing the last lines around to:

```
[pi0w]
#pi0(w) Specific
enable_uart=1
dtoverlay=pi3-disable-bt

[all]
#dtoverlay=vc4-fkms-v3d
# enable raspicam
start_x=1
gpu_mem=128

#MainsailOS Specific
#Enables UART Communication by default
enable_uart=1
dtoverlay=disable-bt

dtoverlay=hyperpixel4
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
enable_dpi_lcd=1
display_default_lcd=1
display_rotate=0
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
hdmi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 
```

Support in Bullseye / Bullseye 64bit is currently spotty since the way Pi supports/works with DPI displays changed quite drastically. Fixing that is a work in progress.

alexanderspacex, 2022-02-16:
@Gadgetoid 
thanks for your reply.

just changes.
this also makes no difference.

nielsaust, 2022-02-16:
I think I have the same issue, or related. I have been using the screen successfully fir wuite a while, but now it's all black (with backlighting) all of the sudden. I do however see that it is not faulty because the Pi OS boot-up screen does show for a fraction of a second, but then reverts to black. I re-installed Pi OS from scratch and the Hyperpixel software with the same result. I'll try some more over the weekend but any feedback is welcome.

Gadgetoid, 2022-02-17:
@alexanderspacex is there a `hyperpixel4.dtbo` in `/boot/overlays`?

@nielsaust trying some of the debugging steps above should help. Are you using Pi OS lite or full? Buster or Bullseye?

alexanderspacex, 2022-02-17:
@Gadgetoid I really appreciate your help!
I don't find hyperpixel4.dtbo in /boot/overlays.
Greetings

BryceAshey, 2022-02-26:
Following up on this thread - for now I downgraded to an old version of [Rasbpian Buster](https://downloads.raspberrypi.org/raspbian/images/raspbian-2020-02-14/). After running ```sudo apt update && sudo apt ugprade -y``` (installs a ton of updates of course) and then running the [hyperpixel install](https://learn.pimoroni.com/article/getting-started-with-hyperpixel-4), the display came right up. So definitely something with the newer versions of Raspbian.

nielsaust, 2022-02-27:
> Following up on this thread - for now I downgraded to an old version of [Rasbpian Buster](https://downloads.raspberrypi.org/raspbian/images/raspbian-2020-02-14/). After running ```sudo apt update && [...]

Thanks for mentioning this. Will try so as well.

raynt, 2022-03-10:
> Thanks for mentioning this. Will try so as well.

I just purchased a hyperpixel (square) and have the issue of blank screen with a pi4b. Tried downgrading using `sudo apt update && sudo apt ugprade -y` but still just a backlit screen with no output.

BryceAshey, 2022-03-10:
@raynt make sure you are starting with the operating system I linked first. Then run the commands I listed.

raynt, 2022-03-10:
Thanks for your reply @BryceAshey. I am a noob with Raspberry sorry for the basic question, but would I use etcher to overwrite the files onto the memory card to start with Buster? My setup came with a preinstalled os, so I want to make sure I'm doing this in a proper order.

BryceAshey, 2022-03-10:
@raynt no worries. Correct, etcher would work well. Raspberry also has a great imager available now: https://www.raspberrypi.com/news/raspberry-pi-imager-imaging-utility/

One thing to be clear on, this will wipe the drive clean and install a new OS. So backup anything you want to keep.

Good luck!

closed 2026-08-25 by Gadgetoid