pimoroni/hyperpixel4 #172: Which branch to use for a Pi Zero 2 W?

evprkr, 2022-03-07:
Hi, it's a bit unclear which branch to select in the script when trying to set up a Pi Zero 2 W, using Debian Buster.

I first tried Pi 4 but that didn't appear to work, I'm trying others now. Posting this issue so perhaps it will be added to the documentation or another option for Zero/Zero 2 will be added.

Thanks.

evprkr, 2022-03-07:
Since I haven't yet been able to even confirm that the screen I got works, I just plugged in a Pi 4 and tried the same SD card without changing anything and it's working with the Pi 4, so this must be an issue with the Zero 2's compatibility. I know my Zero 2 works because I am able to SSH into it fine and I can plug it in and use it through HDMI as well.

I'll see if I can find a Zero and test that as well, please update documentation with either proper instructions for the Zero/Zero 2 or a note that they are unsupported!

TheDodger, 2022-03-07:
I think you need to use the branch for Pi 3 etc., the Pi 4 is somehow different when it comes to graphics. If I remember correctly.

evprkr, 2022-03-07:
> I think you need to use the branch for Pi 3 etc., the Pi 4 is somehow different when it comes to graphics. If I remember correctly.

Just for kicks I decided to flash a brand new buster image onto the SD card using the official Pi imaging software, ran the script selecting Pi 3 and 2021 again, and still nothing but a black screen.

evprkr, 2022-03-08:
I had the Pi on upside down. It wasn't clear with the instructions on the website which way it should sit, and I guessed what I thought was the logical way. Please update the documentation!

6q6, 2022-03-09:
I'm guessing this has more to do with connections either shorting or the display not being attached properly, "the way it should sit' isn't a factor into play since there is 360 degree support.

choiway, 2022-03-15:
@evprkr Did you get it to work with Buster while selecting 3b & older when installing the software?

evprkr, 2022-03-15:
@choiway Yes,  but only after flipping it over so that the bottom of the pi was face up. However, the colors were distorted, I believe I damaged the pi during multiple soldering and soldering attempts, though I havent yet gone back to troubleshoot

meridian640, 2022-03-27:
The screen works well on the PI zero 2.
I selected branch for Pi 3, but in the config.txt file you have to put in comment the line dtoverlay=vc4-kms-v3d otherwise the screen remains black

jeffThompson, 2022-04-30:
I can confirm the legacy installer works without any mods on an old RPi Zero 1.3. I selected "Pi 3 or older" and it had no problems and no need to mod `/boot/config.txt`

closed 2026-08-25 by Gadgetoid