pimoroni/hyperpixel4 #174: When using a Raspberry Pi Zero W Hyperpixel 4.0 Rectangular touchscreen display, the screen is completely black (as in it look like the display is off)

ArcaneShade, 2022-03-19:
**Describe the bug** 

The screen is completely black (as in it looks like its turned off)

**To Reproduce** 

Install through the one-line-installer on a Raspberry Pi Zero W either with Buster Legacy or Bullseye and select rectangular for Raspberry Pi 3B or older, then reboot.

**What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?** 

I installed from the one-line-installer.
 
**Your HyperPixel 4**

**Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel**

 I am using the PIM369 board

**1. Is it Square or Rectangular?** Rectangular
**2. Touch or non Touch?**  It is Touch
**3. Approximately when was it purchased**  December 13th 2021
**4. How is it connected to your Pi**  GPIO pins

**Your Raspberry Pi** 

**Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.** 

I Tried installing on both Bullseye and Buster, neither worked. On Bullseye the Hyperpixel4 display was black (As in it looked like it was turned off). However, even though it was black If I touched it with my finger the mouse cursor would move around on the HDMI monitor which I also had connected to the Raspberry Pi Zero W. When it was on Buster the HDMI monitor got stuck on the rainbow screen and the Hyperpixel4 display still stayed black (As in it looked like it was turned off).

**1. What model of Raspberry Pi are you using** 

I am using a Raspberry Pi Zero W

**2. Which OS release are you using:** 

I have tried on both Debian version: 10 (buster) released January 28th 2022 and Debian version: 11 (bullseye) released January 28th 2022.

**3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?** 

This is a fresh setup

markjhughes, 2022-03-25:
goto /boot/config.txt and comment out dtoverlay=vc4-fkms-V3D

sudo nano /boot/config.txt
find dtoverlay=vc4-fkms-V3D and add a # at the beginning

[ctrl]+[x] to exit and they [y] to save.

sudo reboot

and you should be good to go.

obizues, 2022-06-07:
@markjhughes , this works, but for those of us that also want to rotate the screen, what should we be doing?

I'm specifically running Octopi's image for 3D printing, and without
`dtoverlay=vc4-fkms-V3D`
I cannot rotate my screen 90 degrees.

karimbizid, 2022-06-17:
This indeed fixed it for me on a Pi 4 with PiOS.

feiticeir0, 2023-06-16:
Not working for me. 
Fresh Rasberry PI OS install, commented 
#dtoverlay=vc4-kms-v3d

NOT
dtoverlay=vc4-fkms-v3d

and still black screen

closed 2026-08-25 by Gadgetoid