pimoroni/hyperpixel4 #175: Raspberry Pi Buster esponsive after installing

WillHWThompson, 2022-04-05:
**PLEASE NOTE**

:warning: GitHub issues are for reporting problems with the software, tech support do not monitor these.
Our software team are smol and busy, and you will not receive a timely response.
If you're having a hardware issue with your Hyperpixel 4, you should contact support@pimoroni.com.

:warning: Hyperpixel 4 is only officially supported on Raspberry Pi OS. There are too many operating systems with too many quirks to begin to support them all. Issues relating to other operating systems will be closed.

**Describe the bug**

> I received my Hyperpixel 4 Square and tried to connect it to my Raspberry Pi 4. 
> Initially I has bullseye installed - the screen remained  black without any back-lighting whatsoever. 
> I tried the workaround editing commenting out `dtoverlay=vc4-kms-v3d` in `boot/config.txt` but this had no effect. 
> Next I tried imaging my micro sd card with Buster instead of Bullseye. I ran the curl command to download the software and immediately my monitor went black and I lost my vnc connection. 
> 
> I have since tried restarting the device and connecting via vnc and ssh as well as pinging it - no response at all from the device. 
> I tried again with a fresh install of Buster and experienced the exact same thing. 

**To Reproduce**

What install steps did you go through so far? Did you install from GitHub or use our one-line-installer?

>I used the one-line installer with a fresh installer of Buster on my Raspberry Pi 4. 

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular?

> - Square 

3. Touch or non Touch?

> -Non touch 

5. Approximately when was it purchased

> - March 2022

7. How is it connected to your Pi

> -via GPIO pins

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
2. Which OS release are you using: `lsb_release --description` and `uname -r`
3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular
* `dmesg | grep ft5` for HyperPixel 4.0" Square

And check i2c is working:

`ls /dev/i2c-*`

You should see something like:

```text
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-7
```

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

For example:

```text
pi@raspberrypi:~ $ i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Gadgetoid, 2022-04-11:
On the latest version of Raspberry Pi OS it should suffice just to add `dtoverlay=vc4-kms-dpi-hyperpixel4sq` to `/boot/config.txt` albeit we still need to figure out rotation.

Let me know if this works.

whmckinley, 2022-04-17:
I have tried every suggestion that I could find. MULTIPLE new builds, multiple verifications of config.txt
i2c test shows it at 22, dmesg | grep ft5 states power not found, display has backlight that's it. Getting nothing on RPI 3, RPI4, on RPI 400, I get like a white banding on right half of screen then nothing.

Seems like Neil Young says "  Piece of Crap "

WillHWThompson, 2022-05-01:
I took the advice [here ](https://github.com/pimoroni/hyperpixel4/issues/177) and flashed the new version of bullseye and added `dtoverlay=vc4-kms-dpi-hyperpixel4sq` to my `/boot/config.txt`. I rebooted and the screen remained totally black without any backlight, as if it was off. 

I also tried the legacy version of buster and installing via curling the install script - this did not work either. 

I found a debug script [here](https://github.com/pimoroni/hyperpixel4/blob/master/hyperpixel4-debug.sh below is the output of the script. I am concerned by the output of i2cdetect: `Error: Could not open file `/dev/i2c-2' or `/dev/i2c/2': No such file or directory' - Is something wrong with my i2c busses? 

The full output is here: 

## Platform Information

Revision        : a03115
Description:    Debian GNU/Linux 11 (bullseye)
5.15.32-v8+

## Touchscreen logs

Rectangular: Goodix
Square: ft5
[    6.067701] edt_ft5x06 22-0048: supply vcc not found, using dummy regulator
[    6.068206] edt_ft5x06 22-0048: supply iovcc not found, using dummy regulator
[    6.069636] edt_ft5x06 22-0048: touchscreen probe failed

## I2C Devices and Mappings

/dev/i2c-20
/dev/i2c-21
/dev/i2c-22
Error: Could not open file `/dev/i2c-2' or `/dev/i2c/2': No such file or directory

## Boot Config

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

dtoverlay=vc4-kms-dpi-hyperpixel4sq

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]

Now Starting Raspberry PI