pimoroni/hyperpixel4 #176: Hyperpixel 4 rectangular, touch not working

zero-IQ, 2022-04-08:
Installed a fresh install of rasbian 10 buster lite, then connected the hypepixel 4 rectangular touch and then installed the drivers online. Screen works but touch does not recognized, no touch response at all.

My Pi is model 3B Revision: a02082
Raspbian GNU/Linux 10 (buster). 5.10.103-v7+
Fresh OS setup

dmesg | grep Goodix give the following output:
[ 8.776115] Goodix-TS 11-005d: supply AVDD28 not found, using dummy regulator
[ 8.777005] Goodix-TS 11-005d: supply VDDIO not found, using dummy regulator
[ 8.779100] Goodix-TS 11-005d: ID 911, version: 1060
[ 8.809497] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-11/11-005d/input/input0

i2c is working:
/dev/i2c-11

i2cdetect -y 11 output:
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- UU -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

Gadgetoid, 2022-04-11:
On the latest version of Raspberry Pi OS it should suffice just to add `dtoverlay=vc4-kms-dpi-hyperpixel4sq` to `/boot/config.txt` albeit we still need to figure out rotation.

Gadgetoid, 2022-04-11:
To be *more* specific you should use the 64bit version of Raspberry Pi OS released on 

Also the following script on 64bit Raspberry Pi OS released on 2022-04-04 (or later).

This can currently be found in Raspberry Pi Imager under "Raspberry Pi OS (Other)"

[image]

The following script should allow you to rotate the display, preserving correct touch input mapping:

```bash
#!/bin/bash

UTILITY="hyperpixel4-rotate (Square)"
DEVICE="EP0110M09"
OVERLAY="vc4-kms-dpi-hyperpixel4sq"
XORG_TOUCH_CONF_FILE="/usr/share/X11/xorg.conf.d/88-hyperpixel4-square-touch.conf"
XORG_CONF_FILE="/usr/share/X11/xorg.conf.d/88-dsi-1-rotate.conf"

ORIENTATION=$1

function success() {
    echo -e "$(tput setaf 2)$1$(tput sgr0)"
}

function inform() {
    echo -e "$(tput setaf 6)$1$(tput sgr0)"
}

function warning() {
    echo -e "$(tput setaf 1)$1$(tput sgr0)"
}

function compatibility_check {
    grep -e "^dtoverlay=$OVERLAY.*" /boot/config.txt > /dev/null
    if [ $? -ne 0 ]; then
        warning "Rotation requires hardware support on the Pi 4 or Pi 400"
        warning "Ensure \"dtoverlay=$OVERLAY\" is enabled in /boot/config.txt"
        exit 1
    fi
}

function set_xorg_conf {
    inform "Rotating display $1";
    xrandr --output DPI-1 --rotate $1

    MATRIX="$2 $3 $4 $5 $6 $7 $8 $9 ${10}"

    inform "Setting libinput Calibration Matrix: 1 0 0 0 1 0";
    xinput set-prop "pointer:$DEVICE" "libinput Calibration Matrix" 1 0 0 0 1 0 0 0 1

    inform "Setting Coordinate Transformation Matrix: $MATRIX";
    xinput set-prop "pointer:$DEVICE" "Coordinate Transformation Matrix" $MATRIX

    inform "Saving xorg touch config to $XORG_TOUCH_CONF_FILE";
    sudo tee $XORG_TOUCH_CONF_FILE > /dev/null <<EOF
# Auto generated by $UTILITY, edit with care!

Section "InputClass"
    Identifier "HyperPixel4 Square EP0110M09"
    MatchProduct "EP0110M09"
    Option "CalibrationMatrix" "1 0 0 0 1 0 0 0 1"
    Option "TransformationMatrix" "$MATRIX"
EndSection
EOF

    inform "Saving xorg rotate config to $XORG_CONF_FILE";
    sudo tee $XORG_CONF_FILE > /dev/null <<EOF
# Auto generated by $UTILITY, edit with care!

Section "Monitor"
    Identifier "DPI-1"
    Option "Rotate" "$1"
EndSection
EOF
}

printf "HyperPixel 4: Square - Display/Touch Rotation\n"
printf "This rotate utility only works with the Raspberry Pi OS desktop version or X-based alternatives.\n\n"

compatibility_check

if [ "$DISPLAY" == "" ]; then
    inform "No DISPLAY variable set, trying :0.0"
    export DISPLAY=:0.0
fi

case $ORIENTATION in
    "right")
        set_xorg_conf $ORIENTATION 0 1 0 -1 0 1 0 0 1
        ;;
    "left")
        set_xorg_conf $ORIENTATION 0 -1 1 1 0 0 0 0 1
        ;;
    "inverted")
        set_xorg_conf $ORIENTATION -1 0 1 0 -1 1 0 0 1
        ;;
    "normal")
        set_xorg_conf $ORIENTATION 1 0 0 0 1 0 0 0 1
        ;;
    *)
        printf "Unsupported orientation: $ORIENTATION\n";
        printf "Try one of: left, right, normal, inverted\n";
        exit 1
esac
```

Gadgetoid, 2022-04-11:
Sorry meant to respond to the other issue. Working on Rectangular rotate script right now.

You should still be able to get the board running by adding  `dtoverlay=vc4-kms-dpi-hyperpixel4` to `/boot/config.txt`

Gadgetoid, 2022-04-11:
Sorry I also missed that you were using a Pi 3B. I don't actually know! It's been a long time since I touched a 3B. It *should* work, but it varies quite significantly from the 4 in how it's all handled.

What process did you use to get the drivers installed? And how are you testing touch input?

zero-IQ, 2022-04-11:
can this be the solution?
https://github.com/pimoroni/hyperpixel4touchscreen/tree/master/driver

Gadgetoid, 2022-04-12:
It might help, but things are supposed to be simpler and not more complicated :grimacing:

The main problem with touch is that it's not explicitly scaled to the same size as the display. The touch panel might output values along X and Y from 0-65535 and the software - be that Xorg or a user application - must scale it to match. Normally there are some driver values to at least get *somewhere* with this.

The alternate driver is at least worth a try.

I'll see if I can find a Pi 3B and get anywhere with it.

Gadgetoid, 2022-04-12:
Okay I just put my Pi 4 SD card into my Pi 3B and it works.

You should try the latest OS and the instructions here - https://github.com/pimoroni/hyperpixel4/issues/177

Everything else is outdated, broken and overcomplicated. Things like 3D printer control apps are going to get dragged kicking and screaming into this one true way.

closed 2026-08-25 by Gadgetoid