pimoroni/hyperpixel4 #177: PSA: HyperPixel 4 (Square & Rectangular) on Raspberry Pi OS 64bit 2022-04-04

Gadgetoid, 2022-04-11:
Pi 4 users running the latest and greatest Raspberry Pi OS should forego our installer.

You need only one line in `/boot/config.txt`:

* Rectangular: `dtoverlay=vc4-kms-dpi-hyperpixel4`
* Square: `dtoverlay=vc4-kms-dpi-hyperpixel4sq`

Rotating the display via "Screen Configuration" ("arandr") results in inverted touch input. You can use the below bash script to rotate either display and persist settings into Xorg configuration files.

Supply the `--square` argument last if you're using a Square display. I'll aim to make this auto-detect before we roll it out.

:warning: Note: As usual rotation is *only supported within the desktop environment since Pi 4 has no low-level rotate support for hardware accelerated DSI displays*

```bash
#!/bin/bash

UTILITY="hyperpixel4-rotate (Rectangular)"
XORG_TOUCH_CONF_FILE="/usr/share/X11/xorg.conf.d/88-hyperpixel4-touch.conf"
XORG_CONF_FILE="/usr/share/X11/xorg.conf.d/88-dsi-1-rotate.conf"

ORIENTATION=$1
BOARD=$2

if [ "$BOARD" == "--square" ]; then
    DEVICE="EP0110M09"
    OVERLAY="vc4-kms-dpi-hyperpixel4sq"
else
    DEVICE="Goodix Capacitive TouchScreen"
    OVERLAY="vc4-kms-dpi-hyperpixel4"
fi

function success() {
    echo -e "$(tput setaf 2)$1$(tput sgr0)"
}

function inform() {
    echo -e "$(tput setaf 6)$1$(tput sgr0)"
}

function warning() {
    echo -e "$(tput setaf 1)$1$(tput sgr0)"
}

function compatibility_check {
    grep -e "^dtoverlay=$OVERLAY.*" /boot/config.txt > /dev/null
    if [ $? -ne 0 ]; then
        warning "Rotation requires hardware support on the Pi 4 or Pi 400"
        warning "Ensure \"dtoverlay=$OVERLAY\" is enabled in /boot/config.txt"
        exit 1
    fi
}

function set_xorg_conf {
    if [ "$DISPLAY" == "" ]; then
        inform "No DISPLAY variable set, trying :0.0"
        export DISPLAY=:0.0
    fi

    inform "Rotating display $1";
    xrandr --output DPI-1 --rotate $1

    MATRIX="$2 $3 $4 $5 $6 $7 $8 $9 ${10}"

    inform "Setting libinput Calibration Matrix: 1 0 0 0 1 0";
    xinput set-prop "pointer:$DEVICE" "libinput Calibration Matrix" 1 0 0 0 1 0 0 0 1

    inform "Setting Coordinate Transformation Matrix: $MATRIX";
    xinput set-prop "pointer:$DEVICE" "Coordinate Transformation Matrix" $MATRIX

    inform "Saving xorg touch config to $XORG_TOUCH_CONF_FILE";
    sudo tee $XORG_TOUCH_CONF_FILE > /dev/null <<EOF
# Auto generated by $UTILITY, edit with care!

Section "InputClass"
    Identifier "HyperPixel4 $DEVICE"
    MatchProduct "$DEVICE"
    Option "CalibrationMatrix" "1 0 0 0 1 0 0 0 1"
    Option "TransformationMatrix" "$MATRIX"
EndSection
EOF

    inform "Saving xorg rotate config to $XORG_CONF_FILE";
    sudo tee $XORG_CONF_FILE > /dev/null <<EOF
# Auto generated by $UTILITY, edit with care!

Section "Monitor"
    Identifier "DPI-1"
    Option "Rotate" "$1"
EndSection
EOF
}

printf "HyperPixel 4:  Display/Touch Rotation\n"
printf "This rotate utility only works with the Raspberry Pi OS desktop version or X-based alternatives.\n\n"

compatibility_check

case $ORIENTATION in
    "right")
        set_xorg_conf $ORIENTATION 0 1 0 -1 0 1 0 0 1
        ;;
    "left")
        set_xorg_conf $ORIENTATION 0 -1 1 1 0 0 0 0 1
        ;;
    "inverted")
        set_xorg_conf $ORIENTATION -1 0 1 0 -1 1 0 0 1
        ;;
    "normal")
        set_xorg_conf $ORIENTATION 1 0 0 0 1 0 0 0 1
        ;;
    *)
        printf "Unsupported orientation: $ORIENTATION\n\n";
        printf "Usage:\n"
        printf "    Rectangular: $0 <orientation>\n"
        printf "    Square:      $0 <orientation> --square\n\n"
        printf "Where orientation is one of: left, right, normal, inverted.\n"
        exit 1
esac
```

e28eta, 2022-04-13:
This is almost working for me: it only works in the non-rotated orientation on boot.

I bought my HyperPixel 4 Rectangular a month ago to use with a Pi 400 & Adafruit’s Cyberdeck HAT. I removed the diode on the HAT, and then ran into a black screen (backlight but no picture), and found some of the issues linked above. I was really happy to see this update.

With *only* `dtoverlay=vc4-kms-dpi-hyperpixel4` in `/boot/config.txt` it’ll boot and display, albeit rotated 90° clockwise. This is working well, as far as I can tell.

If I try to use either `,rotate=270` appended to the `dtoverlay` line **or** the separate `dtparam` line that includes the touchscreen options, the display doesn’t work.

If I remove the rotation settings from `/boot/config.txt` and try to use the embedded script (xrandr, etc), it works on that particular session. When I reboot, it breaks.

The failure looks like either a black screen, or a screen where one pixel (the first??) in each row is repeated all the way across the screen (the latter seems to be more common, but I’m not sure). I haven’t been able to determine when each behavior is triggered.

I’ve been trying both with an HDMI monitor connected, and also with *only* the hyperpixel (aka DP1) and changing config over ssh. There may be behavior differences, but neither setup is working correctly.

Is there helpful information I can gather, either for this issue or to file a new one? Sometimes, but not always, there’s a register + call trace dump visible inside of `dmesg` output, related to the `goodix_ts_driver`. I’ve captured several sets of dmesg logs, but didn’t keep any notes on what the particulars of each were 😞

Gadgetoid, 2022-04-13:
Could you paste the lines you have in your config.txt when you're trying to get rotation work? Would help me to recreate.

Gadgetoid, 2022-04-13:
Weird, same config and same SD card in my Pi 3B+ versus my Pi 4 results in two different rotations...

:scream:

e28eta, 2022-04-15:
I've tried three different `/boot/config.txt` variations. [Full contents](https://github.com/pimoroni/hyperpixel4/files/8492897/boot-config.txt), summary of changes from default:

```
dtoverlay=vc4-kms-dpi-hyperpixel4
```

```
dtoverlay=vc4-kms-dpi-hyperpixel4,rotate=270
```

```
dtoverlay=vc4-kms-dpi-hyperpixel4
dtparam=rotate=270,touchscreen-swapped-x-y,touchscreen-inverted-x
```

I have an (occasionally) working setup with the first one (no parameters to the overlay), and having run the bash script you provided :tada: After having ran that bash script, I haven't deleted / reset those settings to go back to a `dtoverly` + `dtparam` *only* solution.

----

Unfortunately, sometimes it fails. I thought the failure was deterministic, but I don't think that's true. I've seen three different failure modes:

* A black screen. The backlight is on and the OS knows a display is connected, but nothing is being drawn.
* One pixel repeated all the way across the screen, maybe skipping every other column? Here's a brief video, in which I move the mouse, and then touch & drag (and the selection rectangle it should be displaying also causes colors to change):

https://user-images.githubusercontent.com/483223/163491202-71712123-d062-48cb-a151-4d862a1ac00c.mov

* every other (I think?) column is skipped, and it throws off the locations of touches:

https://user-images.githubusercontent.com/483223/163491325-d370a209-8a0c-4119-8b9e-71c9281c9d7b.mov

* initially I was *also* trying to get multi-monitor to work (mostly so I had an attached display I could use while fiddling with the hyperpixel4). I observed touches on the hyperpixel being interpreted as happening within the bounds of the HDMI display

----

I have seen all three display failure modes in each of these cases:

- on a cold boot (as usb-c cable initially plugged in), a software-based reboot (command line or GUI), and a boot using the pi400 power key (after a software-based shutdown command)
- with multiple monitors, with the hyperpixel as the primary and not the primary (configured through `arandr` since I don't know `xrandr` well).
- with only the hyperpixel plugged in (I've done most of my testing like this)

I haven't figured if there's a configuration/sequence that always fails, nor do I know if one is more likely to succeed. ex: I thought cold boot was pretty reliable yesterday, but so far today it's failed 3 out of 4 times. Similarly, my initial assertion that it only worked non-rotated on cold boot could have just been random good luck across 3-4 attempts.

----

Here's an example of the `Goodix-TS` failure logs from `dmesg`: [dmesg-goodix-failure.txt](https://github.com/pimoroni/hyperpixel4/files/8492837/dmesg-goodix-failure.txt)
 
My guess is that the `TS` is for touch screen. I don't know if this is a red herring / unrelated. It's not always there when the display fails to work. 

----

Late additions:

* the screen is inverted in the horizontal direction:
[image]
* inverted with some columns missing:
[image]

* looks like every other column is white instead of black?
[image]

----
environment:
Pi 400

```bash
$ uname -a
Linux pi400 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```

`apt` says it's fully up to date, and as far as I can remember the only significant thing I've really done with this device / SD card was run the old hyperpixel4 script. I'd be willing to wipe & re-image if you want.

Gadgetoid, 2022-04-15:
There's a good chance the old hyperpixel script will break this in ways I can't imagine, but it should suffice to run:

```
sudo systemctl disable hyperpixel4-init.service
```

It could easily be this init script half-working that's messing up the display.

Also make sure nothing from the legacy installer is in config.txt.

A wipe and re-image is probably the safest bet though, the old installer is fundamentally a bunch of nasty hacks based around the old config.txt way of setting up DPI.

e28eta, 2022-04-15:
> There's a good chance the old hyperpixel script will break this in ways I can't imagine, but it should suffice to run:
>
> ```
> sudo systemctl disable hyperpixel4-init.service
> ```
>
> It could easily be [...]

That seems to have done it 🎉 🎊 

I feel dumb 😞  I'd previously found and reverted the `/boot/config.txt` changes, and looked through the installer script for other config files that might have been changed, but didn't find the service to disable.

Thanks so much for the help!

cubismod, 2022-04-18:
Howdy so I just installed this on a fresh OS install (April 4th 2022 64 bit on a Pi 4) and I get a blank white display. Using the square display.

cubismod, 2022-04-18:
Nevermind! My issue was that I had removed `dtoverlay=vc4-kms-v3d` which caused the display to not work correctly.

Gadgetoid, 2022-04-19:
> Nevermind! My issue was that I had removed `dtoverlay=vc4-kms-v3d` which caused the display to not work correctly.

No worries. Useful to know exactly how this fails for future reference!

sunsmoke, 2022-04-21:
Should this work with the 2020-and-earlier version of the hyperpixel4sq on a Pi3b+? I tried both the 64bit and 32bit builds of Bullseye, released 2022-04-04. I get what appears to be vertical stripes in both cases. It doesn't seem to matter if hdmi is connected or not. If hdmi is connected, then I get a very low resolution desktop on hdmi. Both installs are fresh, plain installs, and the only change I made was to add "dtoverlay=vc4-kms-dpi-hyperpixel4sq" to the end of /boot/config.txt

Gadgetoid, 2022-04-21:
> Should this work with the 2020-and-earlier version of the hyperpixel4sq on a Pi3b+? I tried both the 64bit and 32bit builds of Bullseye, released 2022-04-04. I get what appears to be vertical [...]

Youch- that's a good question, to which the answer is: no. They require - iirc - very different configurations and bringup. That's something I'm going to have to look into. I had forgotten there were two flavours of Square :grimacing:

sunsmoke, 2022-04-21:
Thanks! If you would like me to test anything, let me know.

brotherlogic, 2022-04-24:
This new version does not work on my Pi Zero 2 W, the screen is just blank - my dtoverlay is:

dtoverlay=rpi-poe,vc4-kms-dpi-hyperpixel4sq

confirmed that the screen works fine on the prior install (with the script config). Is there anything else that I need to configure?

tl8roy, 2022-04-28:
I have done a clean SD card wipe and reinstall on a Pi 400 and I am now getting a blank screen even after adding the new dto in the config.txt.

It was working on my existing installation which had the legacy drivers installed previously. I disabled the legacy service, but I also removed the old hyperpixel.dto. Once I did that and rebooted, the display stopped working. At that point I flashed the new OS to see if that would work.

This is a completely vanilla install with nothing done other than an apt update.

```
sudo cat /boot/config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2
dtoverlay=vc4-kms-dpi-hyperpixel4
#,rotate=270
#,touchscreen-swapped-x-y,touchscreen-inverted-x

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
```

tl8roy, 2022-04-29:
Top Gear Top Tip: i2c and spi both need to be commented out/disabled.

LazLong, 2022-05-03:
Annoyingly, I had rotation working on my Pi4 with the rectangular screen. Filesystem died and I rebuilt and now it doesn't work. The difference being that I'm now running Raspian OS Lite. Do we need X to have the rotation work?

Jeroennl, 2022-05-10:
For some reason I can't get chromium in kiosk mode to display properly (landscape), boot and without chromium is fine. I'm using a PI Zero 2 W and a rectangle screen. My config.txt is attached. Any ideas what i'm missing?
[config.txt](https://github.com/pimoroni/hyperpixel4/files/8661394/config.txt)

[image]
[image]
[config.txt](https://github.com/pimoroni/hyperpixel4/files/8661453/config.txt)

Gadgetoid, 2022-05-10:
> Annoyingly, I had rotation working on my Pi4 with the rectangular screen. Filesystem died and I rebuilt and now it doesn't work. The difference being that I'm now running Raspian OS Lite. Do we need X to have the rotation work?

@LazLong Yes. Or your application must support rotation somehow. This is a frustrating consequence of RPi's move to upstream blessed methods of supporting DPI displays. You can fall back to legacy drivers and the config.txt method, but that has its own tradeoffs.

@Jeroennl as above, there's no hardware rotation in the new drivers and no config.txt setting will affect display rotation. If Kiosk mode chromium uses X under the hood you should be looking at xrandr and xorg config. See the script in my original post.

Jeroennl, 2022-05-10:
> @Jeroennl as above, there's no hardware rotation in the new drivers and no config.txt setting will affect display rotation. If Kiosk mode chromium uses X under the hood you should be looking at xrandr and xorg config. See the script in my original post.

Fixed it using xrandr, thnx!

Footleg, 2022-05-13:
> > Top Gear Top Tip: i2c and spi both need to be commented out/disabled.
> 
> Ooof. I did not spot that. Thanks for reporting it back here!
> 
> I tend to fall into the false assumption that - like the old config.txt magic setup method did - the DPI display will trump anything else, but with device-tree it'll throw a pin conflict instead. 😬

Please add this to the top of this PSA post. I just lost 2 hours trying to find out why my display apparently stopped working when I applied a rotate setting. But it was because I had also enabled i2c for the same reboot while doing initial set up from the refresh OS image. This might also be why the display did not appear to work on my original CM4 hardware stack.

Gadgetoid, 2022-05-13:
@Footleg story of my life! Duly noted and warnings added.

Footleg, 2022-05-13:
I can now confirm my 2019 era Hyperpixel4 rectangular with touch is working on my CM4 mounted on a Waveshare carrier board. I discovered the i2c bus mapped to 22, rather than 3 as told here https://learn.pimoroni.com/article/getting-started-with-hyperpixel-4#using-the-alternate-i2c-interface-on-hyperpixel-4-0-for-advanced-users so maybe a quick update to that article to mention you can see which bus numbers are active using the command ls /dev/i2c-* (as I learned from this comment: https://github.com/pimoroni/hyperpixel4/issues/122#issue-789137822 ).

Note that when rotating the display using the console, the touch does not map to the rotated display. It also does not map to the correct display when a second display is attached to the HDMI output either. I've not looking into options to deal with that yet, but at least I am up and running now.

hothsolo, 2022-06-07:
After doing the instructions exactly as described here and looking thru all the comments on everything that could be wrong, I still cannot get this working.  I get a black screen.  Nothing else.  Any ideas?

locki-cz, 2022-06-07:
Hello, I spend few days tryed everything, and I wasnt able to make it working again, just reinstalled my some old Octoprint with octodash and want to move to mainsail klipper, also tryed all 32bit and 64bit version of great awesome Raspberry Pi OS 64bit 2022-04-04 bullseye and nothing, i can get some picture when i set dtoverlay=vc4-kms-dpi-hyperpixel4sq or some weird desktop picture after display wakeup after sleep ...  just rpi console then only ugly lines :( 

all what i have:

- hyperpixel4
- rpi 3b
- rpi 4b with 4gb ram
- 32gb sandisk extreme card 
- original power supply Raspberry Pi USB-C 5,1V3A

I will try all to test what is wrong! Because it working before ... :(

djbetterly, 2022-06-09:
I'm having zero luck with rotation in any direction, I've disable I2C, SPI, edited the config.txt according to the notes above and zero luck.  Has anyone worked it out?

Footleg, 2022-06-10:
> I'm having zero luck with rotation in any direction, I've disable I2C, SPI, edited the config.txt according to the notes above and zero luck. Has anyone worked it out?

This worked for me as per the instructions at the top of this page with a fresh Raspberry Pi OS image (Bullseye). You do need the latest (use apt to upgrade everything if on an older Bullseye release). Check you are on Bullseye (it lists the repos being used when you run sudo apt update).

J4CY, 2022-06-15:
Hi all, I'm new to the HyperPixel 4.0 and RPI. I've got an RPI 4B w/ 2GB with Raspberry Pi OS 64bit bullseye installed.

Installed the screen and applied the following to `/boot/config.txt`.

```
dtoverlay=vc4-kms-dpi-hyperpixel4 
dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y
```

Rotation and touch are all working as expected, but I am noticing screen tearing of some sort on this specific side of the screen. I've tried other rotations and they all seem to have the same effect. Is this expected, a software bug, or potentially a hardware issue? I've attached some videos to demonstrate my issues.

https://user-images.githubusercontent.com/89369480/173922485-b24e53d6-9450-48c5-bdc5-5938b9fe68d0.mp4

https://user-images.githubusercontent.com/89369480/173923288-7383116a-883c-4140-829a-dcfadda60476.mp4

matthewnreid, 2022-06-18:
> I'm having zero luck with rotation in any direction, I've disable I2C, SPI, edited the config.txt according to the notes above and zero luck. Has anyone worked it out?

I was able to get this working on Pi 400 by moving the hyperpixel lines immediately after "dtoverlay=vc4-kms-v3d" when I had them above it wouldn't rotate properly after the desktop loaded. 

dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dpi-hyperpixel4
dtparam=rotate=270,touchscreen-swapped-x-y,touchscreen-inverted-x

matthewnreid, 2022-06-18:
> I'm having zero luck with rotation in any direction, I've disable I2C, SPI, edited the config.txt according to the notes above and zero luck.  Has anyone worked it out?

This worked for me as well booting into terminal. For the rotation to work in X, I had to rotate left.

rnbarrett, 2022-06-19:
I am looking for some pointers/help trying to get this working with OctoPi/OctoPrint/OctoDash on a Pi4 4GB (think headless), with a HyperPixel 4 Rectangular.  

I have tried the following with a fresh install of OctoPi (0.18.0, 2022-05-24), and also after a full update/reboot (apt).  No other software or configuration has been performed at this point, just the basic OS configured for my WiFi and my locale.  I can SSH into the box, and I can hit the OctoPrint server with a browser, but I have not gone through the OctoPrint setup wizard.

After installing the "legacy" drivers per the top of this thread (because the current install stops and points me to this thread), I can reboot my Pi and the display is working in portrait mode.  It's just a terminal interface, so I'm not sure if touch is working.

If I modify pretty much anything in config.txt, per the comments above, my HyperPixel stops working.  The backlighting is on, but no graphics or text.  SSH and browser (OctoPrint) access remains, so I know the box is still somewhat running.

After the legacy install, here's the contents that work after install, in portrait mode (this is the only part of config.txt that is not commented out, I am pretty sure):
`# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
# enable raspicam
start_x=1
gpu_mem=128

dtoverlay=hyperpixel4
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6`

If I comment out everything (and I mean everything), and only insert this line, the display does not work at all:

**Once installed you need only one line in /boot/config.txt:

Rectangular: dtoverlay=vc4-kms-dpi-hyperpixel4**

If I change the original config.txt so that everything remains, but I change **dtoverlay=hyperpixel4** to **dtoverlay=vc4-kms-dpi-hyperpixel4**, the display does not work.

I'm a bit stuck.  I want to get the display rotated, and make sure that Touch is working with OctoDash (but I can't get to the OctoDash install, yet).

Thanks for any help...

djbetterly, 2022-06-20:
I followed this and had good luck...though the touch screen is still a bit finicky at times 
https://www.instructables.com/Rotate-Raspberry-Pi-Display-and-Touchscreen/

ETE-Design, 2022-07-03:
@nitz Seems like we have same problem, mine is just with KlipperScreen

When I add this to Config.txt:

`[all]
dtoverlay=vc4-kms-dpi-hyperpixel4
dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y`

it is looking correct in landscape on the Prompt, but when it start KlipperScreen it is in is start in Portrait mode but Touch in Landscape... Seems like it is not possible to tell other program to Rotate to Landscape other that Prompt :-(

osfa, 2022-07-12:
> will these bundled drivers work for 2020 and earlier boards? trying to install fresh from mentioned bullseye dist with a 2020 square board and setting `dtoverlay=vc4-kms-dpi-hyperpixel4sq` but only seeing white flickering and glitching. if I instead start fresh and then clone the repo and install via the square-pi4 branch I at least see the boot screen but then after that just black with a white halosque light in the middle. what is the recommended image and install process for a 2020 board on a rpi4? will bullseye work?

to answer my own question: for a 2020 or earlier board one seems to need to install the drivers via the square-pi4 branch and then disable `dtoverlay=vc4-kms-v3d` (and omit `dtoverlay=vc4-kms-dpi-hyperpixel4sq`)

Dianoga, 2022-07-13:
On a Pi 4 running the latest and greatest nightly OctoPi build and using the simplified `/boot/config.txt` settings from above, I'm getting a consistent screen flicker (see video). I've tried going back to the legacy drivers with the same effect. Any suggestions?

Relevant lines from `/boot/config.txt'

```
dtoverlay=vc4-kms-dpi-hyperpixel4
dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y
```

https://user-images.githubusercontent.com/1618956/178836782-b3ac6ffb-90db-42a0-b814-dd52353a0705.mp4

ChrisZ32, 2022-07-18:
**Pi Zero 2 running a freshly installed/updated Bullseye** - The new instructions aren't working for me fully for a new rectangular screen. It installs and displays, and the touchscreen works no problem in portrait and 180 rotated portrait - but 90 and 270 landscape rotations don't work. The boot screen(s) are correct, but the OS GUI remains in portrait. @Gadgetoid FYI as I thought I saw somewhere that you wanted some feedback on a Zero 2.

I ran the legacy installer for a Pi 4 rectangular and the landscape display worked after some tweaks (well technically I ran it for a Pi 3 first, then when that didn't work I ran for a Pi 4). I had to change in `/boot/config.txt`:
- Comment out existing line `dtoverlay=vc4-kms-v3d` (I believe this disables hardware acceleration or something?)
- Add `display_lcd_rotate=1` to rotate the display
- Add `dtparam=touchscreen-swapped-x-y,touchscreen-inverted-y` to rotate the touch input to match the display

This gives you landscape with header at the bottom, will need other combinations for other orientations. I needed both new settings - adding `rotate=270,` to the beginning of the `dtparam` didn't work by itself.

Hopefully this helps a few people struggling with this

jacyhung, 2022-07-21:
> When I add this to Config.txt:
>
> `[all] dtoverlay=vc4-kms-dpi-hyperpixel4 dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y`
>
> it is looking correct in landscape on the Prompt, but when [...]

@ETE-Design still need help figuring out rotation w/ KlipperScreen? I have it working.

In my `/boot/config.txt`,

```
dtoverlay=vc4-kms-dpi-hyperpixel4
dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y
```

Then create a new file,
```
sudo nano /usr/share/X11/xorg.conf.d/90-monitor.conf
```

with the following contents,

```
Section "Monitor"
    Identifier "DPI-1"
    Option "Rotate" "right"
EndSection
```

and restart KlipperScreen. 
[image]

jmacdonald, 2022-07-28:
> On a Pi 4 running the latest and greatest nightly OctoPi build and using the simplified /boot/config.txt settings from above, I'm getting a consistent screen flicker (see video). I've tried going back to the legacy drivers with the same effect. Any suggestions?

I'm seeing these as well. I'm running 2022-04-04 with all of the updates installed. I have _two_ HyperPixel 4 Square displays with two separate RPi Zero 2Ws and am seeing this on both. Here's my `/boot/config.txt`:

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Add HyperPixel support
dtoverlay=vc4-kms-dpi-hyperpixel4sq

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
```

jmacdonald, 2022-08-10:
It's worth noting that these issues were on the 32-bit version of the Bullseye release of Rasperry Pi OS ^. I had tried the 64-bit release, but oddly enough, memory consumption and general performance were so problematic as to make it unusable for my project (which involves Chromium on an RPi Zero 2W).

As a workaround, I've installed the legacy Buster-based version of Rasperry Pi OS alongside the legacy drivers. In this scenario, everything works perfectly; no flickering at all.

hoegge, 2022-08-20:
How do you make the original Hyperpixel display work on Bullseye?

jiki974, 2022-08-22:
> Youch- that's a good question, to which the answer is: no. They require - iirc - very different configurations and bringup. That's something I'm going to have to look into. I had forgotten there [...]

I ran into same situation with the hyperpixel 4 square 2020, any guideline to make it work on bullseye ?

jarrah31, 2022-09-01:
I have a Hyperpixel 4 rectangle bought in 2018. I'm attempting to get this working on a Raspberry Pi 3 B using Bullseye 64-bit 2022-04-04 build. Should this combination work?

The display does actually show the desktop, but the problem is I can't rotate the screen to landscape when trying each of the following lines:
```
dtoverlay=vc4-kms-dpi-hyperpixel4
# with one of the following below
dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y
dtparam=rotate=180
dtparam=rotate=270,touchscreen-swapped-x-y,touchscreen-inverted-x
```

mjrintam, 2022-09-05:
> In my `/boot/config.txt`,
>
> ```
> dtoverlay=vc4-kms-dpi-hyperpixel4
> dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y
> ```
>
> Then create a new file,
>
> ```
> sudo nano [...]

Finally! Thank you so much! Works now also on pi zero w with a few years old Hyperpixel 4.0 and fresh install of 32-bit Bullseye.

Bizzaro, 2022-09-14:
If anyone's confused like me, I just added `dtoverlay=vc4-kms-dpi-hyperpixel4sq` to `/boot/config.txt` and then ran the script with the desired orientation. 

I am using a pi zero and the square display.

Using `dtparam=rotate=180` would rotate the desktop, but not the mouse cursor.

omarandlorraine, 2022-09-15:
> In my `/boot/config.txt`,
>
> ```
> dtoverlay=vc4-kms-dpi-hyperpixel4
> dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y
> ```
>
> Then create a new file,
>
> ```
> sudo nano [...]

Do you know why you needed rotate settings both in xorg.conf.d and in /boot/config.txt? Is one for touch and the other for the framebuffer itself?

sesquipedality, 2022-09-16:
I am not using X so the screen rotation solutions provided here have not worked for me.  However, I have been able to get touchscreen rotation on the Pi 4 with a 2021 Hyperpixel 4.0 Square to work as follows.

In /boot/config.txt

```
dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dpi-hyperpixel4sq,rotate=180,touchscreen-swapped-x-y
```

In /etc/udev/rules.d/98-touchscreen.rules

```
ATTRS{name}=="generic ft5x06 (00)", ENV{LIBINPUT_CALIBRATION_MATRIX}="-1 0 1 0 -1 1"
ATTRS{name}=="EP0110M09", ENV{LIBINPUT_CALIBRATION_MATRIX}="-1 0 1 0 -1 1"
```

This is a bit of a black box to me, as I am not a udev or libinput expert, but as I understand it if the matrix is "a b 1 c d 1" then the touchscreen co-ordinates are multiplied by the matrix

```
(a b)
(c d)
```

to transform the co-ordinates.  Hope this helps someone.

Of course it would be much neater if the rotation parameters worked in /boot/config.txt so hopefully that will get fixed soon.

jeffcolabs1, 2023-02-25:
> **Pi Zero 2 running a freshly installed/updated Bullseye 64bit** - The new instructions aren't working for me fully for a new rectangular screen. It installs and displays, and the touchscreen works no problem in portrait and 180 rotated portrait - but 90 and 270 landscape rotations don't work. The boot screen(s) are correct, but the OS GUI remains in portrait. @Gadgetoid FYI as I thought I saw somewhere that you wanted some feedback on a Zero 2.
> 
> I ran the legacy installer for a Pi 4 rectangular and the landscape display worked after some tweaks (well technically I ran it for a Pi 3 first, then when that didn't work I ran for a Pi 4). I had to change in `/boot/config.txt`:
> 
> * Comment out existing line `dtoverlay=vc4-kms-v3d` (I believe this disables hardware acceleration or something?)
> * Add `display_lcd_rotate=1` to rotate the display
> * Add `dtparam=touchscreen-swapped-x-y,touchscreen-inverted-y` to rotate the touch input to match the display
> 
> This gives you landscape with header at the bottom, will need other combinations for other orientations. I needed both new settings - adding `rotate=270,` to the beginning of the `dtparam` didn't work by itself.
> 
> Hopefully this helps a few people struggling with this
> 
> EDIT: when I booted the pi again the next day the screen was blank - ` journalctl -u hyperpixel4-init.service` showed an error of `ImportError: No module named RPi.GPIO`, editing `sudo nano /usr/bin/hyperpixel4-init` to have `#!/usr/bin/env python3` instead of `#!/usr/bin/env python` fixed it. Just saying in case it helps someone

@ChrisZ32   Thank you for this.   This worked perfectly for me also.   New HyperPixel4, Zero 2W.   Landscape mode works perfectly in either rotation.   There is obviously an issue with the vc4-kms-v3d drivers and rotation.  Reverting back to the old drivers and with your settings worked perfectly.

r-colvin, 2023-03-14:
@jeffcolabs1 / @ChrisZ32 - adding my vote on this answer; brand new Pi Zero 2 WH with latest Raspberry OS (full - 32 bit - 2023-02-04) 

i had the same symptoms as you, and the same solution has worked

curl -sSL get.pimoroni.com/hyperpixel4-legacy | bash
-> select "0"
-> before rebooting, edit /boot/config.txt (in my case, I have the Pi Zero in landscape mode, with the USB ports on the bottom side)
--> comment out dtoverlay=vc4-kms-v3d
--> add display_lcd_rotate=3
--> <im still working out the correct touchscreen parameters : will update when they are working!>

Ruhel786, 2023-03-20:
I have had my Pi 4 4GB and my 2023 Rectangle working with no issues until I had an issue with PINN and wiped the whole card and started again. Now no matter what I do the touchscreen will not rotate to match the screen orientation.
I'm sorry I can't seem to find the OP who posted this so I can't credit it, but the below method had a 100% success rate for me until today and I don't know why it won't work anymore on my installs.

`cd /boot`

`sudo nano config.txt`

Comment the line below:
`**#dtoverlay=vc4-fkms-v3d**`
add the line below
`**display_lcd_rotate=3**`

Save and Exit the Config file.

Save and reboot and the screen will now be orientated with the USB C port at the bottom. 

The TOUCH function will still be incorrect, but to correct this:
`cd /usr/share/X11/xorg.conf.d`

`sudo nano 40-libinput.conf`

Edit the section for the touchscreen - it is normally the second last section - by adding this line:
```
Section "InputClass"
        Identifier "libinput touchscreen catchall"
        MatchIsTouchscreen "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
>>>>>  Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1"  <<<<<<<
EndSection
```
This was working with no issues and now on my Pi OS 64bit it just rotates the display only, not the touchscreen.
Any advice would be appreciated. 
Ruhel™

alucryd, 2023-03-26:
Doesn't seem to work on Arch Linux ARM (armv7, pi zero 2 w) with the new overlay. It was working fine with the old hyperpixel 4 overlay until my last system update, which led me to this page. Touchscreen is working but I have no picture, I am able to interact with klipperscreen but I don't see what I'm clicking on :/

Here's my full config.txt:
```
# See /boot/overlays/README for all available options

dtoverlay=vc4-kms-dpi-hyperpixel4
#dtoverlay=hyperpixel4
dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y
initramfs initramfs-linux.img followkernel

# Automatically load overlays for detected DSI displays
#display_auto_detect=1

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Automatically load overlays for detected cameras
#camera_auto_detect=1

# Uncomment to enable bluetooth
#dtparam=krnbt=on

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

gpu_mem=96
start_file=start_x.elf
fixup_file=fixup_x.dat
disable_camera_led=1

#overscan_left=0
#overscan_right=0
#overscan_top=0
#overscan_bottom=0
#enable_dpi_lcd=1
#display_default_lcd=1
#display_lcd_rotate=1
#dpi_group=2
#dpi_mode=87
#dpi_output_format=0x7f216
#dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```

Running the hyperpixel4-init service doesn't help either, although I understand it's no longer needed.

djammy, 2023-04-20:
Hi all!

Newbie here. Apologies if my question is not meant for this thread. 

I just got my raspberry pi 4 connected with the Hyperpixel 4.0 Square and I followed the instructions as above and the display is working fine. However, I would like to control the individual pixels on the Hyperpixel 4.0 display: for instance I would like to display the entire screen being green only or red, or white etc... Is there a way to do this using Python? If so, which library should I use?

Thanks.

Ruhel786, 2023-04-20:
Can I ask what it's for? Are you testing the pixels on the screen? If so just install the desktop OS and use a full screen image of wash colour.

sesquipedality, 2023-04-20:
Depends quite what you're trying to do.   

 * PyGame is one option for direct control of the screen.   
 * If you want a full window manager, PyQt. with EGLFS. 
 * I think kivy also supports Linux framebuffer if you're after something more modern/high level
 * You can also write to the raw framebuffer device, but that would require a particular kind of masochism.  
 * Needless to say, none of these are compatible with using X, and writing a full screen X11 application in PyQt is also an option.

djammy, 2023-04-20:
So I would like to be able to control the entire screen pixels to display just one colour (255,0,0)/(0,255,0)/(0,0,255) and any combination in between. It is for a photonics experiment. Not for testing the pixels.

I was thinking if I should just use Pillow and display a coloured image.

sesquipedality, 2023-04-20:
I would use pygame or pyqt for that.   Probably pygame as the learning curve is a little less steep.  With a little research you could probably go a little more lightweight and just write the relevant number of raw pixels direct to /dev/fb0, but it's not a very expandable approach if you want to do more later.

Gordon999, 2023-04-21:
with a fresh install of raspiOS 64bit after much trying using the instructions I finally got the hyperpixel4 to work on a Pi3B and rotated with.
add dtoverlay=vc4-kms-dpi-hyperpixel4 to /boot/config.txt, leave dtoverlay=vc4-kms-v3d there
then on the Pi 3B >> Preferences >> Screen Configuration >> DPI >> rotate option to suit, reboot

nmatsuda, 2023-07-07:
> Youch- that's a good question, to which the answer is: no. They require - iirc - very different configurations and bringup. That's something I'm going to have to look into. I had forgotten there [...]

Any update on this?

npimig, 2023-07-21:
So oddly some how I arrived at a somewhat working display that could not rotate to landscape orientation (header on bottom). I used the latest of Bullseye 5/4/2023 image, at first no luck with the `dtoverlay=vc4-kms-dpi-hyperpixel4` then I launched `raspi-config` and enabled GL drivers after reboot a screen turned on. Then trial error I then found to get the touch oriented correctly i had to insert the `dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y` stuff between the two `dtoverlays` (2nd one created after raspi-config `dtoverlay=vc4-kms-v3d`) in the `/boot/config.txt` as well as relying on the command `DISPLAY=:0 xrandr --output DPI-1 --rotate right` once desktop was finally loaded manually at first to orient the display and touchscreen in landscape mode.

To polish I asked ChatGPT (shame), who said to put that command in `~/.xsessionsrc` that needed to be created but loads after the primary boot process/around the initial display rendering time and (viola it works). Takes a while to boot and appears to lock out touch inputs until the "ole-gd-its-broke" period of it and then all is good, presto.  

I'm running on the Pi Zero 2 W with OpenGL 2 drivers enabled with the Hyperpixel 4" (Rectangular Display) with the Raspbian - Bullseye (Release 2023-05-03). 

Prior i was running Buster from 2021 with issues in the legacy driver for this display not allowing me to enable OpenGL features.

Just thought I'd share...

brucediesel, 2023-07-24:
Hi, is there a timeline as to when the Hyperpixel will be able to work with other interfaces enabled?  I have a project which uses a camera through the CSI interface, and a lidar on i2c but even just the camera and the Hyperpixel do not work together.  I am moving over to new libcamera, so I don't want to go back to older OS.

rudiprinsloo, 2023-08-12:
I have been using my HyperPixel Square without issue with the legacy drivers, old Raspbian OS on a Pi Zero W.

Then after upgrading to the latest Raspberry Pi OS (Bullseye) I was initially happy to find that I didnt have to install any drivers, and just needed to add the following line to my /boot/config.txt file:

dtoverlay=vc4-kms-dpi-hyperpixel4sq

After rebooting the screen would just show a bunch of lines running across the display and these would jump around when I touch it.

[image]

I2C and SPI are disabled.

I have read all the comments here but I have not been able to resolve the issue.

Any help would be much appreciated.

omarandlorraine, 2023-08-15:
> Hi, is there a timeline as to when the Hyperpixel will be able to work with other interfaces enabled? I have a project which uses a camera through the CSI interface, and a lidar on i2c but even just [...]

My understanding is it's not going to happen; the hyperpixel4 actually takes over the spi/i2c hardware and corresponding GPIO pins, which is the reason they must be disabled.

[46 more replies]

closed 2026-08-25 by Gadgetoid