pimoroni/hyperpixel4 #180: is not stable.

d4lv, 2022-05-18:
hello.. 
i am testing the screen 2 days now with a raspberry pi 4 and is weird, for now when i rr the raspberry pi the screen is just black every time i need to make a fresh install of the hyperpixel and again is 50/50 after the resrart to working..
i dont know why but is not stable is any fix for that??

i check all the reports and the suggestion but still i don't get it i shutdown the raspberry pi the screen works normal and now again is black :/

another problem is when the screen is close for inactivity and i open again the screen (with the mouse) is start bliking..

Gadgetoid, 2022-05-18:
I don't know if it will make a difference, but:

```
enable_dpi_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
dpi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```

should be removed from config.txt

And you should make sure `hyperpixel4-init` is disabled:

```
sudo systemctl disable hyperpixel4-init
```

And that i2c and SPI are disabled.

It's generally easier - when switching from old to new drivers - to image a fresh card and *only* add the `dtoverlay=vc4-kms-dpi-hyperpixel4` line. This avoids the chance of any old settings or scripts breaking it.

masaeedu, 2023-05-12:
@Gadgetoid Hi there. This is kinda besides the point, but I've experienced unreliable behavior in a clean setup where all three requirements are met (I sent a support ticket last Sunday):

- I2C and SPI are disabled
- hyperpixel4-init isn't installed (I never installed the legacy drivers and used only the dtoverlay approach)
- the aforementioned lines aren't present in /boot/config.txt

I was curious about the first requirement (I2C disabled). There's an I2C breakout on the display, how does one use that? Or is it that the DT overlay for hyperpixel just deals with the details for that I2C breakout internally, so `dtparam=i2c_arm=on` is redundant?

Gadgetoid, 2023-05-16:
The I2C breakout is on the same pins as the touchscreen, whereas `i2c_arm` sets up a bus on GPIO 2 and 3 which are display VSYNC and HSYNC (so we can't use those for i2c since they're absolutely fundamental to the (DPI) display-parallel interface).

The touch i2c bus is on GPIO 10 and 11 and can be access by loading the i2c-dev module: `sudo modprobe i2c-dev` and using the appropriate bus found by `ls /dev/i2c-*` (could be 22).

closed 2026-08-25 by Gadgetoid