pimoroni/hyperpixel4 #184: NOT WORKING ON RPI 3B+ despite the many "fixes"

jacobhjensen, 2022-06-19:
It shouldn't be this hard to get a ~50 GBP monitor to work... 

I have tried the solution mentioned in #177, but still not working. I do get an image though, however it's stuck in portrait mode. And the touch is working if I want to click anywhere else but where I want. So it's not a complete waste of money, as long as I don't expect it to work the way it should.. 👍 

I see that other people have a similar issue with no responses (#183 and #181 ) and no solutions, so here's to hoping the money goes to some sort of development 🥇

jacobhjensen, 2022-06-19:
For anyone using RPI 3B+ and just want it to work.... Just use this fix instead: https://www.youtube.com/watch?v=OJ59hXSyBoI&t=319s&ab_channel=ChrisRiley

helgibbons, 2022-06-20:
On a 3B+ (and on earlier Pi), I needed to rotate the display in 'Display Configuration' after adding the rotation line in `config.txt` - after I did that touch worked as expected.

Might also be worth a try for #181 ?

gaardhus, 2022-07-12:
I cannot make rotating my screen work either. I'm running the latest Raspberry Pi OS Lite and tried the follow configurations in `/boot/config.txt`

```
dtoverlay=vc4-kms-dpi-hyperpixel4,rotate=90
```
or 
```
dtoverlay=vc4-kms-dpi-hyperpixel4
# And one of 
dtparam=rotate=90
display_lcd_rotate=1
display_rotate=1
```
And nothing worked. I cannot figure out how to set 'Display Configuration' to normal in Raspberry Pi OS Lite CLI.

PrivacyPolicy404, 2023-01-17:
To fix the touchscreen rotation i tried asking chatgpt and it totally worked:

> The HyperPixel 4 is a small TFT display that can be used with a Raspberry Pi or other single-board computers. If you need to rotate the touch input on the HyperPixel 4, you will need to use the xinput command-line utility, which is typically pre-installed on most Linux distributions.
> 
> Here are the steps to rotate the touch input on the HyperPixel 4:
> 
> Connect the HyperPixel 4 to your Raspberry Pi or other single-board computer
> 
> Open a terminal window and type the command xinput list and press enter. This command will list all of the input devices connected to your system.
> 
> Look for the input device associated with the HyperPixel 4 touch screen. It should be listed as "FT5406 memory based driver" or similar.
> 
> Once you have located the input device, use the command xinput set-prop 'FT5406 memory based driver' 'Coordinate Transformation Matrix' 0 -1 1 1 0 0 0 0 1 to rotate the touch input 90 degrees clockwise.
> 
> If you want to rotate the input 180 or 270 degrees you can use the following commands respectively:
> xinput set-prop 'FT5406 memory based driver' 'Coordinate Transformation Matrix' -1 0 1 0 -1 1 0 0 1
> xinput set-prop 'FT5406 memory based driver' 'Coordinate Transformation Matrix' 0 1 0 -1 0 1 0 0 1
> 
> If you ever want to go back to normal orientation you can use xinput set-prop 'FT5406 memory based driver' 'Coordinate Transformation Matrix' 1 0 0 0 1 0 0 0 1
> 
> Note that the name of the input device may vary depending on your specific setup.

In my case I had to define the device as " pointer:'Goodix Capacitive TouchScreen' " and rotate 270 degrees for the issue to get fixed.

Hope it helps someone!

closed 2026-08-25 by Gadgetoid