pimoroni/hyperpixel4 #185: Hyperpixel 4 (Rectangular, Touchscreen) not working on Pi Zero W w/Bullseye

Two-Worlds, 2022-07-27:
My Hyperpixel 4 hasn't been used since I bought it in 2019 - been in its box ever since.
I did try it at the time with a Pi 3+ but got no display - didn't have time to resolve so put it to one side.
I've just set it up with a Pi Zero W and still cannot get a display. 
Backlight comes on, so power is getting to the board, but black screen is all I see.
I've set the dtoverlay as per: dtoverlay=vc4-kms-dpi-hyperpixel4
No luck whatsoever.
System is configured remotely over ssh - no keyboard or mouse attached.

Two-Worlds, 2022-07-27:
And here's the debug info:
## Platform Information

Revision	: 9000c1
Description:	Raspbian GNU/Linux 11 (bullseye)
5.15.32+

## Touchscreen logs

Rectangular: Goodix
[   43.514543] Goodix-TS 11-0014: supply AVDD28 not found, using dummy regulator
[   43.541565] Goodix-TS 11-0014: supply VDDIO not found, using dummy regulator
[   43.612018] Goodix-TS 11-0014: i2c test failed attempt 1: -6
[   43.661501] Goodix-TS 11-0014: i2c test failed attempt 2: -6
[   43.712155] Goodix-TS 11-0014: I2C communication failure: -6
[   43.713824] Goodix-TS 11-005d: supply AVDD28 not found, using dummy regulator
[   43.714236] Goodix-TS 11-005d: supply VDDIO not found, using dummy regulator
[   43.716009] Goodix-TS 11-005d: ID 911, version: 1060
[   44.350250] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-11/11-005d/input/input0
Square: ft5

## I2C Devices and Mappings

/dev/i2c-1
/dev/i2c-11
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

## Boot Config

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
dtparam=i2s=off
dtparam=spi=off

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1
# Enable Hyperpixel display
dtoverlay=vc4-kms-dpi-hyperpixel4

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
#max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]

dotMorten, 2022-08-01:
I'm using a Zero 2 with 64bit lite Bullseye and same issue: Doesn't show anything. Backlight is on though.

Two-Worlds, 2022-08-01:
> I'm using a Zero 2 with 64bit lite Bullseye and same issue: Doesn't show anything. Backlight is on though.

I fixed the display by installing the drivers manually (the ones embedded in Bullseye clearly don't work). Still haven't got rotation working though…

dotMorten, 2022-08-01:
@Two-Worlds I'm trying to figure out which one to use for the square one though. There are so many branches, and it isn't obvious which one to use. I used the square branch and it seems to set the right hdmi-timings and frame buffer width but doesn't look right. Pretty much all other branches keeps configuring it to the rectangular version (480x800) even for the one that lets you pick during install what display you got. @Gadgetoid any thoughts here?

dotMorten, 2022-08-01:
Aaah square-2021 was the branch to hit. The install script on the readme page doesn't seem to get this right. Would be nice if doc also mentioned Zero is the same as Pi3 etc.

Two-Worlds, 2022-08-02:
Zero=Pi3 - yes, be useful if that were documented up front - I had to do quite a bit of trial & error there

jeffcolabs1, 2023-02-23:
@Two-Worlds   I' m running into the same issue with a Pi Zero W (32bit) and new rectangular screen.  Could you share your working config?   I'm hoping the screen is still good (ordered another).  Main documentation still shows installing the board the wrong side up which is annoying.  Screen got hot.  I do get just a black screen so its not totally dead.

closed 2026-08-25 by Gadgetoid