pimoroni/hyperpixel4 #188: HQ Camera IMX477 ( Libcamera ) Not working with Hyperpixel

schoolpost, 2022-09-07:
I cannot seem to get the IMX477 HQ Camera to work in tandem with the Hyperpixel under Raspbian OS ( Bullseye )

I'm using a Raspberry Pi 4 / 4GB with the Hyperpixel Square LCD. 

I have added the these two lines to the /boot/config.txt:
dtoverlay=imx477
dtoverlay=vc4-kms-dpi-hyperpixel4sq

The overlay for the imx477 is required now given how cameras are operated with Raspbian using Libcamera.  

As it described above, the LCD + Touch works, however the camera is not even detected by the system and Libcamera apps fail to run:
`pi@raspberrypi:~ $ libcamera-hello 
Made DRM preview window
[0:23:46.526639035] [3466]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
ERROR: *** no cameras available ***`

I believe the Hyperpixel is interfering with the i2c on the HQ camera which I believe is exposed via the GPIO pins. Has anyone found a fix or work around for this? 

SPECS:
Revision        : c03112
Description:    Debian GNU/Linux 11 (bullseye)
5.15.61-v8+

schoolpost, 2022-10-10:
After investigating this issue directly with Raspberry Pi Engineer's over the forum: https://forums.raspberrypi.com/viewtopic.php?p=2044865

It appears the hyperpixel's hardware/software is interfering with the Raspberry Pi HQ Camera ( possibly the other camera models too ) 

By simply having the Hyperpixel attached the to GPIO of the Raspberry Pi ( without the overlay being enabled ) it will interfere with the camera and cause it to not initialize properly:
`
pi@raspberrypi:~ $ dmesg | grep imx
[    7.437015] imx477 0-001a: failed to read chip id 477, with error -5
[    7.438187] imx477: probe of 0-001a failed with error -5
`

Pimoroni should investigate.

schoolpost, 2022-10-13:
Thanks to the hard work by the Raspberry Pi Engineer's ( 6by9 ) a workaround/fix has been determined, add these 3 lines to your /boot/config.txt if you are having the issues I have described above:
`
disable_poe_fan=1
force_eeprom_read=0
camera_auto_detect=0
`

closed 2022-10-13 by schoolpost

6by9, 2022-10-14:
As more detail, HyperPixel4 Square pulls GPIOs 0&1 low more strongly than the internal Pi pull-ups. This upsets the I2C peripheral into thinking it is clock stretching when checking for a HAT EEPROM or PoE HAT, and something in the state isn't reset when Linux takes over. As the same I2C controller is used for the camera (although on different GPIOs) this then causes the I2C reads to the camera to fail.

Oddly HyperPixel4 (not square) doesn't pull 0&1 low, and therefore is quite happy.

feiticeir0, 2023-02-19:
Thank you !
That solved my problem. Beside adding the 3 lines, I didn't had the dtoverlay=imx477 and was not working. After adding that, everything works. !

cpietsch, 2024-09-27:
this does not work on a cm4 with CM4-NANO-A board.
would a clock-frequency added like this `dtoverlay=imx290, clock-frequency=37125000` solve the issue ?

6by9, 2024-09-27:
The clock-frequency override on the imx290 overlay tells the system whether the manufacturer of the imx290 module has used a 37.125 or 72.5MHz oscillator on the camera board. Nothing to do with I2C or GPIOs.

CM4-NANO-A routes out CAM0 to the camera connector, and uses GPIOs 0&1 for the I2C.
The [wiki](https://www.waveshare.com/wiki/CM4-NANO-A) states as much
> CM4-NANO only uses CAM0, so you only need to add "dtoverlay=imx219,cam0".

Seeing as the HyperPixel's DPI interface also uses GPIOs 0&1 you can't use the two at the same time on that board.

If you used a CM4IO board then you should be able to use CAM1 (with I2C on GPIOs 44&45) at the same time as a HyperPixel4 display.

cpietsch, 2024-09-27:
@6by9 thanks for the quick answer. I was fiddling with this for some time now and you provided the answer spot on.
The CM4-NANO-B has a CAM1 out, so could be usable maybe.
All I need is a tiny CM4 board that routes out cam1 and GPIO for hyperpixelsq.

6by9, 2024-09-27:
> The CM4-NANO-B has a CAM1 out, so could be usable maybe. 

I don't believe so. The schematic says that it is also using CAM0 and GPIOs 0&1 - https://files.waveshare.com/upload/f/f5/CM4-NANO-B-SCH.pdf

justDabuK, 2024-12-27:
In full hope that someone here knows why it doesn't work for me. 

I do have a 
- raspberry pi 4B (with a fresh bookworm raspbian installed)
- hyperpixel XP 4.0 square
- camera module 3

I tried adding 
```
dtoverlay=imx477
dtoverlay=vc4-kms-dpi-hyperpixel4sq
disable_poe_fan=1 
force_eeprom_read=0 
camera_auto_detect=0
```
 to my `/boot/firmware/config.txt` and the screen works just fine, but the camera resists to work and I always get an 
 ```
ERROR: *** no cameras available ***
```
when testing it via `rpicam-hello`. 

Am I missing something? Should the config for the camera module 3 be different from what is necessary for the HQ camera? 
Any help would be highly appreciated 🙏🏽

cpietsch, 2024-12-27:
hey @justDabuK : the camera module 3 is a sony imx708 so you will need `dtoverlay=imx708`

justDabuK, 2024-12-27:
thank you so much @cpietsch , that solved it 🥳 

Such that I can learn something from this and hopefully resolve it myself in the future, where did you find the information that it is a sony imx708 chip? 🤔

cpietsch, 2024-12-27:
have a look at https://www.raspberrypi.com/products/camera-module-3/

justDabuK, 2024-12-27:
Ah right first line in Specification
> Back-illuminated, stacked CMOS 12-megapixel Sony IMX708 image sensor

Thanks a lot 🙏🏽

cpietsch, 2024-12-31:
@6by9 I am now on a CM5 + IO Board using CAM1 for the camera. As soon as I add dtoverlay=imx708 to the config the hyperpixel stays black after boot. I hoped using CAM1 would route around the problem :/

cpietsch, 2025-01-01:
Here is the output of [dmesg-hypersq-im708.txt](https://github.com/user-attachments/files/18286085/dmesg-hypersq-im708.txt)
When loading the imx708 this error occurs:
```
[    4.941678] pinctrl-rp1 1f000d0000.gpio: pin gpio0 already requested by 1f00070000.i2c; cannot claim for 1f00148000.dpi
[    4.941685] pinctrl-rp1 1f000d0000.gpio: pin-0 (1f00148000.dpi) status -22
[    4.941687] pinctrl-rp1 1f000d0000.gpio: could not request pin 0 (gpio0) from group gpio0  on device pinctrl-rp1
[    4.941690] drm-rp1-dpi 1f00148000.dpi: Error applying setting, reverse things back
```

cpietsch, 2025-01-01:
I wonder if dtoverlay=cm-swap-i2c0 would help, but as soon I add dtoverlay=imx708 or dtoverlay=imx708,cam1 the hyperdisplay stays black

6by9, 2025-01-02:
`[    4.941678] pinctrl-rp1 1f000d0000.gpio: pin gpio0 already requested by 1f00070000.i2c; cannot claim for 1f00148000.dpi` is indeed the reason for the failure. Just due to the way BSC0 is handled with the dual pinmuxing, it claims 2 sets of pins even when the second isn't desired.

Yes a workaround for Pi3B/3B+/4/02W/CM4 would be `dtoverlay=cm-swap-i2c0,i2c0-gpio44,i2c10-gpio44` to map both buses to the same set of pins. 

For Pi B+/2B/0W/CM1/CM3/CM4S it would be `dtoverlay=cm-swap-i2c0,i2c0-gpio28,i2c10-gpio28`.

For Pi5/CM5 I need to have more of a think, and it also depends on which I/O board (CM4IO vs CM5IO) is used for CM5. The mux isn't actually needed on those as there are sufficient I2C controllers that each CAM/DISP connector has a dedicated one.

cpietsch, 2025-01-04:
I am using the official cm5 IO board. The goal is to create a custom light carrier board that allows me to use the cam1 and the hyperpixel together. Right now I am using the zero2 but am limited by the WiFi and perf.

Was hoping to test the setup using the carrier board before designing the custom board ( which should be quite simple ).

6by9, 2025-01-06:
I'd forgotten the CM5 is slightly different from previous boards.

Testing CM5 on a CM5IO board.
```
dtoverlay=imx477,cam0
dtoverlay=vc4-kms-dpi-hyperpixel4
enable_uart=0
```
And yes you MUST use CAM/DISP0.
CAM/DISP1 uses GPIOs 0&1 (via jumpers on J6) for the I2C that you also want for DPI.

(My display is slightly red. There shouldn't be any issue with the RGB666_1X24_CPADHI DPI mode on CM5/Pi5, but I'm not investigating unless others report similar issues).

cpietsch, 2025-01-06:
ok wow. that worked - out of the box. 🤦‍♂️
somehow I was thinking that I have to use CAM1 to not run into the issue, but that was probably only true for cm4

> If you used a CM4IO board then you should be able to use CAM1 (with I2C on GPIOs 44&45) at the same time as a HyperPixel4 display.