pimoroni/hyperpixel4 #190: HyperPixel Square (Version 2021) not working after update to Bullseye

JJSS-Johannes, 2022-10-31:
I upgraded my Raspberry Pi 3 Model A Plus from Buster to Bullseye. Display is not working any more. 

Display shows top bar of X-window. Left part of screen is flickering.

https://github.com/pimoroni/hyperpixel4/issues/177
I also tried the online installer.

Square with touch purchased in January 2021, connected with the 40 pin gpio - header

Raspberry Pi 3 Model A Plus Rev 1.0
Raspbian GNU/Linux 11 (bullseye)
I didn't try new install yet.

Kind regards,
Johannes

Gadgetoid, 2023-01-20:
Ahoy,

Are you still having trouble with this?

IIRC when the Pi team were kind enough to write some newer drivers for us, the original HyperPixel 4 Square got overlooked. It may also be physically incompatible with how Bullseye does things- since the move from just bunging a few lines into `config.txt` to properly supporting screens via video drivers is quite significant and has *many* caveats that didn't exist when we first developed these boards.

ashenshugarRET, 2023-01-22:
Hello,

I have the same problem with an older square touch screen and a fresh image of the newest Bullseye.

Thanks

JJSS-Johannes, 2023-01-22:
Hello,
It would be great if there is a solution on Bullseye, problem still persists.
Thank You

closed 2026-08-25 by Gadgetoid