pimoroni/hyperpixel4 #191: hyperpixel and i2c devices on pi4

Dtanguy, 2022-11-16:
Hello, i am trying to use i2c devices on a pi4 with a hyperpixel4 rectangular screen.

I have this config in /boot/config.txt :
`dtoverlay=vc4-kms-dpi-hyperpixel4,rotate=180`

When i enable i2c in /boot/config.txt the screen stop working.

I have also try creating another i2c port by adding :
`dtoverlay=i2c-gpio,bus=3,i2c_gpio_delay_us=1,i2c_gpio_sda=17,i2c_gpio_scl=27`

How can i use i2c device and the hyperpixel ? Thanks.

cpietsch, 2023-06-18:
I have the same issue after enabling raspicam.

neozenith, 2023-11-09:
Yeah similar fight getting HyperPixel4 Touch Rectangular working with a PiJuice Hat. Probably a bit too ambitious but I think there are sufficient pins if I can just remap where they are talking.

These docs from Raspberry Pi about DPI is helpful:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#parallel-display-interface

As well as this whitepaper has some details about different colour modes in the protocol that reduce the number of pins required. Just depends if the Hyperpixel4 supports those colour modes. 
https://pip.raspberrypi.com/categories/685-whitepapers-app-notes/documents/RP-003471-WP/Using-a-DPI-display.pdf

[image]

Here are some key extracts from the whitepaper where they mention specifically the Hyperpixel4

[image]
[image]
[image]

neozenith, 2023-11-09:
Actually upon reading this pinout page:
https://pinout.xyz/pinout/pijuice
[image]

And:
https://pinout.xyz/pinout/hyperpixel4
[image]

**_It looks like the direct 40 pin stacking collides on GPIO pins 3,5._** 😭 

Although the Hyperpixel4 exposes 5 pinouts on the underside of the board:
[image]

That pass through might be your best bet to getting extra I2C devices talking.

But there isn't exactly instructions on how to get that working.

neozenith, 2023-11-09:
https://github.com/PiSupply/PiJuice/blob/master/README.md#technical-specs-summary
> The EEPROM can be disabled and its I2C address changed for increased compatibility with other boards

I might be in luck here to set up the PiJuice Hat to get out of the way of the Hyperpixel4.

AlexanderProd, 2023-11-11:
I'm just wondering why the screen wont work as soon as I turn on I2C using `raspi-config` even when no other device is connected.

Gadgetoid, 2023-11-11:
> I'm just wondering why the screen wont work as soon as I turn on I2C using `raspi-config` even when no other device is connected.

radio-config enables *an* i2c bus on specific pins- so you’re telling your Pi to use the display vsync and hsync pins as i2c. They can’t do both.

i2c - on the touch pins- is already provided by the HyperPixel driver. It should be visible as “/dev/i2cN” but since most software assumes 1 you might need to symlink that to avoid modifying the software itself.

You might also IIRC need to “modprobe i2c-dev” for the device node to show up.

neozenith, 2023-11-14:
### Bookworm Config Location Changed

So it took me longer than it should have to realise that using Raspbian Bookworm, they moved `/boot/config.txt` to `/boot/firmware/config.txt` as to why my changes stopped working. 🤦 

[This](https://github.com/neozenith/OAK-D-Lite-Autonomous-LEGO-audi-rs-q-e-tron-42160/blob/f7cf8db38139105fd245959c53466ced7348d891/scripts/config.txt#L65-L66) is my config for the Hyperpixel4 touch on Rasbian Bookworm and it "just works".

```
# https://github.com/pimoroni/hyperpixel4/issues/177
dtoverlay=vc4-kms-dpi-hyperpixel4
```

### Simple i2c testing with BME680

Regarding this comment:

> i2c - on the touch pins- is already provided by the HyperPixel driver

@Gadgetoid did you mean this new driver provided by Raspbian handles this too? 
Or only the driver in this repo?

Any ideas where this new driver is being maintained?

I scaled back my testing from the PiJuice to just using a [BME680 environmental sensor](https://core-electronics.com.au/gravity-i2c-bme680-environmental-sensor-voc-temperature-humidity-barometer.html) and a [JST-PH to male DuPont](https://core-electronics.com.au/pimoroni-hyperpixel-4-0-i2c-cable-jst-ph-to-male-dupont.html) connector on the underside Hyperpixel4 i2c connector.

This device in theory should have an address of `0x77`.

I run:
```sh
$ ls /dev/i2c*

/dev/i2c-20
/dev/i2c-21
/dev/i2c-22
```

`/dev/i2c-22` is the new one that appears when I have the Hyperpixel4 attached and the other 2 already existed when it is not attached.

I ran:
```sh
modprobe i2c-dev
i2c-detect -y 22
```

I didn't capture the output but I couldn't see anything that would indicate a device with address `0x77` was attached. I think I did see a list of the touchscreen addresses though.

(_I'll take some time on the weekend to try again and update this issue with exact output._)

closed 2026-08-25 by Gadgetoid