pimoroni/hyperpixel4 #192: pygame fullscreen shifted down but desktop ok

berarduc, 2022-11-22:
**Describe the bug**

rPi4 / 4GB running buster
kernel 5.10.17-v71+

installed hyperpixel using 1-line installer (legacy version)
running currently in landscape mode using 'hyperpixel4-rotate right' command (but problem manifests itself in 'normal' portrait mode as well).

raspberry desktop looks fine, but when running pygame program that attempts to open a window in fullscreen mode, the window opens up shifted about 1/3 of the way down the y-axis.

What's gone wrong?

uh - I don't know that's why I'm filing a bug :)

as I said desktop looks fine so it appears drivers are working, yet pygame doesn't seem to understand the extents of the screen?

**To Reproduce**

Simple pygame program to reproduce the effect. This python program starts pygame, opens a screen in fullscreen mode, and then fills it with the color red. You will see the red does not cover the entire screen rather top part is black:

import pygame
from pygame import FULLSCREEN
def main():
 pygame.init()
 screen = pygame.display.set_mode([0,0],FULLSCREEN)
 screen.fill((0xff,0,0))
 pygame.display.flip()
 dummy - input("waiting so you can see the display - hit CR to exit")
 pygame.exit()
main()

**Your HyperPixel 4**

1. Is it Square or Rectangular? --> Using rectangular hyperpixel4
2. Touch or non Touch? --> touch enabled and working
3. Approximately when was it purchased ---> purchased over a year ago at least
4. How is it connected to your Pi ---> via GPIO (is there any other way? )

**Your Raspberry Pi*

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
cat/proc/cpuinfo:

[image]

3. Which OS release are you using: `lsb_release --description` and `uname -r`
uname -r: 5.10.17-v71+
lsb_release --description:  Raspian GNU/Linux 10 (buster)

5. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on? --> fresh install. ran sudo apt-get update also.

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors: --> NO problems with touch.

berarduc, 2022-11-24:
Note this issue is a duplicate of issue #193 - leaving it open cause there is supporting info here.

Gadgetoid, 2023-01-20:
See my reply to #193

closed 2023-01-20 by Gadgetoid