pimoroni/hyperpixel4 #198: Not working on Raspberry Pi OS 2023-02-21 on Pi 4 (((pls halp)))

dbech, 2023-03-19:
**PLEASE NOTE**

:warning: GitHub issues are for reporting problems with the software, tech support do not monitor these.
Our software team are smol and busy, and you will not receive a timely response.
If you're having a hardware issue with your Hyperpixel 4, you should contact support@pimoroni.com.

:warning: Hyperpixel 4 is only officially supported on Raspberry Pi OS. There are too many operating systems with too many quirks to begin to support them all. Issues relating to other operating systems will be closed.

**Describe the bug**

Not working on Raspberry Pi OS 2023-02-21 on Pi 4, was working on previous version (I forget which, it was after the overlay was released)

**To Reproduce**

Set overlay and boot, no dice

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or Rectangular? Rectangular
2. Touch or non Touch? Touch 
3. Approximately when was it purchased
4. How is it connected to your Pi? GPIO header

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
Revision        : d03114
3. Which OS release are you using: `lsb_release --description` and `uname -r`
Description:    Debian GNU/Linux 11 (bullseye)
5.15.84-v8+
5. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?
Phresh

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular
[    6.736692] Goodix-TS 22-0014: supply AVDD28 not found, using dummy regulator
[    6.766964] Goodix-TS 22-0014: supply VDDIO not found, using dummy regulator
[    6.769477] Goodix-TS 22-0014: i2c test failed attempt 1: -6
[    6.797112] Goodix-TS 22-0014: i2c test failed attempt 2: -6
[    6.824462] Goodix-TS 22-0014: I2C communication failure: -6
[    6.869142] Goodix-TS 22-005d: supply AVDD28 not found, using dummy regulator
[    6.909040] Goodix-TS 22-005d: supply VDDIO not found, using dummy regulator
[    6.911828] Goodix-TS 22-005d: ID 911, version: 1060
[    6.940839] input: Goodix Capacitive TouchScreen as /devices/platform/i2c@0/i2c-22/22-005d/input/input0

* `dmesg | grep ft5` for HyperPixel 4.0" Square

And check i2c is working:

`ls /dev/i2c-*` 

You should see something like:

```text
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-7
```

/dev/i2c-20  /dev/i2c-21  /dev/i2c-22

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)

For example:

```text
pi@raspberrypi:~ $ i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

 i2cdetect -y 22
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- UU -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

kisdia, 2023-05-13:
What is your /boot/config.txt?
You may need to comment out i2c-arm enable

dbech, 2023-05-17:
this is all I have uncommented in /boot/confit.txt

`
dtoverlay=vc4-kms-dpi-hyperpixel4
dtparam=audio=on
display_auto_detect=1
dtoverlay=vc4-kms-v3d
max_framebuffers=2
arm_64bit=1
disable_overscan=1
[cm4]
otg_mode=1
[all]
`

Gadgetoid, 2023-05-17:
For reference here's mine on a working setup:

```
dtparam=audio=on
camera_auto_detect=1
display_auto_detect=1
dtoverlay=vc4-kms-v3d
max_framebuffers=2
disable_overscan=1
[cm4]
otg_mode=1
[all]
[pi4]
arm_boost=1
[all]
dtoverlay=vc4-kms-dpi-hyperpixel4
```

The presence of `arm_64bit` in your config suggests 64bit Raspberry Pi OS, which might be the culprit. Switching to 64bit to see if I can replicate.

dbech, 2023-05-17:
Yes I am using the 64bit version. 
I also suspect this might be the issue as I recall this screen working flawlessly on a RPi3 running the 32bit OS.

dbech, 2023-05-17:
I just tried the same screen on a 3B with the 32bit OS, I even tried the legacy driver and it didn't work.

I think my display is just dead, RIP to a real one.

closed 2023-05-17 by dbech

Gadgetoid, 2023-05-17:
That’s unfortunate, sure it’s not transposed on the GPIO or loose, and that the ribbon is seated correctly? I had some I pulled out of storage recently that were super flaky until I (carefully) reseated the ribbon.