pimoroni/hyperpixel4 #200: Works in console, stops working in desktop env

addohm, 2023-05-09:
**PLEASE NOTE**

:warning: GitHub issues are for reporting problems with the software, tech support do not monitor these.
Our software team are smol and busy, and you will not receive a timely response.
If you're having a hardware issue with your Hyperpixel 4, you should contact support@pimoroni.com.

:warning: Hyperpixel 4 is only officially supported on Raspberry Pi OS. There are too many operating systems with too many quirks to begin to support them all. Issues relating to other operating systems will be closed.

**Describe the bug**

The display works perfect until I set the pi to start into desktop.  I can see it goes through the boot sequence but once it startx it goes black (but still backlit).  I've skinned this cat 9 different ways and every variation produces the same results.  Console good, X bad.

**To Reproduce**

sudo apt update
sudo apt upgrade 
sudo apt dist-upgrade
sudo apt clean
sudo reboot
sudo /home/addohm/scripts/install-desktop
sudo reboot
git clone https://github.com/pimoroni/hyperpixel4 -b pi4
cd hyperpixel4
sudo ./install.sh
edit /boot/config.txt - dtoverlay=vc4-fkms-v3d
sudo reboot
(Display works at this point)
sudo raspi-config - boot to desktop
(Display stops working once boot reaches startx)

**Your HyperPixel 4**

Let us know which HyperPixel 4 board you're using. Note: if you're having a problem with the original HyperPixel you should go to: https://github.com/pimoroni/hyperpixel

1. Is it Square or **Rectangular**?
2. **Touch** or non Touch?
3. Approximately when was it purchased
Last week
4. How is it connected to your Pi
Directly using the riser header

**Your Raspberry Pi**

Give as much detail about your Pi and OS as possible. We only officially support Raspbian, but might be able to point you in the right direction if the problem is with another OS.

1. What model of Raspberry Pi are you using (the result of `cat /proc/cpuinfo | grep Revision` can tell us this exactly
Revision        : b03115

2. Which OS release are you using: `lsb_release --description` and `uname -r`
Description:    Raspbian GNU/Linux 11 (bullseye)

3. Is this a fresh OS setup, or one you might have previously installed scripts/add-ons on?
"Other specific purpose OS" > Octopi

**Extra debugging information**

If you're having a problem with touch, try checking `dmesg` for related errors:

* `dmesg | grep Goodix` for HyperPixel 4.0" Rectangular*
[    7.119080] Goodix-TS 22-0014: supply AVDD28 not found, using dummy regulator
[    7.119487] Goodix-TS 22-0014: supply VDDIO not found, using dummy regulator
[    7.126802] Goodix-TS 22-0014: Error reading 1 bytes from 0x8140: -6
[    7.154612] Goodix-TS 22-0014: Error reading 1 bytes from 0x8140: -6
[    7.181055] Goodix-TS 22-0014: I2C communication failure: -6
* `dmesg | grep ft5` for HyperPixel 4.0" Square

And check i2c is working:

`ls /dev/i2c-*`
/dev/i2c-22

You should see something like:

```text
pi@raspberrypi:~ $ ls /dev/i2c-*
/dev/i2c-7
```

And your HyperPixel 4 touch is showing up (there should be an address blocked out with UU in the below command):

`i2cdetect -y X` ( where X is the number of i2c bus found in the command above)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- 5d -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

For example:

```text
pi@raspberrypi:~ $ i2cdetect -y 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

addohm, 2023-05-09:
Disabled kms and fkms.  Problem solved.

Gadgetoid, 2023-06-13:
I think this is an Octopi issue, since this works on Raspberry Pi OS. At an absolute guess it doesn't do any of the weird monkeying with /usr/share/dispsetup.sh that Pi OS uses to make displays work.

Do you see anything if you `cat /usr/share/dispsetup.sh`?

What Window manager/desktop environment does `sudo /home/addohm/scripts/install-desktop` install?

Without kms you're at the behest of whatever window manager you're using to handle rotation in software, effectively, which isn't very desirable.

addohm, 2023-06-22:
> I think this is an Octopi issue, since this works on Raspberry Pi OS. At an absolute guess it doesn't do any of the weird monkeying with /usr/share/dispsetup.sh that Pi OS uses to make displays [...]

The _only_ way I have been able to get this to work on the pi3a is disabling kms and fkms.  I've spent two+ days on this.  The pi3b+\pi4 method actually works but I couldn't get the screen to rotate in ratpoison.

closed 2026-08-25 by Gadgetoid